Edzelf/ESP32Radio-V2 #2: Sound with major interrupts on VS1003

skartalov, 2021-10-06:
I am using VS1003 insted of VS1053.

The Terminal shows correct about using VS1003, but when the stream starts it is interrupted a lot.
Like 0.5 sec sound, 5 sec - no sound.
On every stations.

What could be the issue? Did someone else used VS1003 ?

Edzelf, 2021-10-06:
You have the wrong files in SPIFFS.  In the data directory there should be a file "config.html" without the "MP3 player" tab:
```
 <ul>
   <li><a class="pull-left" href="/index.html">Control</a></li>
   <li><a class="pull-left" href="/search.html">Search</a></li>
   <li><a class="pull-left active" href="/config.html">Config</a></li>
   <li><a class="pull-left" href="/about.html">About</a></li>
  </ul>
```

skartalov, 2021-10-07:
Yes, first I have build the WEB-RADIO-V1 project in it.
But how to start over? How to erase SPIFFS and initialize again with current project?