Edzelf/ESP32Radio-V2 #12: ST7789

skartalov, 2021-11-07:
I spend few days trying to set the project to work with ST7789 controller, but no success.
(I use small 240x240resolution ST7789 lcd)

Can you make it work?

As you already include Adafruit library for ST7735 and ST7789, it should be easy. Seems like I am missing something...

skartalov, 2021-11-19:
In fact ST7789 works with ILI9341 drivers! I dont know how, but it is inversed. Anyway i spend 3 days trying to config ST7789 to work, but not success.
Would be grateful if someone helps me with this issue.
I have designed ultra compact size PCB for this project.[image]
[image]

timoluebeck, 2022-01-20:
Could you share your Code with me?
Regards