Edzelf/ESP32Radio-V2 #15: No an issue exactly ;-)

skartalov, 2021-12-25:
This is what I came up with this modified project:

https://youtu.be/CQTEDK5EOqk

Ultracompact size: 95x30x10mm enclosure.
Hope you like it.

Edzelf, 2021-12-25:
Nice!
More pictures?

skartalov, 2021-12-25:
OK, more pictures!

One thing I want to ask you. Would you help me to make the MP3 player function work in this revision?

I have microSD card on my PCB, so it would be nice if it can be used.

I can ship one assembled unit to you so you can work on it.

RDS

Simeon

[image]
[image]
[image]
[image]

Edzelf, 2021-12-25:
You got a deal!

skartalov, 2021-12-25:
Great, please let me know your shipping address on my email:  spotlite@abv.bg