Edzelf/ESP32Radio-V2 #16: Simple Question (not an issue)

xmicron, 2021-12-28:
Great project! Thanks!

I would like to ask what you think is better to use: (less sound glitches, less decoding formats..etc)
1) The VS1053 to do all the steam decoding on all supported formats
2) An I2S -> DAC 

I would like to design a pcb and I can't choose what is better in the long run.
Thanks

Edzelf, 2021-12-28:
I think option 1 is the best.  It will save some CPU power.  The second option will save some PCB/housing space.
If you already have the VS1053, go for option 1.
For the lowest price, or for a commercial project, go for option 2.

xmicron, 2021-12-28:
Thanks for your quick response.

Against functionality? Decoding glitches and stability? 
Which one will do a better job?