Edzelf/ESP32Radio-V2 #21: Right or Wrong ESP32 board

wa8lbz, 2022-01-31:
The Hardware section of the ESP32-radio V2.0 Manual states that the ESP32 module is basically an ESP32 on a small print. You used a DOIT ESP32 Development Board.

In the Arduio IDE version 1.8.19, the closest board I could select is the DOIT ESP32 DEVKIT V1. Unfortunately the vitrifaction of the sketch stops with the this error message: exit status 1  error compiling for board DOIT DEVKIT V1.

This probably indicates that I have selected the wrong ESP32 board. Do you know which ESP32 board I should select?

Edzelf, 2022-01-31:
The board is okay.  But there should be more error info than the exit status.  Please find that info.

wa8lbz, 2022-02-01:
Here is the information you requested:

Arduino: 1.8.19 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Al Mecozzi\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\ALMECO~1\AppData\Local\Temp\arduino_build_891436 -warnings=none -build-cache C:\Users\ALMECO~1\AppData\Local\Temp\arduino_cache_522243 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Al Mecozzi\Documents\Arduino\ESP32RadioV2\ESP32RadioV2.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Al Mecozzi\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\ALMECO~1\AppData\Local\Temp\arduino_build_891436 -warnings=none -build-cache C:\Users\ALMECO~1\AppData\Local\Temp\arduino_cache_522243 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\Al Mecozzi\Documents\Arduino\ESP32RadioV2\ESP32RadioV2.ino

Using board 'esp32doit-devkit-v1' from platform in folder: C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

Using core 'esp32' from platform in folder: C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

cmd /c if exist "C:\\Users\\Al Mecozzi\\Documents\\Arduino\\ESP32RadioV2\\partitions.csv" copy /y "C:\\Users\\Al Mecozzi\\Documents\\Arduino\\ESP32RadioV2\\partitions.csv" "C:\\Users\\ALMECO~1\\AppData\\Local\\Temp\\arduino_build_891436\\partitions.csv"

cmd /c if not exist "C:\\Users\\ALMECO~1\\AppData\\Local\\Temp\\arduino_build_891436\\partitions.csv" copy "C:\\Users\\Al Mecozzi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "C:\\Users\\ALMECO~1\\AppData\\Local\\Temp\\arduino_build_891436\\partitions.csv"

Detecting libraries used...

"C:\\Users\\Al Mecozzi\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Al Mecozzi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\Al Mecozzi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-IC:\\Users\\Al Mecozzi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-IC:\\Users\\Al Mecozzi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-IC:\\Users\\Al Mecozzi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" "-IC:\\Users\\Al Mecozzi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bt" "-IC:\\Users\\Al Mecozzi\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/coap" "-IC:\\Users\\Al Mecozzi\\App [... cut ...]
Using library AsyncTCP at version 1.1.1 in folder: C:\Users\Al Mecozzi\Documents\Arduino\libraries\AsyncTCP 

Using library SPIFFS at version 1.0 in folder: C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SPIFFS 

Using library Adafruit_VS1053_Library at version 1.2.1 in folder: C:\Users\Al Mecozzi\Documents\Arduino\libraries\Adafruit_VS1053_Library 

Using library SD at version 1.0.5 in folder: C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD 

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

Edzelf, 2022-02-01:
Wow, that's quite a story.  I searched for the first occurence of "fatal" and found:
`\main.cpp:99:82: fatal error: utils.h: No such file or directory`
So the compiler cannot find "utils.h".  That means that you did not include this file in the working directory.  I will make a description on how to compile the project with Arduino IDE and include this in the documentation.  I let you know when it is ready for use.

Edzelf, 2022-02-01:
See the new ESP32-radio.odt file in the doc directory, chapter 15.

wa8lbz, 2022-02-01:
I did exactly as specified in Chapter 15 and still got the following error message: Arduino: 1.8.19 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None"

C:\Users\Al Mecozzi\Documents\Arduino\libraries\Adafruit_VS1053_Library\Adafruit_VS1053.cpp:26:29: fatal error: Adafruit_VS1053.h: No such file or directory

compilation terminated.

Multiple libraries were found for "SD.h"

 Used: C:\Users\Al Mecozzi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD

 Not used: C:\Program Files (x86)\Arduino\libraries\SD

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I should state that the DOIT ESP32 DEVKIT 1 board is not connected to any circuit. The board is connected directly to the computer using a USB cable only. It is using COM7 at an upload speed of 115200.

I really want to get this project working. I had success with your ESP8266 radio. I will keep plugging.

Edzelf, 2022-02-01:
Please remove the adafruit vs1053 library.

wa8lbz, 2022-02-02:
Removed Adafruit VS1053 library but got this error message while verifying the sketch:

Arduino: 1.8.19 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None"

ESP32RadioV2:110:79: fatal error: VS1053.h: No such file or directory

compilation terminated.

exit status 1

VS1053.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Edzelf, 2022-02-02:
VS1053.h should be in your working directory if you followed the procedure in chapter 15 of the document.  Please show the contents of your working directory.

wa8lbz, 2022-02-02:
Your are correct, I did miss that line. I have fixed this. Here is a image of my work directory.

[image]

Now, when verifying the sketch, it stops at line 4061 with this error message:  exit status 1 'TFTSECS' was not declared in this scope.

I'm still plugging away. I do want to make this project work. If you need verbose  error message I will provide it.

Edzelf, 2022-02-02:
You need at leidt one display driver in your working directory.  Check chapte4 15 again.
Include and nextion should not be in your working directory.

wa8lbz, 2022-02-02:
Here is my work directory no after you last reply.
[image]

Now verifying the sketch stops at line 544 with this error message: exit status 1 bluetft.h: No such file or directory.

Edzelf, 2022-02-02:
You need that file if you configured bluetft in configh.  Show the config.h file, after you have edited it for your set-up.

wa8lbz, 2022-02-03:
config.h has been changed as shown below:

[image]

Finally success Ed. The sketch verified with out any error messages. I have already flashed the preferences into the ESP32 with no problems. Next to compile and load the sketch into the ESP32. I will keep you posted on this.

Edzelf, 2022-02-07:
You do not need the ino file.  After uploading the sketch and the data directory, connect to the AP at 192.168.4.1 and use the config page of the webinterface to configure WiFi and other settings.

wa8lbz, 2022-02-09:
I am now using Platformio. I uploaded  main.cpp to the ESP32 DOIT DEVKIT V1 and did pio run -t upload and pio run -t uploadfs. This happened without a problem. When finished I received this information in the serial monitor:

> Executing task in folder ESP32Radio-V2: C:\Users\Al Mecozzi\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- forcing DTR inactive
--- forcing RTS inactive

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Miniterm on COM7  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 06 Oct 2021 14:35:00 GMT.  Free memory 298180
D: Display type is NEXTION
D: SPIFFS is okay, space 1378241, used 28614
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 6 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_vs_cs set to -1  
D: pin_vs_dcs set to -1 
D: pin_vs_dreq set to -1
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_i2s_bck set to -1
D: pin_i2s_lck set to -1
D: pin_i2s_din set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Create list with acceptable WiFi networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - ARRIS-E4B1                Signal: -60 dBm, Encryption WPA_WPA2_PSK,
D:  2 - ARRIS-E4B1_2GEXT          Signal: -64 dBm, Encryption WPA_WPA2_PSK,
D: End of list
D: Command: preset with parameter 0
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 0
D: Connect to WiFi
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
D: Start server for commands
D: Rotary encoder is disabled (-1/-1/-1)
D: STOP requested
D: Starting I2S playtask..
D: Output to internal DAC
D: Playtask stop song

I checked the the defaultprefs.txt file in the Data directory to make sure properly inserted the name/id properly for my two networks. I did. I don't know why the WiFi failed.

Edzelf, 2022-02-09:
Connect your PC/tablet/phone to the ESP32Radio network and go with your browser to 192.168.4.1.  Go to the config page and set your WiFi credentials.  There should be some info in the documentation.

wa8lbz, 2022-02-10:
1. Earlier in this discussion you mentioned that the Esp32-radio-init.ino file was not needed. Then why was it included in the ESP32 Radio project?

2. Is the defaultfs.txt file in the Data directory used for anything other than for the Web Interface?

3. I do not want to use the Web interface to control the radio. Right now I intend to use an IR remote to control the radio. Do I still need to perform the steps in Chapter 6 of the manual?

I hate to be asking you so many questions but I not understanding how all the parts of the project go together. The ESP8266 Radio was much easier for me to accomplish. Do you use Skype? If so, maybe I could ask the questions to you directly. I do thank you for sticking with me on making this project work.

Edzelf, 2022-02-10:
The .ino file is not needed.  You may use it to prefill ​the NVS.
The defaults.txt can be used to speed-up the configuration in the NVS by using the web interface.
Leave the webinterface as it is.  You can do without it, but it will make things more difficult.
About the undefined messages while compiling: it is impossible to have these errors AND having a working version.  So I don't understand.

wa8lbz, 2022-02-11:
Here is what I see when the main.cpp file is loaded into the Platformio IDE:

[image]

Edzelf, 2022-02-12:
Looks like a problem in the NEXTION part.

wa8lbz, 2022-02-23:
Now I have a Nextion NX4024T032 display. Where can I get the Nextion .HMI and .TFT file for this project?

Edzelf, 2022-02-24:
I have added the nextion files to the "nextion" directory.

wa8lbz, 2022-02-24:
Thank you for providing the .HMI and .TFT files for your project. However, when try to load the .HMI into the Nextion Editor (version 1.63.3, the latest), I get the following message: Wrong resource file or resource file has been damaged.

Edzelf, 2022-02-26:
It seems that .HMI files are not supported by github.  I have added a .zip file in the nextion directory.

wa8lbz, 2022-02-26:
No .zip file in the github Nextion folder.

Edzelf, 2022-02-26:
I probably uploaded it in the version 1 repositry.

wa8lbz, 2022-02-26:
Here is my setup for the RESP32 Radio:

1. In the config.h file I have defined only the DEC_Helix (MP3/ACC I2S output) and the NEXTION display.

2. In the defaultprefs.txt file:
   a. Added my local networks names and id's.
   b. Added additional preset streaming radio stations.
   c. Set the clock Daylight Savings Time to 1 and offset to 5.
   d. IR to GPIO 35, i2s_bck_pin to 5, i2slck_pin to 32 and i2s_din_pin to 4.
   e. Set in the IR Remote codes to control the upvolume, downvolume, uppreset, downpreset, mute and default station.

3. In the nextion.h file set NXT_RX_PIN to 16,  NXT_TX_PIN to 17, display width to 400 and display height to 240 to match my Nextion display.

4. Uploaded the main.cpp sketch and executed and the pio run -t upload and pio run -t uploadfs commands. All with no problems.

5. Turned power off and bread boarded the ESP32DOIT-DEVKIT-V1 board, MAX98357A I2S decoder/audio board, Nextion NX4024T032 display module and VS1838B IR receiver board.

6. Applied power. No sound. The background images appeared on the display but no text appeared. The display never moved from page 1 to page 2.

7. The last few line of the platformio serial monitor are:
   D. Connected to my local network
   D. IP=192.168.0.13
   D. Start server for commands
   D. Network found. Starting mp3 client, mqtt and OTA
   D. MDNS responder started
   D. Rotary encoder disabled (-1/-1/-1)
   D. STOP requested
8. Section 2.1 Preferences in the documentation states "Remove the lines ia a device is not connected. I will be controlling the ESP32 Radio with the display and IR rmote. So removed pin_enc_clk = 25, pin_enc_dt = 26 and pin_enc_sw = 27. And the same for all the other devices I will not be using.

9. Some additional questions:
   a. If the file ESP32_radio_init.ino is not needed, why is included in the platformio project?
   b. What are pio run -t upload and pio run -t uploadfs commands uploading loading?
   c. In defaultprefs.txt file, pin_i2s_bck = 5, pin_i2s_din = 4 and pin_i2s_lck = 32. However, in the platformio serial monitor is shows pin_i2s_bck = 26, pin_i2s_din = 27 and pin_i2s_lck = 25. How does this happen?

10. I have read the documentation over and over. Either I am missing and/or not understanding something.

Edzelf, 2022-02-26:
A lot of questions.  I will answer some here.
defaultprefs.h is only used if you load and save the defaults in the webinterface config page.
You may use esp32_radio_init.ino to preload the NVS.  Not necessary if you use the config page of the webinterface.  The values you see in the config page of the webinterface are the values that are used in the sketch.
pio run -t upload uploads the sketch, pio run -t uploadfs uploads the SPIFFS from the data directory.

wa8lbz, 2022-02-27:
This time I am sending you the full serial monitor display:
--- Miniterm on COM7  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
D: Connected to ARRIS-E4B1
D: IP = 192.168.0.13      
D: Start server for commands
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400f68bb  PS      : 0x00060830  A0      : 0x800f5405  A1      : 0x3ffe14c0
A2      : 0x00000000  A3      : 0x3f40d649  A4      : 0x0000003f  A5      : 0x3ffcca24
A6      : 0x3ffe14fc  A7      : 0x3ffe14fc  A8      : 0x800f6889  A9      : 0x3ffe1480
A10     : 0x0000000d  A11     : 0x00000040  A12     : 0x3f40d649  A13     : 0x3ffe1540
A14     : 0x3ffe14c0  A15     : 0x00000008  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400f68bb:0x3ffe14c0 0x400f5402:0x3ffe1560 0x400d3c3c:0x3ffe1580 0x4008a692:0x3ffe15b0
  #0  0x400f68bb:0x3ffe14c0 in Print::printf(char const*, ...) at C:\Users\Al Mecozzi\.platformio\packages\framework-arduinoespressif32\cores\esp32/Print.cpp:261
  #1  0x400f5402:0x3ffe1560 in NEXTION_displayvolume(unsigned char) at lib\NEXTION\src/NEXTION.cpp:123
  #2  0x400d3c3c:0x3ffe1580 in handle_spec() at src/main.cpp:4300
      (inlined by) spftask(void*) at src/main.cpp:4317
  #3  0x4008a692:0x3ffe15b0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

I hope this helps.

Edzelf, 2022-02-27:
This is not the full logging.  The first part with the configuration settings is missing.  Also show config.h.

wa8lbz, 2022-02-28:
Here is complete serial monitor printout:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 06 Oct 2021 14:35:00 GMT.  Free memory 298136
D: Display type is NEXTION
D: SPIFFS is okay, space 1378241, used 29869
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 29 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_vs_cs set to -1
D: pin_vs_dcs set to -1
D: pin_vs_dreq set to -1
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_i2s_bck set to 26
D: pin_i2s_lck set to 25
D: pin_i2s_din set to 27
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: Enable pin 35 for IR
D: Create list with acceptable WiFi networks
D: Added XXXXA to list of networks
D: Added XXXXB to list of networks
D: End adding networks
D: Scan Networks
[... 59 lines cut ...]

Rebooting...
ets Jun  8 2016 00:22:57

Here is the the config.h file:

//***************************************************************************************************
// config.h                                                                                         *
//***************************************************************************************************
//
// Define (just one) type of MP3/AAC decoder
//#define DEC_VS1053                                      // Hardware decoder for MP3, AAC, OGG
//#define DEC_VS1003                                      // Hardware decoder for MP3
#define DEC_HELIX                                         // Software decoder for MP3, AAC. I2S output
// Define (just one) type of display.  See documentation.
//define BLUETFT                                           // Works also for RED TFT 128x160
//#define OLED1306                                        // 64x128 I2C OLED SSD1306
//#define OLED1309                                        // 64x128 I2C OLED SSD1309
//#define OLED1106                                        // 64x128 I2C OLED SH1106
//#define DUMMYTFT                                        // Dummy display
//#define LCD1602I2C                                      // LCD 1602 display with I2C backpack
//#define LCD2004I2C                                      // LCD 2004 display with I2C backpack
//#define ILI9341                                         // ILI9341 240*320
#define NEXTION                                         // Nextion display. Uses UART 2 (pin 16 and 17)
//

Edzelf, 2022-03-01:
I have tested with your set-up.  The problem was in the NEXTION option.
I saw a difference is in the logging:
After "`D: Enable pin 35 for IR`", I miss in your logging the lines:
```
D: Start display
D: Init Nextion, pins 16, 17
D: Nextion command cls BLACK
D: Nextion command page 0

```
The NEXTION initialisation was not executed.  I fixed the bug and will upload the new version.

wa8lbz, 2022-03-01:
This is I got when I tried to unzip the Nextion .zip file in the previous version of the ESP32 Radio.
[image]

Also I uploaded the new main.cpp. I got sound on for about 60 seconds then off for 60 seconds and then back on for 60 seconds. This keeps on repeating. There was nothing on the display except the static background. Here is the complete printout of the serial monitor:

[Serial Monitor Printout 2.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8163983/Serial.Monitor.Printout.2.txt)

wa8lbz, 2022-03-02:
I uploaded the new main.cpp file. Now the radio plays for while then stops and then starts again. Maybe it is the station (NL 1). The display still only displays the static Page 0 (P0). No text of startup activities is displayed. I used the .TFT file you left in the earlier ESP32 Radio. It didn't help. I suggest the you place the .HMI AND .TFT files in the Nextion folder in the V2 project. That way others will not have to hunt for them. The problem with the display might be that RX2 and TX2 are not being turned on. If you need more information, let me know what it is and I will provide it.

wa8lbz, 2022-03-05:
Unless I missed them, a search of main.cpp Version Tue 01Feb2022 09:20:00 GMT showed dbgprints as follows:

D: Enable pin 35 for IR is found in line 2355
D: Start display found in line 2361
D: Init Nextion, pins 16, 17 not found
D: Nextion command cls BLACK not fount
D: Nextion command page 0 not found

Radio plays on and off. On, off times vary. Sometimes long, sometime short. Still nothing on the Nextion display except a static page 0. No radio startup information or other information displayed. I hope this additional feedback helps.

Edzelf, 2022-03-05:
What do you mean by "not found"?  Is it "not printed" or "not found in the siyrce code"?

wa8lbz, 2022-03-05:
They are not printed in the serial monitor and they are not found in the code. The radio is playing right now, but turns on and off periodically. This the serial print out that was shown when the radio first came on. Seems to be some problems with Python.
[Serial Monitor Printout.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8191397/Serial.Monitor.Printout.txt)

Edzelf, 2022-03-05:
Line 2362 contains:
dsp_ok = dsp_begin ( INIPARS ) ;                     // Init display
This will call NEXTION_dsp_begin() ;
This will print "Init Nextion, pins....."
See lib/NEXTION/src/NEXTION.cpp line 31.

The python errror has to do with a bad serial connection to your PC.

wa8lbz, 2022-03-06:
The following lines do not show up the the serial monitor as you indicated they should. Not only does the Nextion display not work properly, the IR remote commands for uppreset (FFE21D) and  downpreset (FFA25D) do not work. I am not including the FF in the commands in the code. I have to ESP32DOIT DEVKIT V1 boards they both exhibit the same radio play not play problem. It is rather remote that the UART's on both boards are having a problem. Replaced USB cable but the problem continues. The RX and TX connections between the Nextion display and the ESP32 board have been checked and rechecked. No problem there. I don't know what else could be causing a bad serial connection to the computer. Bottom line is Nextion display and IR remote codes do not work.

Edzelf, 2022-03-06:
Make sure that line 2362 is executed.  Only then the NEXTION should work.

wa8lbz, 2022-03-07:
In line 2362: dsp_ok = dsp_begin (INIPARS)
Cursor over dsp_begin brings up box saying
identifier "dsp_begin" is unidentified C/C++(20).
Same notification for lines 2363 through 2375.

Edzelf, 2022-03-07:
If NEXTION is defined in config.h, you should see:
[image]

So it looks if NEXTION is not defined.

wa8lbz, 2022-03-07:
1. Since I am still learning how to use Platformio, how do I make sure that line 2362 is executed.
2. Since Platformio can't run .ino files, the esp32_radio_init.ino, included with the ESP32Radio project, must be run using the Arduino IDE. Where does it go in ESP32 board memory when it is run?
3. Do you know of anyone who has got the ESP32Radio project to work with a Nextion display and IR receiver? Neither the Nextion display or IR receiver seem to be working.

wa8lbz, 2022-03-15:
There is good news and bad news. The bad news first. The Nextion display still does not display anything except the static P0. The good news is the IR receiver is now working. I had the wrong IR codes for the remote I am using. With the correct codes it works.

I have decided to switch from the I2S decoder/audio board because of the lack of integratedl volume control. I am now using the VS1052 decoder/audio board. Unfortunately no audio output from the headphone jack into headphone. Below is the serial monitor printout.

[Serial Monitor Printout 3.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8256524/Serial.Monitor.Printout.3.txt)

Edzelf, 2022-03-15:
Please supply the logging from the beginning.

wa8lbz, 2022-03-15:
I forgot this printout:
[Debug Printout 1.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8256662/Debug.Printout.1.txt)

wa8lbz, 2022-03-16:
More progress. Replaced some intermittent DuPont wires connecting the VS1053 to the ESP32 and now there is sound coming out of the earphone jack to the headphones. Now, I will continue to work on getting the Nextion display to work.