Edzelf/ESP32Radio-V2 #22: Songs audio play OK but announcer auudio is garbled.

artiezz, 2022-02-15:
Hi,
I have implemented ESP32Radio-V2 and it mostly works. I have added some stations I want, but problem is, while music is playing, it sounds fine. But when the song stops and the announcer comes on, the audio is garbled. An example is tuning into  "wwfm.streamguys1.com/live-mp3." Here is the terminal output:

_D: Start server for commands
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Command: status (without parameter)
D: Command: status (without parameter)
D: Command: status (without parameter)
D: Command: status (without parameter)
D: Rotary encoder is enabled
D: STOP requested
D: New preset/file requested (11/0) from wwfm.streamguys1.com/live-mp3
D: Connect to new host wwfm.streamguys1.com/live-mp3
D: Connect to wwfm.streamguys1.com on port 80, extension /live-mp3
D: Song stopped correctly after 0 msec
D: Connected to MP3 host at 216.246.14.119 on port 80
D: send GET command
D: Switch to HEADER
D: Headerline: Accept-Ranges: none
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:128
D: Headerline: ice-audio-info: channels=2;samplerate=44100;bitrate=128
D: Headerline: icy-br:128
D: Headerline: icy-description:WWFM
D: Headerline: icy-genre:Misc
D: Headerline: icy-name:/live-mp3
D: Headerline: icy-pub:1
D: Headerline: icy-url:WWFM
D: Headerline: Server: Icecast 2.4.0-kh1
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Pragma: no-cache
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 128, metaint is 16000
D: Command: status (without parameter)
D: Metadata block 16 bytes
D: Streamtitle found, 15 bytes
D: StreamTitle='';
D: Command: status (without parameter)
D: Command: status (without parameter)
D: Command: status (without parameter)__

The song stopped and the announcer came on after the second to last line in the log. Is it possible that the stream changes mode to 'MONO' or something when the announcer is on?

This URL plays music and announcer just fine in Winamp and VLC.

Edzelf, 2022-02-16:
This station is bad on my set-up too, always not enough data in the stream.
Are other stations playing normal on your set-up?

artiezz, 2022-02-18:
Some work, some do not. All the stations that I want to listen to exhibit this strange music is OK but announcer is not behavior. Another example is ideastream.streamguys1.com/wclv.mp3, plays music with lots of "clicks." I'm looking into your debug output to see if I can find any more clues.