Edzelf/ESP32Radio-V2 #25: Nextion display does not work

wa8lbz, 2022-03-24:
I have the VS1053 module (MP3/ACC decoder), VS1338B module (IR), KCX_BT_EMITTER module (Bluetooth transmitter), BUT not the Nextion NX4024T032_011 display is not working. Other than the display the radio works fine.

I connected a oscilloscope to the RX2 in on the ESP32 module and see the signal displayed in the attached picture. The signsl stays the same after switching to another signal. I don't know what the signal should look like at the RX2 pin but this is what I see.

Is there anyone who built this project using a Nextion display and got the display working. I have tried everything I know but can't fix the problem. The display is brand new. I will greatly appreciated any help I can get.

[image]

Edzelf, 2022-03-24:
Connect your RIGOL to the TX2 pin of the ESP32 and che k if there is a signal.  Remember that lines should be crossed: TX2 to the Nextion RX and RX2 to the Nextion TX.

wa8lbz, 2022-03-24:
The blue lead on the display is TX and it connected to the RX2 pin on the ESP32. The yellow lead on the display is RX and it is connecte to the TX2 pin on the ESP32. Yes there is a signal on the ESP32 TX2 pin.

Edzelf, 2022-03-24:
I will test this set-up again.

wa8lbz, 2022-03-25:
Here is a new picture showing both the Nextion display RX anf TX. The yellow trace is the RX signal and the blue trace is the TX signal. I changed the sweep range to get both signals on the scope at the same time. I hope this helps.
[image]

Edzelf, 2022-03-26:
I fixed the NEXTION display.

Edzelf, 2022-03-28:
Show me your config.h

Edzelf, 2022-03-28:
It looks like you have defined both NEXTION and DUMMYTFT.  Just define one type of display.

wa8lbz, 2022-03-28:
[Config dot h.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8365406/Config.dot.h.txt)

wa8lbz, 2022-03-28:
After making some changes, I got the pio run -t upload and uploads to upload without problems. However, not only does the Nextion display still not work but the radio no longer plays. This happened after down loading and installing the new zip files.

Edzelf, 2022-03-28:
Just download the project. Compiles without errors, Radio plays, tested with DEC_HELIX_INT and NEXTION.
I found that the NEXTION did not update program info.  Fixed that in NEXTION.h and NEXCTION.cpp.
[image]

wa8lbz, 2022-03-29:
This is what I did:
[What I Did.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8374009/What.I.Did.txt)

Jefke1, 2022-03-29:
Hi, 
I've been following this for a while and decided to give it a try.
I did the test with a pcm5102A and Nextion NX4832T035_011 this is my log, can't connect to wifi also not with web interface. keeps rebooting.

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Sat, 26 Mar 2022 10:50:00 GMT.  Free memory 113792
D: Display type is NEXTION
D: SPIFFS is okay, space 1378241, used 33634
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 50 keys from NVS
D: pin_ir set to 35
D: Pin 0 is already reserved!
D: Pin 0 is already reserved!
D: Pin 0 is already reserved!
D: Pin 0 is already reserved!
D: pin_sd_cs set to 22
D: pin_i2s_bck set to 27
D: pin_i2s_lck set to 25
D: pin_i2s_din set to 26
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start display
D: Init Nextion, pins 16, 17
D: Nextion command cls BLACK
D: Nextion command page 0
D: NEXTION output 'Starting......' to 1
D: NEXTION output 'Sat, 26 Mar 2022' to 2
D: NEXTION output 'By Ed Smallenburg' to 3
D: Create list with acceptable WiFi networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 7
D:  1 - DIRECT-xy46" Philips 8008 Signal: -58 dBm, Encryption WPA2_PSK,
D:  2 - Orange-82fa0              Signal: -70 dBm, Encryption WPA2_PSK, 
D:  3 - Orange-486AE              Signal: -72 dBm, Encryption WPA2_PSK,
D:  4 - Guest-Orange-82fa0        Signal: -76 dBm, Encryption WPA2_PSK, 
D:  5 - Orange-1d6e6              Signal: -80 dBm, Encryption WPA2_PSK,
D:  6 - telenet-2122642           Signal: -90 dBm, Encryption WPA2_PSK,
D:  7 - telenet-05670             Signal: -91 dBm, Encryption WPA2_PSK, 
D: End of list
D: Connect to WiFi
D: NEXTION output 'Connect to WiFi' to 4
D: WiFi Failed!  Trying to setup AP with name ESP32-Radio and password ESP32-Radio.
D: IP = 0.0.0.0
D: NEXTION output 'IP = 0.0.0.0' to 5
D: Nextion command page 1
D: Start server for commands
D: Rotary encoder is disabled (0/0/0)
D: Radiofuncs cmd is 1
D: Connect to host 109.206.96.34:8100
D: Connect to 109.206.96.34 on port 8100, extension /
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400da9df  PS      : 0x00060530  A0      : 0x800daa90  A1      : 0x3ffb1c50
A2      : 0x00000000  A3      : 0x3ffb1ca4  A4      : 0x00001fa4  A5      : 0x00000000  
A6      : 0x3ffb1c90  A7      : 0x00000004  A8      : 0x3f40d764  A9      : 0x3ffb1c60
A10     : 0x00000000  A11     : 0x3ffb1cac  A12     : 0x400da6d8  A13     : 0x00000000  
A14     : 0x00000002  A15     : 0x0000000d  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 0000000000000000

Backtrace: 0x400da9df:0x3ffb1c50 0x400daa8d:0x3ffb1ca0 0x400d2c27:0x3ffb1ce0 0x400d5231:0x3ffb1f40 0x400d95c9:0x3ffb1f90 0x400f970c:0x3ffb1fb0 0x4008a6ee:0x3ffb1fd0
  #0  0x400da9df:0x3ffb1c50 in AsyncClient::connect(IPAddress, unsigned short) at .pio\libdeps\esp32\AsyncTCP\src/AsyncTCP.cpp:1118
  #1  0x400daa8d:0x3ffb1ca0 in AsyncClient::connect(char const*, unsigned short) at .pio\libdeps\esp32\AsyncTCP\src/AsyncTCP.cpp:1118
  #2  0x400d2c27:0x3ffb1ce0 in connecttohost() at src/main.cpp:4451
  #3  0x400d5231:0x3ffb1f40 in radiofuncs() at src/main.cpp:4451
  #4  0x400d95c9:0x3ffb1f90 in loop() at src/main.cpp:3362
  #5  0x400f970c:0x3ffb1fb0 in loopTask(void*) at C:\Users\josva\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #6  0x4008a6ee:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

wa8lbz, 2022-03-29:
Here is another clue. This the Serial Monitor printout when power is applied to the radio. And as far as I can tell the VS1053 is properly installed and it is defined in config.h.
[Serial Monitor 1.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8374796/Serial.Monitor.1.txt)

Edzelf, 2022-03-29:
What is wrong with GPIO18.  This should be the SPI clock,  but it is pulled LOW.

Jefke1, 2022-03-29:
I used a new Esp32, radio plays well, in serial monitor Nextion text is visible but the display remains blank. I have used a standard and an enhanced display.
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Sat, 26 Mar 2022 10:50:00 GMT.  Free memory 113792
D: Display type is NEXTION
D: SPIFFS is okay, space 1378241, used 33634
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 45 keys from NVS
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_i2s_bck set to 27
D: pin_i2s_lck set to 25
D: pin_i2s_din set to 26
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is LOW, probably no PULL-UP
D: GPIO26 is LOW, probably no PULL-UP
D: GPIO27 is LOW, probably no PULL-UP
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Start display
D: Init Nextion, pins 16, 17
D: Nextion command cls BLACK
D: Nextion command page 0
D: NEXTION output 'Starting......' to 1
D: NEXTION output 'Sat, 26 Mar 2022' to 2
D: NEXTION output 'By Ed Smallenburg' to 3
D: Create list with acceptable WiFi networks
D: Added Orange-486AE to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 7
D:  1 - Guest-Orange-82fa0        Signal: -54 dBm, Encryption WPA2_PSK,
D:  2 - Orange-82fa0              Signal: -56 dBm, Encryption WPA2_PSK,
D:  3 - Orange-1d6e6              Signal: -57 dBm, Encryption WPA2_PSK, 
D:  4 - Orange-486AE              Signal: -71 dBm, Encryption WPA2_PSK, Acceptable
D:  5 - telenet-2122642           Signal: -77 dBm, Encryption WPA2_PSK,
D:  6 - DIRECT-EB-HP ENVY 5000 series Signal: -87 dBm, Encryption WPA2_PSK,
D:  7 - telenet-05670             Signal: -87 dBm, Encryption WPA2_PSK, 
D: End of list
D: Connect to WiFi
D: NEXTION output 'Connect to WiFi' to 4
D: NEXTION output 'SSID = Orange-486AE' to 5
D: SSID = Orange-486AE
D: IP = 192.168.0.72
D: NEXTION output 'IP = 192.168.0.72' to 6
D: Nextion command page 1
D: Start server for commands
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 00:09:33
D: Radiofuncs cmd is 1
D: Connect to host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to MP3 host at 145.58.53.148 on port 80
D: send GET command
D: Nextion t0.txt="ESP32-Radio"
D: Nextion t1.txt=""
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: Headerline: icy-br:192
D: Headerline: icy-genre:Talk
D: Headerline: icy-metadata:1
D: Headerline: icy-name:NPO Radio1
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://www.radio1.nl
D: Headerline: Server: Icecast 2.4.0-kh15
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: Connection: Close
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, SOURCE, PUT, HEAD, STATS
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: Nextion t2.txt="NPO Radio1"
D: Starting I2S playtask..
D: Output to I2S, pins 27, 25 and 26
D: Playtask start song
D: helixInit called for audio/mpeg
D: Playtask stop song
D: Streamtitle found, 103 bytes
D: Nextion t1.txt="NPO Radio 1\rNooit Meer Slapen: Robin van den Maagdenberg (journalist schrijver) - VPRO"
D: Sync found at 0x0100
D: MP3Decode error -2
D: helixInit called for audio/mpeg
D: Sync found at 0x0080
D: Bitrate     is 192000
D: Samprate    is 48000
D: Channels    is 2
D: Bitpersamp  is 16
D: Outputsamps is 2304
D: MP3Decode error -6
D: helixInit called for audio/mpeg
D: Sync found at 0x0060
D: Bitrate     is 192000
D: Samprate    is 48000
D: Channels    is 2
D: Bitpersamp  is 16
D: Outputsamps is 2304

About GPIO18 == ??

Edzelf, 2022-03-30:
Have you uploaded the nextion program to the nextion?  Is the nextion baudrate correct?

Jefke1, 2022-03-30:
uploaded with SD card and TTL, baud rate 115200.
Set GPIO18 ==>> HIGH
Settings or OK but no result.
Radio plays well, volume control works fine but not visible in the serial monitor.

Edzelf, 2022-03-30:
I don't understand how you get this JSON info and what it means.

Jefke1, 2022-03-30:
it's very weird, when I open src main.cpp in pio I get 42 JSON errors. all related to Display or TFT.
[image]

Edzelf, 2022-03-30:
If it compiles without an error, the problem is in intelli-sense.  Sometimes a clean-up will help.  There is a button for this, looks like a garbage can.

wa8lbz, 2022-03-30:
I performed a clean up as you suggested and got the serial monitor printout during pio run -t upload:
[pio run -t upload 1.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8383999/pio.run.-t.upload.1.txt)

wa8lbz, 2022-03-30:
Could this be a problem? The last serial monitor printout for pio run -t upload shows Python to be version 3.1. The command Python --version show it to be 3.9.10.

Maybe there is a problem with preferences. When issuing the command pio run -t uploadfs, specifically what is being uploaded? defaultprefs.txt, ESP32_Radio_init.ino or something else. Both of these have been changed for my setup.

Edzelf, 2022-03-30:
It looks there is something wrong with COM7.  Usually exit Platformio will help, otherwise power cycle your PC.

Edzelf, 2022-03-30:
-t uploadfs uploads the files in the data directory to the SPIFFS.
-t upload uploads the sketch.

Jefke1, 2022-03-31:
I did a clean up. 
Everything uploaded, compiled without errors. 
Nextion doesn't work, only flickers on startup.
No signal on Pin 16 - 17.
[Serial.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8390051/Serial.txt)

Edzelf, 2022-03-31:
Check these lines in NEXTION.h:
```
// Various macro's to mimic the ST7735 version of display functions
#define dsp_setRotation()                                          // Use standard landscape format
#define dsp_print(a)        NEXTION_dsp_print(a)                   // Print a string
#define dsp_println(b)      NEXTION_dsp_println(b)                 // Print a string followed by newline 
```

Jefke1, 2022-03-31:
Identical to yours.

Jefke1, 2022-04-01:
Still a problem. Prev and Next do not work with the Nextion or in the web interface.
[Prev-Next.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8397439/Prev-Next.txt)

Edzelf, 2022-04-01:
I will look at that later.

Jefke1, 2022-04-01:
Everything adjusted. Result remains the same.

wa8lbz, 2022-04-01:
Changing the statement from nxtserial-> (115200) to 9600 in the nextion.cpp file brought my Nextion display to life. This fix was brought to my attention by Jefke1. Now the remaining problem is shown here:
[Power Up Serial Monitor.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8399883/Power.Up.Serial.Monitor.txt)

wa8lbz, 2022-04-02:
A test of Port 80 reveals this:
[image]

Edzelf, 2022-04-02:
What does it mean?  You try to connect to port 80 of your Windows machine?  This has nothing to do with your Nextion nor the ESP radio.