Edzelf/ESP32Radio-V2 #26: Wired Ethernet support?

dreamy1, 2022-03-31:
***not an issue*** :-)

Hello,

is it planned to add wired ethernet support for this great work?

Actually, there are some modified variants of your software with support for e.g. LAN8720. The only difference for the software part is that the SPI bus for the lcd and sd card has to be moved to the second hardware SPI bus (called HSPI) because the RMII pins are needed for the LAN chip. In consequence, some pins have to be switched but this should not be a problem because of the GPIO definition in the NVS text file. 

An ethernet connection is for most users with an automation systems the best choice. The RMII interface of the ESP32 support wired ethernet out-of-the-box.

Here is an example schematic showing the new pin definitions - VS1053, Ethernet via RMII and LCD are possible at the same time:

[image]

Edzelf, 2022-04-12:
I guess the radio will work well on a WT32-ETH-01 board.  The Ethernet interface will use som GPIO pins, so maybe some features must be eliminated.

dreamy1, 2022-04-12:
Hi Ed,

it would be great if there would be a possibility in your code to switch to HSPI pins for the SPI bus. Only renaming the pins does not work because the standard SPI bus is VSPI and im not good enough in software to get that solved (as what i learned, i ne SPI class has to be defined for VSPI).

With this modification there would be a maximum flexibility for other hardware parts. It would be great if you have a look on it.

Thanks,
Stefan

tsctrl, 2022-04-13:
i guess you could use any vspi or hspi pin by just changing the pin in config. if that is not working then you can just create adruino SPIClass SPI2(pin, pin, pin, speed)
and replace the existing SPI to SPI2 you just created before the SPI begin function.

is very simple 2 line of codes. give it a try.

tsctrl, 2022-04-13:
hi @dreamy1, can you point or share the url for the other variant or fork of the internet radio that was using the lan module?

dreamy1, 2022-04-14:
Hi @tsctrl,

the radio with LAN is here: https://sites.google.com/site/pcusbprojects/5-custom-projects/er-secure-esp32-internet-radio-with-wired-ethernet-connection

It is a modified version of the V1 of the radio from Ed. There is also a modification for using HSPI, but this doesn't compile (it seems that the SPI.h of the ESP32 was changed in the meantime by Arduino IDE). I tried to get it work with some code you posted above (new SPI Class), but with no success. 

Regards,
Stefan

dreamy1, 2022-04-15:
Did you have a look on it @tsctrl ?

tsctrl, 2022-04-15:
just dont use the arduino ide. use at least platform io. i doubt you couldnt change the spi pin. SPI class is predefined in the lib and you should able to use the VSPI pin with that predefine SPI class. in fact i just use VSPI pin using the SPI class and codes that was created by Ed and it works.

if that was not work either. the worst case is you have to modify the esp32 arduino library. but i dont think you have to do this.

tsctrl, 2022-04-15:
here is the example you can follow: https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino

dreamy1, 2022-04-15:
Thanks for the input - i tried that already but it seems that there is a bit more than a declaration at the entry of the code because the whole spi commands must contain a "hspi" and the whole declaration of the VS1053 is based on VSPI. 

And i am worse at software...but good in hardware :-)

tsctrl, 2022-04-15:
you need to have this declaration and have all your code to use this SPI Class to handle the SPI bus data transfer instead of Predefined SPI Class. And you have to point the initialization to that Class. 

https://github.com/schreibfaul1/ESP32-vs1053_ext/blob/0014147924b574987d11e730cfa50d315d9edf73/src/vs1053_ext.h#L153

for Ed implementation is more simpler and use the SPI Class declared in the library that use VSPI by default. 
https://github.com/espressif/arduino-esp32/blob/6cfe4613e4b4846e1ab08c7f78b7ea241f52c7da/libraries/SPI/src/SPI.h#L56

so, to change is by:
1. create new SPIClass spi2 with HSPI/VSPI.
2. change all SPI text to spi2. such as SPI.beginTransfer to spi2.beginTransfer

basically if you use the same code you share it should work.

dreamy1, 2022-04-16:
Thanks, i tried exactly what you suggested. But i got a lot of errors while compiling, all related to SPI. Maybe the place of the declaration was wrong.

Edzelf, 2022-04-16:
Next week I hope to get a WT32-ETH-01 board. And I will see if I  an run the radio software on it.

dreamy1, 2022-04-16:
Thanks a lot Ed, that would be great to get it to work!

Stefan

dreamy1, 2022-04-21:
Hi Ed,

did you get your WT32-ETH-01 board? 

Regards,
Stefan

Edzelf, 2022-04-21:
Yes, just one hour ago.

dreamy1, 2022-04-25:
Boah....Ed, you're absolutely great! I am really very impressed that you are able to present a solution so quick :-)

I will try it - just two questions: 
1) Is it possible to not use "pin_eth_power"? This is because in my application the activation of the LAN8720 is controlled externally.
2) Does SPI now still use hardware HSPI? Or is this still VSPI with routed signals by the GPIOMUX of the ESP32?

BTW: please let me know how i can estimade and support your great work by a paypal donation or you can get one of my hardware solution for this radio for free...i will take some pictures when it is running final :-)

Edzelf, 2022-04-25:
Yes, you may set the 3 pin definitions in the config page of the webinterface, or change the defaults in the source of main.cpp.
The software is still using the same SPI bus.  I did not test it yet, just connected a I2S module to a bare WT32-ETH01.
Surely I'm interested in your hardware solution.

dreamy1, 2022-04-25:
Just battling with compilation errors...i added the lib ESPAsyncWebServer but it seems to be not the correct one:

C:\Users\Stefan\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:33:10: fatal error: AsyncTCP.h: No such file or directory
 #include <AsyncTCP.h>
          ^~~~~~~~~~~~
compilation terminated.

dreamy1, 2022-04-25:
Problem solved, it was the AsyncTCP-master library missing.

Unfortunately, a lot of new compilation errors comes with ST7789 defined in config.h, e.g.:

C:\Users\Stefan\Desktop\ESP Radio\ESP32_Edzelf\ESP32Radio\ESP32Radio.ino: In function 'void tftset(uint16_t, const char*)':
ESP32Radio:848: error: 'TFTSECS' was not declared in this scope
   if ( inx < TFTSECS )                                  // Segment available on display
              ^~~~~~~
ESP32Radio:852: error: 'tftdata' was not declared in this scope
       tftdata[inx].str = String ( str ) ;               // Yes, set string

...and much more regarding the tft declarations.

Edzelf, 2022-04-25:
Show your config.h.

dreamy1, 2022-04-25:
This is my config.h, the adafruit libs are installed too:

//***************************************************************************************************
// config.h                                                                                         *
//***************************************************************************************************
//
//#define NAME "ESP32-Radio"                              // Define name of the radio, also AP SSID
                                                          // Default is "ESP32-Radio"

//#define SDCARD                                          // For SD card support (reading MP3-files)
#define ETHENET
//#define FIXEDWIFI "MYWIFI/password"                       // Add a fixed SSID to the list

// Define (just one) type of MP3/AAC decoder
#define DEC_VS1053                                        // Hardware decoder for MP3, AAC, OGG
//#define DEC_VS1003                                      // Hardware decoder for MP3
//#define DEC_HELIX                                       // Software decoder for MP3, AAC. I2S output
//#define DEC_HELIX_INT                                   // Software decoder for MP3, AAC. DAC output

// Define (just one) type of display.  See documentation.
//#define BLUETFT                                           // Works also for RED TFT 128x160
#define ST7789                                          // 240x240 TFT
//#define OLED1306                                        // 64x128 I2C OLED SSD1306
//#define OLED1309                                        // 64x128 I2C OLED SSD1309
//#define OLED1106                                        // 64x128 I2C OLED SH1106
//#define DUMMYTFT                                        // Dummy display
//#define LCD1602I2C                                      // LCD 1602 display with I2C backpack
//#define LCD2004I2C                                      // LCD 2004 display with I2C backpack
//#define ILI9341                                         // ILI9341 240*320
//#define NEXTION                                         // Nextion display. Uses UART 2 (pin 16 and 17)
//

// Define ZIPPYB5 if a ZIPPY B5 Side Switch is used instead of a rotary switch
///#define ZIPPYB5

//bat0   = 2178   # ADC for 0% battery capacity left
//bat100 = 3256   # ACD for 100% battery capacity
#
clk_dst = 1
clk_offset = -1
clk_server = pool.ntp.org
#
//gpio_00 = uppreset = 1
//gpio_12 = upvolume = 2
//gpio_13 = downvolume = 2
#
//ir_40BF = upvolume = 2
//ir_C03F = downvolume = 2
#
//mqttbroker = none
//mqttpasswd = none
//mqttport = 1883
//mqttuser = carol
#
//pin_enc_clk = 25     # GPIO rotary encoder CLK
//pin_enc_dt = 26      # GPIO rotary encoder DT
//pin_enc_sw = 27      # GPIO rotary encoder SW
pin_spi_sck = 14
pin_spi_miso = 12
pin_spi_mosi = 13
pin_ir = 39          # GPIO IR receiver VS1838B
pin_tft_bl = 2 
pin_tft_cs = 4      # GPIO TFT CS
pin_tft_dc = 5       # GPIO TFT DC
pin_vs_cs = 15        # GPIO VS1053 CS
pin_vs_dcs = 32      # GPIO VS1053 DCS
pin_vs_dreq = 33      # GPIO VS1053 DREQ
#
preset = 8
preset_000 = 109.206.96.34:8100                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
preset_001 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida
preset_002 = streamer.radio.co/sa7bf61271/listen      #  2 - JB's Rock N Roll
preset_003 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_004 = stream.laut.fm/kinderlieder-123          #  4 - Kinderlieder
preset_005 = icecast.omroep.nl:80/radio1-bb-mp3       #  5 - Radio 1, NL
preset_006 = icecast.omroep.nl:80/radio5-bb-mp3       #  6 - Radio 5, NL
preset_007 = icecast.omroep.nl:80/radio2-bb-mp3       #  7 - Radio 2, NL
preset_008 = www.antenne.de/webradio/antenne.m3u
preset_009 = webstream.schlagerparadies.de:80/schlagerparadies128k.mp3
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 5
#
wifi_00 = ADSL-11_plus/xxxxxx
wifi_01 = NETGEAR-11/xxxxxx
wifi_02 = ADSL-11/xxxxxx

Edzelf, 2022-04-25:
Define ETHERNET, not ETHENET.

dreamy1, 2022-04-25:
Already corrected, the compilation errors are the same - all with the TFT functions.

dreamy1, 2022-04-25:
This is my workspace:
[image]

Edzelf, 2022-04-25:
Complicated, because it is different than in the Github repositry:
[image]
And what do you mean by installing Adafruit libraries?  All librararies are installed by teh platformio.ini file.

dreamy1, 2022-04-25:
I don't use platformio, i still use the regular arduino IDE since years now. I checked every single library needed, they are all installed correctly. 

I just tried to switch to an ILI9341, but the compilation errors are the same. It has something to do with the use of the graphic libs.

Edzelf, 2022-04-25:
Yes, I see now.  Be sure to use one of my display libraries.  There  'TFTSECS' is defined in the header file of the configured display.

dreamy1, 2022-04-25:
Yes, of course i use your ST7789 lib. It seems that this lib is not included correctly from the main sketch.

dreamy1, 2022-04-25:
All .h and .cpp seems to be loaded correctly:
[image]

tsctrl, 2022-04-26:
hi @dreamy1,

i have question for you custom hardware design,
do you have oscillator en pin for LAN8720 connected to esp32 gpio pin?

dreamy1, 2022-04-26:
I handle that with a GPIO pin of the VS1053.

Edzelf, 2022-04-26:
I have updated the procedure for compiling with the Arduino IDE in the document.  There is also a change in main.cpp that effects the position of the config.h file.  I tested the procedure and it compiles without problems.

dreamy1, 2022-04-26:
Thank you very much Ed!

I still got a lot of errors while compiling and it seems most are related to some "#" in my config.h. After removing them, only the following errors occure:

**_In file included from C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino:92:
config.h:30: error: 'pin_spi_sck' does not name a type
 pin_spi_sck = 14
 ^~~~~~~~~~~
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino: In function 'void nvsopen()':
ESP32Radio:667: error: 'NVS_READWRITE' was not declared in this scope
     nvserr = nvs_open ( NAME, NVS_READWRITE, &nvshandle ) ;  // No, open nvs
                               ^~~~~~~~~~~~~
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino:667:31: note: suggested alternative: 'S_IWRITE'
     nvserr = nvs_open ( NAME, NVS_READWRITE, &nvshandle ) ;  // No, open nvs
                               ^~~~~~~~~~~~~
                               S_IWRITE
ESP32Radio:667: error: 'nvs_open' was not declared in this scope
     nvserr = nvs_open ( NAME, NVS_READWRITE, &nvshandle ) ;  // No, open nvs
              ^~~~~~~~
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino:667:14: note: suggested alternative: 'nvsopen'
     nvserr = nvs_open ( NAME, NVS_READWRITE, &nvshandle ) ;  // No, open nvs
              ^~~~~~~~
              nvsopen
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino: In function 'esp_err_t nvsclear()':
ESP32Radio:684: error: 'nvs_erase_all' was not declared in this scope
   return nvs_erase_all ( nvshandle ) ;                // Clear all keys
          ^~~~~~~~~~~~~
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino: In function 'String nvsgetstr(const char*)':
ESP32Radio:700: error: 'nvs_get_str' was not declared in this scope
   nvserr = nvs_get_str ( nvshandle, key, nvs_buf, &len ) ;
            ^~~~~~~~~~~
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino:700:12: note: suggested alternative: 'nvsgetstr'
   nvserr = nvs_get_str ( nvshandle, key, nvs_buf, &len ) ;
            ^~~~~~~~~~~
            nvsgetstr
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino: In function 'esp_err_t nvssetstr(const char*, String)':
ESP32Radio:736: error: 'nvs_set_str' was not declared in this scope
     nvserr = nvs_set_str ( nvshandle, key, val.c_str() ) ; // Store key and value
              ^~~~~~~~~~~
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino:736:14: note: suggested alternative: 'nvssetstr'
     nvserr = nvs_set_str ( nvshandle, key, val.c_str() ) ; // Store key and value
              ^~~~~~~~~~~
              nvssetstr
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino: In function 'void nvschkey(const char*, const char*)':
ESP32Radio:758: error: 'nvs_erase_key' was not declared in this scope
     nvs_erase_key ( nvshandle, oldk ) ;                    // Remove key
     ^~~~~~~~~~~~~
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino: In function 'bool nvssearch(const char*)':
ESP32Radio:774: error: 'nvs_get_str' was not declared in this scope
   nvserr = nvs_get_str ( nvshandle, key, NULL, &len ) ; // Get length of contents
            ^~~~~~~~~~~
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino:774:12: note: suggested alternative: 'nvsgetstr'
   nvserr = nvs_get_str ( nvshandle, key, NULL, &len ) ; // Get length of contents
            ^~~~~~~~~~~
            nvsgetstr
C:\Users\Stefan\Documents\Arduino\SJS\ESP32Radio\ESP32Radio.ino: In function 'void writeprefs(AsyncWebServerRequest*)':
ESP32Radio:2904: error: 'nvs_commit' was not declared in this scope
   nvs_commit( nvshandle ) ;
   ^~~~~~~~~~_**

I have no idea what is the problem here...

dreamy1, 2022-04-26:
Would you please test if you can compile with my config.h?

//***************************************************************************************************
// config.h                                                                                         *
//***************************************************************************************************
//
//#define NAME "ESP32-Radio"                              // Define name of the radio, also AP SSID
                                                          // Default is "ESP32-Radio"

//#define SDCARD                                          // For SD card support (reading MP3-files)

//#define FIXEDWIFI "MYWIFI/password"                       // Add a fixed SSID to the list
#define ETHERNET
// Define (just one) type of MP3/AAC decoder
#define DEC_VS1053                                        // Hardware decoder for MP3, AAC, OGG
//#define DEC_VS1003                                      // Hardware decoder for MP3
//#define DEC_HELIX                                       // Software decoder for MP3, AAC. I2S output
//#define DEC_HELIX_INT                                   // Software decoder for MP3, AAC. DAC output

// Define (just one) type of display.  See documentation.
//define BLUETFT                                           // Works also for RED TFT 128x160
#define ST7789                                          // 240x240 TFT
//#define OLED1306                                        // 64x128 I2C OLED SSD1306
//#define OLED1309                                        // 64x128 I2C OLED SSD1309
//#define OLED1106                                        // 64x128 I2C OLED SH1106
//#define DUMMYTFT                                        // Dummy display
//#define LCD1602I2C                                      // LCD 1602 display with I2C backpack
//#define LCD2004I2C                                      // LCD 2004 display with I2C backpack
//#define ILI9341                                         // ILI9341 240*320
//#define NEXTION                                         // Nextion display. Uses UART 2 (pin 16 and 17)
//
pin_spi_sck = 14
pin_spi_miso = 12
pin_spi_mosi = 13
pin_ir = 39
pin_tft_bl = 2 
pin_tft_cs = 4
pin_tft_dc = 5
pin_vs_cs = 15
pin_vs_dcs = 32
pin_vs_dreq = 33

toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
volume = 5

clk_dst = 1
clk_offset = -1
clk_server = pool.ntp.org

Edzelf, 2022-04-26:
I wiil.  Are you aware of the second half of this file?  Only "#define" lines are expected.

Edzelf, 2022-04-26:
Compilation is okay.  Please check line 92 of the main program.  It should read:
`#include "config.h"                                       // Specify display type, decoder type`

dreamy1, 2022-04-26:
Oha :-) Now it compiles without any errors. Big sorry, that was my mistake with misunderstanding config.h and the config page in NVS :-(

So my last question for today:
Without a specific pin definition vor various signals *before* uploading the code into the hardware i expect that my hardware will not start correctly. I think this will apply to other hardware too because the VS1053 can't start without defining the corresponding pins before.

For example, the following pin definitions are "fixed" in my hardware:
pin_spi_sck = 14
pin_spi_miso = 12
pin_spi_mosi = 13
pin_ir = 39
pin_tft_bl = 2 
pin_tft_cs = 4
pin_tft_dc = 5
pin_vs_cs = 15
pin_vs_dcs = 32
pin_vs_dreq = 33 

I think the best solution in my case would be to change the lines 1863ff - so instead of "-1" i would place the pin used in my hardware:

  **struct iosetting klist[] = {                            // List of I/O related keys
    { "pin_ir",        &ini_block.ir_pin,           -1 },
    { "pin_enc_clk",   &ini_block.enc_clk_pin,      -1 }, // Rotary encoder CLK
    { "pin_enc_dt",    &ini_block.enc_dt_pin,       -1 }, // Rotary encoder DT
    { "pin_enc_up",    &ini_block.enc_up_pin,       -1 }, // ZIPPY B5 side switch up
    { "pin_enc_dwn",   &ini_block.enc_dwn_pin,      -1 }, // ZIPPY B5 side switch down
    { "pin_enc_sw",    &ini_block.enc_sw_pin,       -1 },
    { "pin_tft_cs",    &ini_block.tft_cs_pin,       -1 }, // Display SPI version
    { "pin_tft_dc",    &ini_block.tft_dc_pin,       -1 }, // Display SPI version
    { "pin_tft_scl",   &ini_block.tft_scl_pin,      -1 }, // Display I2C version
    { "pin_tft_sda",   &ini_block.tft_sda_pin,      -1 }, // Display I2C version
    { "pin_tft_bl",    &ini_block.tft_bl_pin,       -1 }, // Display backlight
    { "pin_tft_blx",   &ini_block.tft_blx_pin,      -1 }, // Display backlight (inversed logic)
    { "pin_sd_cs",     &ini_block.sd_cs_pin,        -1 }, // SD card select
    { "pin_vs_cs",     &ini_block.vs_cs_pin,        -1 }, // VS1053 pins
    { "pin_vs_dcs",    &ini_block.vs_dcs_pin,       -1 },
    { "pin_vs_dreq",   &ini_block.vs_dreq_pin,      -1 },
    { "pin_shutdown",  &ini_block.shutdown_pin,     -1 }, // Amplifier shut-down pin
    { "pin_shutdownx", &ini_block.shutdownx_pin,    -1 }, // Amplifier shut-down pin (inversed logic)
    { "pin_i2s_bck",   &ini_block.i2s_bck_pin,      -1 }, // I2S interface pins
    { "pin_i2s_lck",   &ini_block.i2s_lck_pin,      -1 },
    { "pin_i2s_din",   &ini_block.i2s_din_pin,      -1 },
  #ifdef ETHERNET
    { "pin_spi_sck",   &ini_block.spi_sck_pin,      -1 },
    { "pin_spi_miso",  &ini_block.spi_miso_pin,     -1 },
    { "pin_spi_mosi",  &ini_block.spi_mosi_pin,     -1 },
    { "pin_eth_mdc",   &ini_block.eth_mdc_pin,      23 },
    { "pin_eth_mdio",  &ini_block.eth_mdio_pin,     18 },
    { "pin_eth_power", &ini_block.eth_power_pin,    16 },
  #else
    { "pin_spi_sck",   &ini_block.spi_sck_pin,      18 },
    { "pin_spi_miso",  &ini_block.spi_miso_pin,     19 },
    { "pin_spi_mosi",  &ini_block.spi_mosi_pin,     23 },** 

Would that be correct?

Edzelf, 2022-04-26:
I guess it's best in this case to change the defaults in the source code, at least for the ethernet pins.  As soon as you have the webinterface running, you can set the ethernet- and other pins in the config page of the webinterface.
Then the defaults have no effect anymore.

dreamy1, 2022-04-26:
So the place above in the main.cpp would be right or is there another position what has to be changed?

I think for optional hardware it would be suffcient to define that later by the webinterface, but in my case TFT, VS1053 und Ethernet is always fixed. And these pins should have defined signal levels before the first starting of the software.

How is the software reacting at the moment if e.g. VS1053 is not found by the SPI bus because the pins are not defined correctly before updating them via the webinterface? Maybe that would be a general improvement defining that in the main.cpp?

Edzelf, 2022-04-26:
The idea is to have as little as possible pin definitions in the source code.  If you do not have WiFi, it will be very difficult to "bootstrap" the device.  You may however use the sketch in the tools directory to fill the NVS at forehand.
Once the NVS it set, the drivers for mp3 decoding, display, rotary encoder, touch and IR will read the pin settings from NVS.  You can play around with the NVS settings without changing the software.

tsctrl, 2022-04-26:
Hi @Edzelf ,

i have tested the latest codes with standalone LAN8720 ETH board with Wifi AP and Station enabled and it works without any issue.

I (27) boot: ESP-IDF v4.3.2 2nd stage bootloader
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (74) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (82) boot:  2 app0             OTA app          00 10 00010000 00140000
I (89) boot:  3 app1             OTA app          00 11 00150000 00140000
I (97) boot:  4 int_storage      Unknown data     01 82 00290000 00170000
I (104) boot: End of partition table
I (631) cpu_start: ESP-IDF:          v4.3.2
I (703) esp_core_dump_uart: Init core dump to UART
I (706) cpu_start: Starting scheduler on PRO CPU.
I (305) esp_apptrace: Initialized TRAX on CPU1
I (5) cpu_start: Starting scheduler on APP CPU.

D: Starting ESP32-radio running on CPU 0 at 240 MHz.
D: Version Fri, 15 Apr 2022 11:30:00 GMT.  Free memory 108544
D: Display type is BLUETFT
D: FILESYSTEM is okay, space 1507328, used 77824
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'int_stor' at offset 0x290000 with size  1507328
D: Read 61 keys from NVS
D: Create list with acceptable WiFi networks
D: Added MYWIFI to list of networks
D: Added SSID2 to list of networks
I (2942) wifi:wifi driver task: 3ffcb270, prio:23, stack:6656, core=0
I (2955) wifi:wifi firmware version: eb52264
I (2956) wifi:wifi certification version: v7.0
I (2992) wifi_init: rx ba win: 6
I (3030) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (3129) wifi:mode : sta (3c:61:05:4c:3a:b8) + softAP (3c:61:05:4c:3a:b9)

D: Scan Networks
D: Scan completed
D: Number of available networks: 1
D:  1 - redmi-9a                  Signal: -89 dBm, Encryption WPA2_PSK, 
D: End of list
D: updateNr 0 <= 0 to 0, relative is 0
D: updateNr result is 0
D: nextPreset is 0, host 109.206.96.34:8100                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
D: Slow SPI, Testing VS1053 read/write registers...
D: This is not a VS1053, but a VS1003 instead!
I (7551) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
D: Fast SPI, Testing VS1053 read/write registers again...
D: This is not a VS1053, but a VS1003 instead!
D: Connect to WiFi
I (9048) wifi:station: 48:ee:0c:24:5d:ef leave, AID = 1, bss_flags is 655395, bss:0x3ffd2220
I (9049) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:6
I (9114) wifi:Total power save buffer number: 16
I (21348) wifi:ap channel adjust o:6,1 n:1,1
W (21882) wifi:<ba-add>idx:0 (ifx:0, 34:e8:94:3e:0f:db), tid:0, ssn:0, winSize:64
D: WiFi AP, Trying to setup AP with name ESP32Radio and password ESP32Radio.
I (31871) wifi:Total power save buffer number: 16
D: IP = 1.0.0.0
D: ETH pins 17, 23 and 18
I (31909) esp_eth.netif.glue: 3c:61:05:4c:3a:bb
D: ETH event 17
D: ETH Stopped
D: ETH cable disconnected
I (34438) esp_netif_handlers: eth ip: 192.168.0.107, mask: 255.255.255.0, gw: 192.168.0.1
D: IP = 1.0.0.0
D: Start server for commands
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 02:37:45
D: Radiofuncs cmd is 1
D: Connect to host 109.206.96.34:8100
D: Connect to 109.206.96.34 on port 8100, extension /
D: send GET command
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -   808
D:   1 -    33
D:   2 -     0
D:   3 -  6000
D:   4 -     0
D:   5 -  AC45
D:   6 -  5084
D:   7 -  1E06
D:   8 -     0
D:   9 -     0
D:   A -     0
D:   B -  F8F8
D:   C -     0
D:   D -     0
D:   E -     0
D:   F -     0
D: Switch to HEADER
D: Headerline: icy-name:NAXI LOVE RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) - 128k
D: Headerline: icy-genre:Love and Romance
D: Headerline: icy-url:http://www.naxi.rs
D: Headerline: content-type:audio/mpeg
D: Headerline: icy-pub:0
D: Headerline: icy-metaint:8192
D: Headerline: icy-br:128
D: Switch to DATA, bitrate is 128, metaint is 8192

The issue i have is i not able to use LAN8720 as source 50Mhz clock and have to use esp to feed in GPIO0. 
And it is the only approach work for my ETH module.

Thanks for the ETH addition!

dreamy1, 2022-04-26:
Hi Ed,

for getting the thing about the fixed pins in the main.cpp round...besides the changes in the pin list above, what does the following in your code?

progpin_struct   progpin[] =                             // Input pins and programmed function
{
  {  0, false, false,  "", false },
  //{  1, true,  false,  "", false },                    // Reserved for TX Serial output
  {  2, false, false,  "", false },
  //{  3, true,  false,  "", false },                    // Reserved for RX Serial input
  {  4, false, false,  "", false },
  {  5, false, false,  "", false },
  //{  6, true,  false,  "", false },                    // Reserved for FLASH SCK
  //{  7, true,  false,  "", false },                    // Reserved for FLASH D0
  //{  8, true,  false,  "", false },                    // Reserved for FLASH D1
  //{  9, true,  false,  "", false },                    // Reserved for FLASH D2
  //{ 10, true,  false,  "", false },                    // Reserved for FLASH D3
  //{ 11, true,  false,  "", false },                    // Reserved for FLASH CMD
  { 12, false, false,  "", false },
  { 13, false, false,  "", false },
  { 14, false, false,  "", false },
  { 15, false, false,  "", false },
  { 16, false, false,  "", false },                      // May be UART 2 RX for Nextion
  { 17, false, false,  "", false },                      // May be UART 2 TX for Nextion
  { 18, false, false,  "", false },                      // Default for SPI CLK
  { 19, false, false,  "", false },                      // Default for SPI MISO
  //{ 20, true,  false,  "", false },                    // Not exposed on DEV board
  { 21, false, false,  "", false },                      // Also Wire SDA
  { 22, false, false,  "", false },                      // Also Wire SCL
  { 23, false, false,  "", false },                      // Default for SPI MOSI
  //{ 24, true,  false,  "", false },                    // Not exposed on DEV board
  { 25, false, false,  "", false },                      // DAC output / I2S output
  { 26, false, false,  "", false },                      // DAC output / I2S output
  { 27, false, false,  "", false },                      // I2S output
  //{ 28, true,  false,  "", false },                    // Not exposed on DEV board
  //{ 29, true,  false,  "", false },                    // Not exposed on DEV board
  //{ 30, true,  false,  "", false },                    // Not exposed on DEV board
  //{ 31, true,  false,  "", false },                    // Not exposed on DEV board
  { 32, false, false,  "", false },
  { 33, false, false,  "", false },
  { 34, false, false,  "", false },                      // Note, no internal pull-up
  { 35, false, false,  "", false },                      // Note, no internal pull-up
  //{ 36, true,  false,  "", false },                    // Reserved for ADC battery level
  { 39, false,  false,  "", false },                     // Note, no internal pull-up
  { -1, false, false,  "", false }                       // End of list
} ;

Is this to prohibit a new pin declaration in the webinterface? So if i want to get some pins fixed in the main.cpp and not user changeable in the webinterface, i would:
- change a specific pin in the struct iosetting klist[] from -1 to the GPIO name
- change the GPIO in the progpin_struct   progpin[]  to "true" (first of the 3 boolean entries)

Is that correct?

BTW: Thanks tsctrl for the feedback, sounds great :-)

dreamy1, 2022-04-26:
@tsctrl : you can apply a 4k7-10k pulldown resistor from the enable pin of the LAN8720 oscillator to ground and wire this enable pin to a free GPIO of the ESP32. This GPIO you can use as PHY_power_pin, so while booting the clock signal of the oscillator is not active and after booting and activating ethernet the GPIO is set to high and the clock signal is then applied to the GPIO0 of the ESP32.

This is the actual workaround for the LAN8720 modules.

Edzelf, 2022-04-26:
The propping table is used to hold information about the programmable pins.

tsctrl, 2022-04-26:
yes, all are active @dreamy1, just i not able to use the external oscillator. try that setup to disable to oscillator but id did not communicate to esp and driver didnt load during boots.

wondering why use LAN8720 instead of W5500 lan module @dreamy1 , @Edzelf?

W5500 is spi based and use lesser pin

dreamy1, 2022-04-27:
@tsctrl : maybe you need a second resistor as pulldown for the GPIO0 of the ESP32: https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

With this modification the LAN8720 should work without problems.

tsctrl, 2022-04-28:
thanks @dreamy1, i follow that blog during my test. have pull up hook up on gpio0 and pull down on gpio17.
Same as what @Edzelf does in the lib to use GPIO0OUT as esp generate the oscillator. maybe @Edzelf also have the same issue by using the W32 ETH01 module. 

look like esp does not detect the freq during boot. so i am too lazy to hook osciloscope and debug further and just use esp oscillation mode as what had work.

esp32 is very strong chip. 240mhz it is quite fast. i guess spi w5500 could work well. i am still waiting my w5500 module to test.

@Edzelf can consider to integrate with w5500 module too? would be great as gpio is very limited. we lost 6 pins by using LAN8720 and only 2 by using W5500.

Edzelf, 2022-04-28:
Further testing:
I tested the WT32-ETH01 module with the VS1053 (SPI) without problems.
Pins used:
```
pin_vs_cs = 2        # GPIO VS1053 CS
pin_vs_dcs = 12      # GPIO VS1053 DCS
pin_vs_dreq = 4      # GPIO VS1053 DREQ
pin_spi_sck = 5      # GPIO SPI SCK
pin_spi_miso = 33    # GPIO SPI MISO
pin_spi_mosi = 17    # GPIO SPI MOSI
```

Next step: SPI display.

Edzelf, 2022-04-28:
Further testing:
I tested he WT32-ETH01 module with VS1053 and SPI display (BLUETFT) without problems.  Station at 320 kbps.
Pins used:
```
pin_vs_cs = 2        # GPIO VS1053 CS
pin_vs_dcs = 12      # GPIO VS1053 DCS
pin_vs_dreq = 4      # GPIO VS1053 DREQ
#
pin_spi_sck = 5      # GPIO SPI SCK
pin_spi_miso = 33    # GPIO SPI MISO
pin_spi_mosi = 17    # GPIO SPI MOSI
#
pin_tft_cs = 35      # GPIO BLUETFT CS
pin_tft_dc = 14      # GPIO BLUETFT A0
```
[image]

dreamy1, 2022-04-28:
Thanks Ed for the update!

It seems that the WT32-ETH01 has other GPIO definitions than the "standard" ESP32? For example, the GPIO35 is input only, so that would not work as chip_select for the TFT.

One thing what is not clear for me: do you use real hardware spi of the second spi bus (HSPI) or does your code still use the first spi bus (VSPI) which is routed over the GPIOMUX to the non-standard-pins? That will cost extra performance - but maybe it is not a problem.

tsctrl, 2022-04-28:
is there any gpio left for ir and rotary switch? i hope we dont have to integrate an i2c io extender. my experience  integrating those is a nightmare. it is slow and not suitable for rotary encoder, use lots of codes to deal with i2c bus and need it own process to run. @Edzelf please consider to add w5500 instead.

tsctrl, 2022-04-28:
hi @dreamy1 i not sure if it is gpio gmux involve. but the codes use existing vspi bus with a non standard pin definition. i not sure if using existing bus with hspi pin definition will use secondary bus automatically or will use gmux. this need to be address or asked to expressif if you will. else current implementation need to follow multiple bus implementation as example i had mention previously. 

i have use the codes that was done by @Edzelf without any changes on the bus implementation. just change the pin definition to hspi pins and it work without any lags on the spi bus. run with lan vs decoder.

Edzelf, 2022-04-28:
@dreamy1: You are right about pin35, it is input only.  But as you can see in the picture, the display functions normally.  You may connect CS of the display to ground and it will still work. You may also use pin 15, it is still unused
@tsctrl: there are pins for the rotary encoder (al inputs): 35, 36, 39 (ypu ave to supply PULL-UPs I think).
I cammot test the W5500, as I do not have one, but I quess it will work as well.

tsctrl, 2022-04-29:
hi @Edzelf 

i have use gpio as below for my setup but the rotary encoder only response to volume up and down rotation.
no 2 or single press mute was detected.

pin_enc_clk = 34                                     # GPIO Pin number for rotary encoder CLK
pin_enc_dt = 39                                     # GPIO Pin number for rotary encoder DT
pin_enc_sw = 36                                     # GPIO Pin number for rotary encoder SW

i have added 3 pull up 10k to all the pin without any efect.

Edzelf, 2022-04-29:
Pin 36 is reserved for battery level, see the progpin table.  Better use pin 35.

tsctrl, 2022-04-29:
Hi @Edzelf ,

could sent me your address to tsc7rl@gmail.com? will try to sent you the ethernet modules to your address.
will try if possible to sent this from china.

thanks!

tsctrl, 2022-05-24:
hi @Edzelf , any choppy playback on 320kbps stream? any valid channel with 320kbps to try? 

look like ethernet are slower than wifi. i am only getting around 1400 bytes length per client read, do you have this issue once before migrating to AsyncTCP?

Edzelf, 2022-05-24:
With the search function in yhe webinterface you can find 320 kbps streams.
AsyncTCP has a 5 kB buffer if I remember well.