Edzelf/ESP32Radio-V2 #27: Preset Stations

tini1617, 2022-04-01:
Hello,
I used v1 of this wonderful software and had problems switching from one preset to another by using previous/next in web interface or with the rotary switch: the current station does not change.

Now I tried v2: Similiar problem. The switching works only between preset 6 and 7 and some other stations of the defaultprefs. And switching to a other new presets/station/stream (which worked in v1!) leads to a crash and reboot.

Are there any new prerequisits for the urls of the stations? Or any other ideas?
Best regards.

Edzelf, 2022-04-01:
I will test it later.  Meanwhile you may try the Search function.

tini1617, 2022-04-01:
Thanks for replying!
I did some testing. Here is some additional information. For example:

From "Search Function": For example....
ANTENNE BAYERN (64kbps AAC)   0k       ok
Bayern 1 Franken            MP3        128k    **no switching to, remaing at previous station**
Bayern 1 Oberfranken     MP3        128k    **no switching to, remaing at previous station**
Antenne Sauerland         MP3        128k    ok
Antenne Kaiserslautern  AAC+      128k    **Crash and reboot**

Stream ok with V1/ not ok mit V2:
streams.br.de/br24live_2.m3u              **Crash and reboot** 
streams.br.de/bayern3_2.m3u              **Crash and reboot**
streams.br.de/bayern1obb_2.m3u       **Crash and reboot**
(These stations are my favorites....)

Thank you again for your activities and best regards.

tini1617, 2022-04-04:
New experiences:
A general problem  in my environment seems to be:  freertos (queue.h and freertos/task.h).
I had to get rid of the code line "xQueueReceive(sdqueue, &specchunk, 0)" in QUEUETOPT --> no further crashes at preset change.
Tried to use the latest ESP32 package 2.0.2 --> GURU Meditation ! No success.
Program running only with package 1.0.6.

The above mentioned streams lead to SPI semaphores -  and/or  wdt - issues und reboots.

I`m happy that for example
"dispatcher.rndfnk.com/br/br1/obb/aac/low" works.
Best regards.

timoluebeck, 2022-04-08:
Same problem here, got GURU Meditation. maybe a wrong parameter?
Also,when i hit the "station+" button in the webinterface once, i come to preset 1, if i do hit 2 times, it's preset 2, if i hammer 10 times on it, its preset 10. before the change, everything worked normal, just incrementing the number of the preset,
Hopefully this helps a bit finding the bug. 
Thanks for the great work!

Edzelf, 2022-04-12:
The NEXT problem, that causes crashes is solved in the 12 april version.

tini1617, 2022-04-13:
Hi Ed,
thank you very much for (nearly) solving the NEXT issue.  Great job!!! 
Rotary switch stable (up/down) - important for me!  (NEXT in webinterface still with problems). 
Still 2 problems remaining.:
- not all stations can be played -  for example: streams.br.de/bayern3_2.m3u  - Crash and reboot (working in V1).
- after turning the rotary switch: After confirmation (single click) stationname is displayed shortly than it is erased (Information 
   lines are continously displayed). When choosing station with (web-)presets: Stationname is stable displayed.
   We tried to find the bug in the software - no chance........

Again thank you very much for your activities - I appreciate this project very much!
Best regards.

Edzelf, 2022-04-13:
streams.br.de/bayern3_2.m3u is not a station, but a playlist.
It seems that the current version is not able to handle a playlist correctly.
Use "dispatcher.rndfnk.com/br/br3/live/mp3/mid" instead.  This is the first entry in the playlist.

The NEXT function is corrected in the 13 april version.

Edzelf, 2022-04-13:
Playlist function is corrected now.
In this case the first entry in the playlist points to a second site which redirects to a third site....
The German way of doing things?

timoluebeck, 2022-04-13:
Thanks Ed, switching forward works a bit better now,but after reaching the
last preset Station and trying to switch one over to the first one, the
Radio stops playing and counts still up the preset list. Also switching
from preset 0 back results in preset "-1" which is illegal. Die not have
the time yet to dig further,but hopefully the discription helps a bit

Regards
Timo

tini1617, 2022-04-15:
Hi Ed,
thank you very much for the optimization of the code. The rotary switch and the german playlists work perfectly. Great!
The only point for further optimization is the missing/disappearing playlistname after changing it with rotary switch as I mentioned allready.
Again thank you very much - I continue the hardware integration in an old  tube radio......
Best regards
Lutz

Edzelf, 2022-04-15:
Fixed.

tini1617, 2022-04-16:
Hi Ed,
perfect! Thank you very(!) much for these improvements. It works fine!
Best regards.
Lutz

closed 2022-04-16 by tini1617

tini1617, 2022-10-11:
Thanks for replying!
I did some testing. Here is some additional information. For example:

From "Search Function": For example....
ANTENNE BAYERN (64kbps AAC) 0k ok
Bayern 1 Franken MP3 128k  no switching to, remaing at previous station
Bayern 1 Oberfranken MP3 128k  no switching to, remaing at previous station
Antenne Sauerland MP3 128k ok
Antenne Kaiserslautern AAC+ 128k  Crash and reboot

Stream ok with V1/ not ok mit V2:

streams.br.de/br24live_2.m3u  Crash and reboot
streams.br.de/bayern3_2.m3u  Crash and reboot
streams.br.de/bayern1obb_2.m3u  Crash and reboot
(These stations are my favorites....)

Thank you again for your activities and best regards.

Dr.-Ing. Lutz Kühne

Edzelf, 2022-10-12:
I have tried your 3 favorite stations: no problem found....

tini1617, 2022-10-15:
Thank you, Ed for this information. I will test the latest software version in the next days.
Nevertheless I`m happy with my current version of software. I use the radio very often.
Best regards.