Edzelf/ESP32Radio-V2 #28: Which partition scheme should be used for a 16MB ESP32 board?

dreamy1, 2022-04-03:
Hello,

the arduino IDE allows several partition schemes to choose, for a ESP32 devkit board following options e.g.:

[image]

Which option is the right one to choose?

Thanks,
Stefan

Edzelf, 2022-04-03:
The top one is okay.  You need SPIFFS, and OTA can be handy, but is not mandatory.

dreamy1, 2022-04-03:
Hi Ed,

thank you very much for the quick response. So the top one works including your web update bot with no reserved space for the web update of new firmware?

Edzelf, 2022-04-03:
If you want OTA,  use "minimal SPIFFS".

dreamy1, 2022-04-03:
Thanks again Ed, i will try it. 

Stefan