Edzelf/ESP32Radio-V2 #29: Fails Setting Up Access Point

wa8lbz, 2022-04-04:
I purchased a new ESP32 DOIT DEVKIT. Used the pio run -t upload and pio run -t uploadfs commands to upload the current sketch and preferences. Both were successful. Nothing was connected to the ESP32 except power, a capacitor from ESP32 EN pin to ground and USB cable to computer COM7. It's not working. See attached file...
[Power Up Serial Monitor 2.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8413753/Power.Up.Serial.Monitor.2.txt)

Edzelf, 2022-04-05:
Do you see the ESRadio access point?  If so, connect to it and go to http://192.168.4.1.  Then configure you WiFi network.  For Nextion: Check if the baudrate is 115200.

wa8lbz, 2022-04-05:
No I do not see the access point (AP). Yes, the baud rate is 115200. I could see the AP before the March 26 code update.

Edzelf, 2022-04-05:
Then show the complete logging.

wa8lbz, 2022-04-05:
Here is all logging. Upload, uploadfs and debug.
[Uploaad.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8422509/Uploaad.txt)
[Uploadfs.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8422510/Uploadfs.txt)
[Debug.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8422511/Debug.txt)

Edzelf, 2022-04-06:
NVS is still empty, but you should see the AP.  Otherwise the ESP32 is defective.

wa8lbz, 2022-04-06:
Now I am really confused. The pio run -t uploadfs command was successful, but no AP. I have three ESP32 DOIT DEVKIT modules, one brand new, all giving the same results.

wa8lbz, 2022-04-06:
Below is my defaultprefs.txt file, in the data directory, that was uploaded. Is there anything wrong with it?
[Defaultprefs.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8429475/Defaultprefs.txt)

wa8lbz, 2022-04-07:
[Uploasfs Terminal Output 2.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8447475/Uploasfs.Terminal.Output.2.txt)

wa8lbz, 2022-04-07:
In the last message, the part about the missing spiffs.bin file is in error. The file is there. My mistake. But I am still concerned about the terminal logging for the pio run -t uploadfs command. You said the NVS was empty and I am trying to find out why. And I want to know what the "keys" are that are going into the NVS.

wa8lbz, 2022-04-08:
Is there anything wrong with theplatformio.ini file:
[Platformio dot ini.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8448313/Platformio.dot.ini.txt)

Edzelf, 2022-04-08:
Defaultprefs.txt is not needed at all.  It is just a help to fill NVS after you have AP running.  Also uploadfs is not vital.  The first thing that is important is the AP.  If you cannot get AP to work, you can not proceed with the radio.

Edzelf, 2022-04-08:
NVS contains the configuration parameters.  Examples are the presets, volume, wifi credential.  Read the documentation for details.  NVS will be filled using the config page of the web interface.

wa8lbz, 2022-04-09:
I understand that if you cannot get the AP to work, I can't proceed with the radio. I was able to get the AP until the code changed to fix the Nextion problem. After that, no AP. I have tried three ESP32 DOIT DEVKIT boards all with the same results. I have tried every thing I know to do but cannot solve the no AP problem. I get the same result using the Arduino IDE. What can I check next to solve the no AP problem? You say the NVS contains the configuration parameters and you also said my NVS is empty. I don't know what else to do. Do you want to see more logging outputs?

Edzelf, 2022-04-09:
Again, step 1 is to have AP working.  Don't mind NVS or defaultprefs.txt.

Edzelf, 2022-04-09:
Sometimes it helps to erase the flash completely and then install SPIFS and sketch again.

wa8lbz, 2022-04-10:
I will try anything to get the AP to work. Will the command )pio run erase -target) erase the flash? If not, how do I erase the flash?

Edzelf, 2022-04-10:
The command is `pio run -t erase`.
You can also find the command in the left pane if you select the pio icon:
[image]

wa8lbz, 2022-04-11:
Still no AP. Here is the debug logging after performing these commands: pio run -t erase; pio run -t upload; and pio run -t uploadfs. All completed successfully. Could there be something wrong in the platformio.ini file or some other platform specific files?
[Debug Logging.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8466512/Debug.Logging.txt)

Edzelf, 2022-04-12:
The debug shows an error in the serial communication (PlatformIO to ESP32).  So the platformIO crashes, not the ESP32.  Maybe the serial chip on the devkit fails.
It seems that the crash happens at the moment the WiFi of the ESP32 is switched on.
Are you sure you have a good  power supply (at least 1 Amp) for the ESP32 connected to "VIN"?  If you only connect via the USB of a PC/laptop there may be insufficient power.
I always use a USB hub with external powe rsupply.

wa8lbz, 2022-04-12:
Good thought. I am in the process of getting a powered USB hub. Today for the first time I received this error message:
[image]

Lesior, 2022-04-13:
Hi I got the same problem. In my case I got debug message: 
```
D: Connect to WiFi
D: WiFi Failed!  Trying to setup AP with name ESP32-Radio and password ESP32-Radio.
D: IP = 0.0.0.0
```
When I use the old build (https://github.com/Edzelf/ESP32-Radio) then the AP works fine. Therefore, I think there's something messed up with the libs or dependencies. I don't know what, though. 
Thanks for hard work Ed, by the way! 

Pawel.

Edzelf, 2022-04-13:
This is not an error message.  The IP address should be 192.168.4.1 and the AP will be available.Verzonden vanaf mijn Galaxy

"target": "https://github.com/Edzelf/ESP32Radio-V2/issues/29#issuecomment-1098307877",
"url": "https://github.com/Edzelf/ESP32Radio-V2/issues/29#issuecomment-1098307877",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

Lesior, 2022-04-13:
Exactly, I got IP address 192.168.4.1 when I use the old ESP-Radio version. I disconnected all the peripherals and even created a new project in PlatformIO from scratch. Effect is that I can connect to the AP with WiFi, but the index.html page can't be displayed. And again when I do the same with the old version of the ESP-Radio I can connect and the index.html is displayed correctly.

Edzelf, 2022-04-13:
You probably forgot to upload the webpages to SPIFFS.

wa8lbz, 2022-04-14:
I am now using a powered USB hub. Voltage measured at Vin in on ESP32 is 4.34 VDC and on 3V3 pin 3.37 VDC. So, the 3.3V regulator on the ESP32 board is functioning properly. Still no AP. Same results with three different ESP32 DOIT DEVKIT V1 boards. Was able to get an AP with the old code but not with the new code. I used a separate sketch I acquired and it established an AP (192.168.1.4), with two way communications, with no problem. There must be something wrong with the main.cpp file, one of the libraries, a platformIO specific file or platformIO itself. I have no more Ideas. I now see someone else is having the same problem of not being able establish an AP.

tsctrl, 2022-04-14:
i actually have the same issue.

i run the arduino sketch to set the preference.
in v1 it set the value. in v2 it is not.

after run the sketch, i upload the firmware, connect to station 192.168.4.1. open the config page and it is asking to create new default config. endup i manually fill up the config in v2.

in v1 all preferences was set once firmware was uploaded.

tsctrl, 2022-04-14:
@wa8lbz , that issue shows if you install vs with client installer. need host installer to work with pio. and not related to the wifi issue

@Lesior you need to use spiff upload tools to upload the data folder to your esp spiffs partition.

i did have wifi connection issue on v1.  there is some issue with the wifi library need to have it to be initialize with some additional commands. maybe wifi.disconnect() or something i cant recall.

wa8lbz, 2022-04-16:
Ed, Thanks for you hard work. After you last changes I can get the AP Web Interface at 192.168.0.4. All is working except for a problem with preset. This will be a new issue for me.

wa8lbz63, 2024-07-22:
I am also having the same problem as the others but can't seem to find the problem. I Have this question ... What should be entered into Config.h line 16 for #FIXEDWIFI? My SSID/PW doesn't work. When I push the EN button on the ESP32, the first error that appears in the terminal output is SSID too long or is missing. Below is the printout. I am confused. 

[device-monitor-240722-150512.log](https://github.com/user-attachments/files/16339176/device-monitor-240722-150512.log)

Edzelf, 2024-07-23:
Please use the "Fri, 05 Jul 2024 13:40:00 GMT" version.

wa8lbz63, 2024-07-24:
I am still having problem running the 5 July 2024 version. The following commands pio run -t upload and pio run -t uploadfs uploaded with no problems. However, pressing the reset button on the ESP32 revealed errors in the device monitor readout. The full readout is supplied below. I am still confused as to what should be entered for FIXEDWIFI in the config.h. I tried my credentials and ESP32-Radio/ESP32-Radio and apparently neither of them are correct.

[device-monitor-240724-144748.log](https://github.com/user-attachments/files/16366554/device-monitor-240724-144748.log)

Edzelf, 2024-07-24:
Just edit the definition in config.h:
`#define FIXEDWIFI "ARRIS-E4B1/ppppppppp"                    // Add a fixed SSID to the list (WiFi only)`
where "ppppppppp" is the password.

Otherwise, connect to de ESP-radio access point, browse to 192.168.4.1 and edit the wifi setting in de config page of the webinterface as described in the documentation.

wa8lbz63, 2024-07-24:
I defined FIXEDWIFI in config.h to be ARRIS-E4B1/my password. Again the pio run -t upload and pio run -t uploadfs uploaded with no errors. I get the same results. The complete device monitor output is below. 192.168.4.1 does not connect to the ESP-radio.
[device-monitor-240724-194030.log](https://github.com/user-attachments/files/16368506/device-monitor-240724-194030.log)

wa8lbz63, 2024-07-25:
I think I might have found the problem, but I don't understand it. See Photo's 1 and 2. Photo 1 shows the wifi networks my computer sees and Photo 2 shows the wifi networks discovered in the device monitor output. Notice my wifi network is ARRIS-E4B1-5 but the device monitor shows ARRIS-E4B1. I only have one wifi network and it is ARRIS-E4B1-5.

[image]
[image]

Edzelf, 2024-07-25:
You are close.  It must be the name of the network.  Maybe the second hyphen ("-") is a problem.  Can you change the network name to a simple name or add a (guest) network with a simple name?
Note also that the network must be 2.4 GHz, 5 GHz does not work for an ESP32.

Did you try to connect your PC with the WiFi network of the ESP32.  It name is "ESP32-Radio" and the key is also "ESP32-Radio".

wa8lbz63, 2024-07-25:
If I select the ESP32-Radio network and enter 192.168.4.1 I can gain access to ESP32-Radio. I changed my network name to ARRIS-E4B1, but I still get the same results. No such network found. I would really like to access ESP32-Radio without having to change networks like I did before.

Edzelf, 2024-07-26:
Sorry, I'm out of solutions.  There must be someting strange with your WiF network or with your ESP32.  Could you try to connect to the network of a friend or neighbour?

wa8lbz63, 2024-07-26:
I did a little checking. My WiFi network is capable of operating at two speeds: ARRIS-E4B1 is 2.4 GHz and ARRIS-E4B1-5 is 5 GHz. I am using ARRIS-E4B1 (2.4 GHz). I too have come to the end of my ability to troubleshoot this problem. I am sorry because I really wanted the ESP32 Radio V2 project to work for me like the V1 did.

Edzelf, 2024-07-26:
Sometimes it helps to fully erase the flash and to upload again.

wa8lbz63, 2024-07-27:
I just completed the following steps using the same ESP32:

1. Checked to make sure that my local network is ARRIS-E4B1 2.4 GHz and it is.

2. Started PIO and performed a PlatformIO Clean. Result was Success.

3. Performed a ESP32 erase with pio run -t erase. Result was Success.

4. Uploaded firmware to ESP32 with pio run -t upload. Result was Success.

5. Uploaded file system to ESP32 with pio run -t uploadfs. Result was Succes.

6. Pressed reset button on ESP32 and observed Device Monitor output. Result, I can now gain control of the ESP32 Radio with 192.168.0.62.

Something in the above steps seems to have solved the problem.