Edzelf/ESP32Radio-V2 #30: Crashens on mp3 stop

on7lds, 2022-04-10:
The version I downloaded on 1/4/2022 did work well.
The actual version has some problems:
* 'stop' command gives '**D: ESP32-Radio called with illegal parameter: stop**'
* same for 'resume' and 'settings'
* changing preset (rotary, remote control, command) crashes (and reboots with same preset)
 
[crash.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8458778/crash.txt)

using VS1053, VS1838B, rotary encoder, NEXTION

Edzelf, 2022-04-12:
At the moment I cannot reproduce the crash.  Commands "stop" and "resume" are not supported in this version, the buttons are also removed from the webinterface.  The command "settings" works, it is used in the webinterface to show the current settings and the presets.

Edzelf, 2022-04-12:
I managed to reproduce the crash, so I was able to debug the program.
It is fixed now.

on7lds, 2022-04-13:
this is fixed, indeed.
Thanks.

closed 2022-04-13 by on7lds