Edzelf/ESP32Radio-V2 #31: Recomended setup for DAC

tsctrl, 2022-04-12:
hi, 

anyone could share feedback on using internal/external dac for this radio or maybe using spi i2s module for this radio.

is it worth to use external circuit for better audio bits and quality audio output? what dac was used?

this could be more question or discussion and not a bug report. move to discussion section if required. 

thanks at the earliest!

tsctrl, 2022-04-13:
hi @Edzelf,

what benefits using dac instead of vs1053? stereo dac such as pcm5102 are more expensive than vs1053. 

why dac preferable for production environment than a vs? i guess dac will cost more than vs

tsctrl, 2022-04-13:
agree, roughly the same for simple pcm module. pin are lesser on pcm. just tested both pcm and vs1053. quality is more on vs. but vs1053 got loud FM hissing kind tone for few millis intermittently on my end during playback. but not by using PCM module.

do you encounter the same problem?

Edzelf, 2022-04-13:
No, no distorsion with the VS1053.  But I am using 2 audio transformers.

tsctrl, 2022-04-13:
i have use audio transformer and voltage isolator below to separate power used. i even has use 2 power source for module with usb power and the pam with external power bench.

audio transformer help a lot to remove noise, hissing and clicking sound. but using separate power source and isolator dint help much.

pam 8403 module did not do the job well, to many noise and i have distortion on volume higher then 80%. 

[image]

Edzelf, 2022-04-13:
The thing in picture looks like a filter, not like an transformer or isolator.

tsctrl, 2022-04-14:
i see, just look back at the item description and is actually an amplifier filter. interesting where it is to be connected to the volt and not to speaker. i guess it will also or suppose to filter noise from the pam8403 vcc line. but it is not in my case.

wondering if there is amplifier or modules that can resolve the common ground issue. but my guess is it is not common ground issue. is how the amp it self to be connected to the ground. 

do you have simple or maybe connection diagram with a simple snubber or capacitor based isolator or breaker like the approach use in adafruit vs breakout board? maybe a larga caps on the audio output line before the amps? 

hardly to understand the adafruit circuit and keep wondering the simpler approach to connect smt capacitor in breadboard for that.

dreamy1, 2022-04-14:
This is all related to the mismatching circuit of VS1053B to the class D. For the PAM8403, you can find here some useful hints (german only):

https://esp8266-server.de/Internetradio.html#SchaltplanA 

The quick&dirty solution which will give you a big improvement is here: https://www.mikrocontroller.net/attachment/330246/amp.png

Good luck!
Stefan

dreamy1, 2022-04-14:
BTW: i have designed my own hardware with VS1053 and applied class D. The sound is amazing, no transfomer etc. are needed.

tsctrl, 2022-04-15:
thanks @dreamy1! that really help for me to understand the issue. vs module has lots of caps and resistors. i dont want to get my mind to start to look into designing my own vs. but i guess, thats really worth it. probably sharing your schematic would be something lot of people will appreciate by a lot. :-P
so you still opt in the vs option instead of dac. wondering if there is dac that could probably beats the vs.

tsctrl, 2022-04-15:
@dreamy1, did you made your vs-pam having connection as the solution mention as below?
[image]

anyone use common pam 8403 module as below with modified connection as above?
[image]

dreamy1, 2022-04-15:
My application circuit is very complex and i use another amplifier with different audio input path than the PAM, so it think that doesn't really help. If you follow the instructions above, all your problems with the PAM are a thing of the past. 

The VS1053 is a DAC with integrated streaming option and with very acceptable audio path, so i don't see any benefit to use a separate DAC and burden the ESP32 additional work in an application most running at maximum CPU usage.

dreamy1, 2022-04-15:
All you have to do is a connection from GBUF pin of the VS1053 to the VREF pin of the PAM, then the noise will disappear.

tsctrl, 2022-04-15:
great! new milestone archive for me to start looking for a PAM replacement and to start designing my own VS module. i wish i could get yours. :smile: :smile: :joy: :stuck_out_tongue_closed_eyes:

dreamy1, 2022-04-15:
VS1053 pcb design is really complex to get a good result, there are a lot of things to pay attention for like ground planes, current paths and ripple handling of the power supplies. 

Here is a photo of my actual design with ethernet support, a class D amp is stacked on top (not in the picture):
[image]

dreamy1, 2022-04-15:
What is missing...is somebody who can help to get the HSPI bus to work...see https://github.com/Edzelf/ESP32Radio-V2/issues/26

tsctrl, 2022-04-23:
hi @Edzelf , do the i2s will support other DAC than the pcm5102? plug and play possible?

Edzelf, 2022-04-23:
It is standard I2S, so most devices with I2S input should work.

tsctrl, 2022-05-15:
hi @Edzelf ,

figure out new dac module. someone already created library for this module too:
[image]

with prebuild amps, line out and line in in one package.

library:
https://github.com/CelliesProjects/wm8978-esp32/tree/master/src