Edzelf/ESP32Radio-V2 #32: Bluetooth A2D

tsctrl, 2022-04-12:
Hi @Edzelf,

do the Bluetooth A2D feature ready? 
it does mean we could pair mobile phone to esp as external speaker right?

looking forward to try this awesome feature
thank you!

Edzelf, 2022-04-13:
I don't think this is possible.  Using WiFi and BT with only one antenna is too much.

tsctrl, 2022-04-13:
i guess antenna is not an issue, just flash space and memory constraints.

tsctrl, 2022-04-13:
probably adding the where a2d chunk from bt to be feed to the i2s is something that mostly valueable and to have that to work. i did not understand those or know how to do it.

we could just simply turned off the wifi, disable the webserver to use bt or other services such as mdns or mqtt as an option. having all to be working together as a future enhancement.

hope you could shed some light on how is a2d stream could be feed to the chunk and queue for processing so we could start looking into it.

thanks!