Edzelf/ESP32Radio-V2 #33: uppreset and downpreset not working

on7lds, 2022-04-13:
uppreset always selects preset 1
downpreset always selects preset -1 (!)

```
D: IR code A05F received. Will execute uppreset = 1
D: Preset is now 1
D: Radiofuncs cmd is 1
...
D: IR code A05F received. Will execute uppreset = 1
D: Preset is now 1
D: Radiofuncs cmd is 1
...
D: IR code A05F received. Will execute uppreset = 1
D: Preset is now 1
D: Radiofuncs cmd is 1
...
D: IR code E21D received. Will execute downpreset = 1
D: Preset is now -1
D: Radiofuncs cmd is 1
...
D: IR code E21D received. Will execute downpreset = 1
D: Preset is now -1
D: Radiofuncs cmd is 1
```

Edzelf, 2022-04-13:
Does it work with the web interface?

on7lds, 2022-04-13:
no, same problem (prev = preset -1 and next = preset 1)

Edzelf, 2022-04-13:
I can not reproduce it, show me your presets.

on7lds, 2022-04-13:
clk_dst = 1
clk_offset = 2
clk_server = 192.168.250.6
#
ir_0AF5 = status
ir_18E7 = tonehf = 1
ir_1AE5 = preset=4
ir_22DD = status
ir_3AC5 = preset=0
ir_40BF = preset=3
ir_50AF = upvolume = 2
ir_728D = tonehf = 15
ir_7887 = preset=1
ir_7A85 = preset=2
ir_906F = downvolume = 2
ir_A05F = uppreset = 1
ir_A25D = mute
ir_A857 = toneha = 1
ir_C03F = mute
ir_C23D = toneha = 15
ir_E21D = downpreset = 1
ir_FA05 = reset
#
mqqprefix = none
mqttbroker = none
mqttpasswd = ESP32
mqttport = 1883
mqttuser = none
#
pin_enc_clk = 32     # GPIO rotary encoder CLK
pin_enc_dt = 35      # GPIO rotary encoder DT
pin_enc_sw = 34      # GPIO rotary encoder SWitch
pin_ir = 33                    # GPIO IR receiver VS1838B
pin_vs_cs = 5                  # GPIO Pin number for VS1053 "CS"  (If DEC_VS1053 defined)
pin_vs_dcs = 21                # GPIO Pin number for VS1053 "DCS" (If DEC_VS1053 defined)
pin_vs_dreq = 4                # GPIO Pin number for VS1053 "DREQ"(If DEC_VS1053 defined)
#
preset = 0
preset_000 = icecast.vrtcdn.be/radio1-high.mp3
preset_001 = icecast.vrtcdn.be/ra2vlb-high.mp3
preset_002 = icecast.vrtcdn.be/radio1_classics-high.mp3
preset_003 = radios.rtbf.be/classic21-128.mp3
preset_010 = icecast.omroep.nl:80/radio1-bb-mp3
#
toneha = 15
tonehf = 15
tonela = 15
tonelf = 11
#
volume = 30
#
wifi_00 =*****/*****

Jefke1, 2022-04-13:
same problem for me, as I mentioned before in Nextion display does not work(next & prev). when you press next the preset does not increase it always chooses preset 1, even if preset 7 is used as startup. When using prev the sound drops out and reboots.
[Prev-Next.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8479614/Prev-Next.txt)

Edzelf, 2022-04-13:
Some presets in the list above are not real stations.  If a stations redirects to another station, the original preset number was not memorized.  I could not reproduce this bug because I never used redirect stations.
It is fixed in the current version.
Note that in the list above presets 4..9 are missing.  This may result in unexpected behaviour.

on7lds, 2022-04-14:
This problem seems to be solved.
I somtimes still have crashes when changing preset often, but will try to find a way to reproduce it consistently

closed 2022-04-14 by on7lds

Edzelf, 2022-04-14:
I am redesigning the station selection now.