Edzelf/ESP32Radio-V2 #34: Problem with presets.

wa8lbz, 2022-04-16:
I have some problems with presets. They are itemized in this document:
[Presets.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8500547/Presets.txt)

Edzelf, 2022-04-16:
Show the presets and the version date/ time.

wa8lbz, 2022-04-18:
I forgot to mention earlier that I am using IR codes to control the radio. The version is Thur 14 Apr 2022 09:10:00 GMT. The presets are shown below:
[image]

Edzelf, 2022-04-18:
The newest version of 15 April is much better.

wa8lbz, 2022-04-18:
I have uploaded the April 15 version. Now uppreset and downpreset IR command don't work at all. A glitcy sound is heard on the radio when IR commands uppreset or downpreset are issued but nothing else happens. The upvolume and downvolume IR commands work OK.

Edzelf, 2022-04-18:
Show the config.h.

wa8lbz, 2022-04-18:
Here is my config.h:
[Config dot h.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8508646/Config.dot.h.txt)

Edzelf, 2022-04-19:
I have tested it, but I could not reproduce the bug.  Check the webinterface if there are more than 1 presets.  When you press the remote-button for CH+ ou should see something like this:
```
D: IR code E21D received. Will execute uppreset = 1
D: updateNr 3 <= 7 to 1, relative is 1
D: updateNr result is 4
D: nextPreset is 4, host stream.laut.fm/kinderlieder-123          #  4 - Kinderlieder
D: Preset is now 4
D: Radiofuncs cmd is 1
D: MP3Decode error -6
```

wa8lbz, 2022-04-19:
Below is a copy of my whole Web Interface config screen and the results of buttons pressed on my IR remote:
[My Web Interface Config.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8515238/My.Web.Interface.Config.txt)
[IR Remote Button Presses.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8515241/IR.Remote.Button.Presses.txt)

Edzelf, 2022-04-19:
As you can see:
```
D: updateNr 0 <= 0 to 1, relative is 1
D: updateNr result is 0

There is only one station. Check the config page of the webinterface to check the presets.  During startup you should see the presets in the logging.

wa8lbz, 2022-04-20:
My preset problem is solved. One of the presets was not streaming. When uppreset or downpreset to this non streaming station causes the radio to get confused and do nothing. I removed the non streaming preset and the uppreset and downpreset commands seem to be working properly now. Ed, thanks for your help with my preset problem.

Next I will report a problem with the items printed on page 0 of the Nextion display during startup.

wa8lbz, 2022-04-20:
I spoke too soon. The preset problems are back. I added a number of preset stations to bring the total up to 11. I checked the Web Interface config screen for errors but did not find any. Here are the problems I see:

1. The default preset station is set to (preset = 9) but the radio comes on with preset 00.

2. Performing uppreset and downpreset comands works normally for a while. Then the stations stop changing, the radio sound stops and the radio just sets there and does nothing. However, all the other IR commands are still working normally.

The trouble might be timing errors. Is the timing of receiving IR commands and associated reactions done using timing loops or interrupts?
Except for the IR preset up and down commands, all of the other IR commands work properly. It is the preset up and down commands that are causing the confusion/problem. By the way all of the preset stations are streaming. If you need more information just ask for it. Below is information that might help:
[Web Interface Config 2.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8524707/Web.Interface.Config.2.txt)

Edzelf, 2022-04-20:
You should see the preset 9 selection in the serial output.  And also the station selection by IR.  From there it would be possible to debug this problem.

wa8lbz, 2022-04-21:
Included below are radio power up logging and logging for each button pressed on the IR remote.
[Powerup Logging.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8533786/Powerup.Logging.txt)
[IR Remote Buttons.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8533788/IR.Remote.Buttons.txt)

Also, something strange showed up in the radio power up logging: This is not a VS1053, but a VS1001 instead. I am using the same VS1053 board I have been using since I began this project. I have changed nothing. And the radio no longer makes any sounds. This is quite a mystery for me.

Edzelf, 2022-04-21:
Look at the type number on the chip, not on the board.

wa8lbz, 2022-04-21:
In the Config screen of the Web Interface, when the Save button pressed, how long does it take for the saving process to complete?
When the status indicator says Config Saved? Or longer? Maybe the configuration is not being correctly saved to the NVS.

Edzelf, 2022-04-21:
About 0.1 seconds.

wa8lbz, 2022-04-22:
I replaced the VS1053 breakout board with a new one. The marking on the chip is VS1053B. The radio played normally today until I powered down and then powered up again. The power up logging still shows that the VS1053 is a VS1001. It is not. The fact that the radio played normally when first powered up today tells me the VS1053 breakout board is OK. When the radio was playing normally, the uppreset and downpreset IR commands worked normally. But the default preset set to (preset = 9), radio still came up 00. There are 11 presets (00 through 10). Once the radio stopped I could not get it playing again. The Web Interface is available with the IP 192.168.0.4. I checked the configuration screen for errors but could not find any. I don't know what to check next.

Edzelf, 2022-04-22:
If you still see the SPI errors, there is probably a problem with the power supply.

wa8lbz, 2022-04-22:
I will look into the power supplies I have used as you suggested.

More confusing and bad news to report. I go to the Web Interface with IP 192.168.0.4. I go to the Config screen, make a change, click on Save, then click on Restart. The radio restarts but now there is no sound from the radio. At the same time, the Web Interface is non responsive. After doing a lot of checking I determined that the IP address has changed from 192.168.0.4 to something else. Some of the other IP addresses that gave me access to the Web Interface after restarts are: 192.168.0.26, 192.168.4.1, and 192.168.0.13.

To complicate matters, the IP address is not printed on Page 0 of the Nextion display. From excerpts of the logging:

D: Nextion command cls BLACK
D: Nextion command page 0
D: NEXTION output 'Starting......' to 1
D: NEXTION output 'Fri, 15 Apr 2022' to 2
D: NEXTION output 'By Ed Smallenburg' to 3

The above are printed to the display.

D: Connect to WiFi
D: NEXTION output 'Connect to WiFi' to 4
D: NEXTION output 'SSID = ARRIS-E4B1' to 5
D: SSID = ARRIS-E4B1
D: IP = 192.168.0.4
D: NEXTION output 'IP = 192.168.0.4' to 6
D: Nextion command page 1
D: Start server for commands

The above are not printed to lines 4, 5, 6 on the display This means I have to restart the radio each time and look at the logging to determine what IP to use to access the Web Interface.

wa8lbz, 2022-04-23:
At the moment I am using a regulated DC power supply. In the circuit are the Nextion display, VS1053B decoder, VS1838B IR detector and ESP32 DOIT DEVKIT V1. The power supply output voltage is +5.00 Vdc and the total current drain is 274 mA. I can access the Web Interface at IP 192.168.0.4. The radio still comes on with preset 00 even though the default is to preset = 9. All the IR remote commands are working properly, right now. There is no sound coming from the radio from any preset.

wa8lbz, 2022-04-25:
I discovered why the radio starts up playing preset = 00. In the Web Interface at IP 192.168.0.4 in the Config screen, I change preset = 0 to preset = 9, press Save and press Restart. The Config screen now shows preset = 9 but the radio is still playing preset = 0. I close the Web Interface and turn off the radio. When I restart the Web Interface and the radio, the radio is still playing preset = 0. The Web Interface Config screen still shows preset = 0. It appears the the configuration information is not being saved or correctly saved to the NVS when the Save button is pressed.

Edzelf, 2022-04-25:
Solved in today's version.

wa8lbz, 2022-04-25:
Not solved here. With the default preset to preset = 10 the radio still comes up playing preset 00. Something else I discovered. When I make changes in the Web Interface configuration screen, press Save, press Restart, the radio will not come on playing. But if I press the Reset button on the ESP32 the radio comes on playing. This is repeatable every time.