Edzelf/ESP32Radio-V2 #36: I2S Problem

wa8lbz, 2022-04-28:


wa8lbz, 2022-04-28:
Since I am having on and off results with the VS1053, I am going to try I2S again. I am using a PCM5102A board connected as follows: BCK to GPIO 5, DIN to GPIO 32 and LCK to GPIO 4. I have changed the Config.h accordingly. See below. When I issued the PIO run -t upload command it fails. See below.
[image]
[image]

Edzelf, 2022-04-28:
Pio update?

wa8lbz, 2022-04-29:
I updated PlatformIO. Same results.

Edzelf, 2022-04-29:
When compiling, do you see this platform and these packages?  Check the version numbers.

```
PLATFORM: Espressif 32 (3.3.2) > DOIT ESP32 DEVKIT V1
..
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
```

wa8lbz, 2022-04-29:
This what I see:
[What I See.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8591688/What.I.See.txt)

Edzelf, 2022-04-29:
Version is okay.  Change I2S_COMM_FORMAT_STAND_I2S to **I2S_COMM_FORMAT_I2S**.  I will update it.

wa8lbz, 2022-04-29:
This is what I see after making the change:
[image]
:

Edzelf, 2022-04-29:
Can you find out out what is refering to "internal_dac"?  I cannot see any reference.
Compare your compiler output to the attached listing.
[compile.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8592793/compile.txt)
.

wa8lbz, 2022-04-29:
I compared your output to my output but could not find any reference to "internal_dac". Below is my output:
[My Output.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8594482/My.Output.txt)

wa8lbz, 2022-04-29:
This is one place I found reference to "inter_dac"
[image]
.

Edzelf, 2022-04-29:
Mm, that's an old version.  I replaced helixfuncs.h with the current version.

wa8lbz, 2022-04-30:
The current version of helixfuncs.h allowed a successful upload. The I2S radio now plays. At this time there are still two outstanding problems to be resolved.

Problem #1: No matter how many times I change the default preset = xx, click on Save and click on restart, the radio starts up playing preset 00. This still happens when I power the radio down then power it back up. The radio is still playing preset 00. All of the other preset functions work as they should.

Problem #2: According to the debug output: 'SSID = xxxxxxxxx' should appear on page 0 line 5 (t4) of the Nextion display and 'IP= 192.168.0.13' should appear on page 0 line 6 (t5) of the display. But they don't. It would be very helpful to have the IP address appear on the display.

Ed, thanks for all your help.

Edzelf, 2022-05-04:
Problem 1 should be corrected now.

wa8lbz, 2022-05-04:
Yes, problem 1 is now solved. Looking forward to the resolution of problem 2.

wa8lbz, 2022-05-12:
I get the error messages shown below when uploading the latest code. It looks similar to the error messages when older Helixfuncs.h file was used.
[Maybe Old Helixfuncs dot h.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8681529/Maybe.Old.Helixfuncs.dot.h.txt)

Edzelf, 2022-05-12:
Use  the last version of the espressif32 framework (pio update).

wa8lbz, 2022-05-12:
I down graded to Espressif 4.2.0. Now I only get one error message during upload. See below.
[image]

Edzelf, 2022-05-12:
Do a "clean" and/or remove .pio and .vscode directories.

wa8lbz, 2022-05-13:
Performed a Clean successfully. Removed .pio and .vscode directories. The upload gave the error message below.
[image]

wa8lbz, 2022-05-13:
I forgot to mention that the .pio directory comes back immediately after deleting it. Maybe this is normal. I tried an upload again and this time it was successful. The the radio just continues to reboot over and over again.

wa8lbz, 2022-05-13:
Successful uploads but endless radio reboots. Below is the Serial Monitor logging:
[image]

Edzelf, 2022-05-15:
The traceback is about a crash of platformio caused by a failing serial to USB driver.  Could be the powersupply.

armecozzi, 2022-05-17:
Sorry for not getting back to you sooner, but my Internet provider changed and am having all kinds of problems. You are right, the constant rebooting was the result of a faulty power supply. Everything is now working as it should. Now all the lines on page 0 of the display are shown. Suggestion, add a 4 or  5 second delay after page 0 is complete. This gives time to view what is on page 0 before it changes to page 1. Ed, thank you for all of your hard work.

Edzelf, 2022-05-18:
Add some extra time, se line 1640..1643:
```
  #ifdef NEXTION
    delay ( 2000 ) ;                                    // Show for some time
    dsp_println ( "\f" ) ;                              // Select new page if NEXTION 
  #endif
```

wa8lbz, 2022-05-24:
Made the appropriate changes in the code. Timing now good.

Ed, I want to thank you for all of your hard work. Thanks to you, all of the functions/hardware in my ESP32 Radio V2 are now working as they should.

wa8lbz, 2022-05-25:
I think I have found a problem. The picture below shows the GPIO pins for the I@S interface. Notice the description of the data output pin as i2s_din_pin.  
[image]

Now look at the picture below showing the Preferences like... here the data output pin is i2s_bin_pin.
[image]

One of the descriptions is wrong.