Edzelf/ESP32Radio-V2 #41: SD Card Mount Failed!

tsctrl, 2022-05-10:
Hi @Edzelf , to raise issue regarding sd implementation.

i was unable to load/start sd  for unknown reason. i have swapping sd library, using different lib version and add some modification to the codes. i guess this is related to use shared spi bus for sd, tft with no cs connected, and the vs on a wrover module. am using spi instance with vspi pin definition. 
```
D: Starting ESP32-radio running on CPU 1 at 240 MHz.
D: Version Thu, 05 May 2022 09:25:00 GMT.  Free memory 113792
D: Display type is BLUETFT
D: SPIFFS is okay, space 1378241, used 33634
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 39 keys from NVS
D: pin_ir set to 5
D: pin_enc_clk set to 35
D: pin_enc_dt set to 34
D: pin_enc_sw set to 39
D: pin_tft_dc set to 2
D: pin_tft_scl set to 14
D: pin_tft_sda set to 13
D: pin_sd_cs set to 33
D: pin_vs_cs set to 15
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: Pin 14 is already reserved!
D: pin_spi_sck set to 14
D: pin_spi_miso set to 12
D: Pin 13 is already reserved!
D: pin_spi_mosi set to 13
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is HIGH
D: GPIO39 is HIGH
D: Enable pin 5 for IR
D: Start BLUETFT display
D: Create list with acceptable WiFi networks
D: updateNr 0 <= 0 to 5, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: Slow SPI, Testing VS1053 read/write registers...
D: This is not a VS1053, but a VS1003 instead!
D: Fast SPI, Testing VS1053 read/write registers again...
D: This is not a VS1053, but a VS1003 instead!
D: Connect to network
D: Start web server
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 21:33:01
D: SD Card Mount Failed!
D: updateNr 0 <= 2 to 5, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: Radiofuncs cmd is 1
D: Connect to host 109.206.96.34:8100
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -   808
D:   1 -    33
D:   2 -     0
D:   3 -  6000
D:   4 -     0
D:   5 -  AC45
D:   6 -  5084
D:   7 -  1E06
D:   8 -     0
D:   9 -     0
D:   A -     0
D:   B -  F8F8
D:   C -     0
D:   D -     0
D:   E -     0
D:   F -     0
D: send GET command
D: Switch to HEADER
D: Headerline: icy-name:NAXI LOVE RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) - 128k
D: Headerline: icy-genre:Love and Romance
D: Headerline: icy-url:http://www.naxi.rs
D: Headerline: content-type:audio/mpeg
D: Headerline: icy-pub:0
D: Headerline: icy-metaint:8192
D: Headerline: icy-br:128
D: Switch to DATA, bitrate is 128, metaint is 8192
D: Streamtitle found, 47 bytes
D: StreamTitle='Ja biram sta slusam';StreamUrl='';
```

Edzelf, 2022-05-10:
Pin 12 and 13 double used.

tsctrl, 2022-05-10:
cant get it, pin 12 is only for miso spi. and 13 is for mosi. and the tft sda which is mosi set to 13
Pin 13 is already reserved! this error is because i declare 13 as mosi which is tft sda which also a mosi which is correct?

tsctrl, 2022-05-13:
cause: incompatible sd module. problem solved

closed 2022-05-13 by tsctrl