Edzelf/ESP32Radio-V2 #42: i2c LCD2004 display not displaying title or radio station

timo619, 2022-05-12:
Hi, 

I have connected a LCD2004 display to my ESP-wroom-32 with an encoder. Audio via i2s `#define DEC_HELIX `.

The display only shows date and time. When I press the encoder (singleclick) it says Mute in the 2nd line. 
When I doubleclick it shows the station and after 2 seconds  the text "Turn to select station Press to confirm".
The preset is not shown when I turn the encoder. Also the Mute is not going away when I unmute. Volume bar is also not shown when turning the encoder. 
[image]
[image]
[image]

Do you have any tips for me where to search the problem in the source code?

Edzelf, 2022-05-12:
Please try today's version.  I have tested it, although without the encoder.

timo619, 2022-05-12:
Thank you very much. Great work. It is working now as expected. See picture below:
[image]

closed 2022-05-12 by timo619