Edzelf/ESP32Radio-V2 #43: Feature request: Bluetooth audio streaming

timo619, 2022-05-12:
Hi,

is it possibly to add Bluetooth audio streaming? Could also be as a preset station to save buttons.  

Here are some examples I found online:
[https://www.pschatzmann.ch/home/2020/03/27/1255/](https://www.pschatzmann.ch/home/2020/03/27/1255/)

[https://github.com/MrBuddyCasino/ESP32_MP3_Decoder](https://github.com/MrBuddyCasino/ESP32_MP3_Decoder)

Edzelf, 2022-05-12:
As far as I know, a combination of WiFi and BT simultaneously is not possible.

timo619, 2022-05-12:
I meant that you can select if you want to play a Webradio station or receive a Bluetooth audio stream from your smartphone for example. But not at the same time. 
But then there is the problem with the webserver and mqtt. Didn't think of that.

Edzelf, 2022-05-12:
You may use a BT receiver and feed the output to the radio amplifier.

timo619, 2022-05-15:
Hi, during my websearch I found the following content, which shows that Bluethooth and Wifi should be possible in generell. But a totally different setup would be necessary: 
[https://github.com/BM45/esp-adf/tree/master/examples/advanced_examples/wifi_bt_ble_coex](https://github.com/BM45/esp-adf/tree/master/examples/advanced_examples/wifi_bt_ble_coex)

A webradio with the adf framework, but bluetooth not implemented (yet):
[https://github.com/BM45/iRadioMini](https://github.com/BM45/iRadioMini)