Edzelf/ESP32Radio-V2 #45: Unable to connect WiFi

dkotte, 2022-05-13:
i am unable to connect wifi.
Config is nearly empty. Only 2 Wifi networks with correct Password ( i try only numbers)
`D: Create list with acceptable WiFi networks
D: Added MYWIFI to list of networks
D: Added STV to list of networks
D: Added SophosGuest to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 11
D:  1 - SGuest                  Signal: -73 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - STV                       Signal: -74 dBm, Encryption WPA2_PSK, Acceptable
D:  3 - ....
...
D: Connect to network
D: WiFi Failed!  Trying to setup AP with name ESP32-Radio and password ESP32-Radio.
D: IP = 1.0.0.0
D: Start web server
`
What can i do or try?

Edzelf, 2022-05-13:
Try to specify one network in config.h instead of MYWIFI.

dkotte, 2022-07-30:
Thanks for the hint ... very useful to predefine one wifi...
.. But the problem is configured local ... mot an esp32radio-problem.
Thanks!

closed 2022-08-01 by dkotte