Edzelf/ESP32Radio-V2 #49: AAC playback problem

Sanc0Pansa, 2022-05-17:
Hello. I'm having trouble playing the AAC stream http://str.pcradio.ru/Abba-hi
The example in the schreibfaul1 / ESP32-audioI2S library works correctly. (https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/I2Saudio/I2Saudio.ino)

Sanc0Pansa, 2022-05-17:
Serial output schreibfaul1 demo play correctly :

info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 225568 bytes
info        Connect to new host: "http://str.pcradio.ru/Abba-hi"
info        Connect to "str.pcradio.ru" on port 80, extension "/Abba-hi"
info        Connection has been established in 57 ms, free Heap: 222736 bytes
info        HTTP/1.1 200 OK
info        server: nginx
info        date: Tue, 17 May 2022 17:13:18 GMT
info        content-type: audio/aac, format is aac
info        AACDecoder has been initialized, free Heap: 202700 bytes
bitrate     64000
info        ice-audio-info: channels=2;samplerate=44100;bitrate=64
bitrate     64000
info        icy-name: Abba
station     Abba
icyurl      http://www.pcradio.ru/
info        access-control-allow-origin: *
info        access-control-allow-headers: Origin, Accept, X-Requested-With, Content-Type
info        access-control-allow-methods: GET, OPTIONS, HEAD
info        Switch to DATA, metaint is 16000
lasthost    http://str.pcradio.ru/Abba-hi
info        stream ready
info        buffer filled in 3 ms
info        syncword found at pos 0
info        Channels: 1
info        SampleRate: 22050
info        BitsPerSample: 16
info        BitRate: 64771
info        AAC HeaderFormat: ADTS
info        AAC Codec: MPEG-4
info        AAC Profile: LowComplexity
info        VBR recognized, audioFileDuration is estimated
info        StreamTitle=''
streamtitle 
info        StreamTitle='ABBA - People Need Love'
streamtitle ABBA - People Need Love
info        StreamTitle='ABBA - Hasta MaĂ±ana'
streamtitle ABBA - Hasta MaĂ±ana

tsctrl, 2022-05-27:
hi @Sanc0Pansa you could try to comment out this line to see if it is working
https://github.com/Edzelf/ESP32Radio-V2/blob/7d2a1ebc7a3b6f01bf8e3858b592929e202c398b/src/main.cpp#L1419

Sanc0Pansa, 2022-05-27:
Thanks for the reply, I'll try tomorrow. But I really don't believe it will help. But that stream plays ridiculously fast and with outages. The problem will be in decoding AAC VBR 64.

Sanc0Pansa, 2022-05-28:
> hi @Sanc0Pansa you could try to comment out this line to see if it is working
> 
> https://github.com/Edzelf/ESP32Radio-V2/blob/7d2a1ebc7a3b6f01bf8e3858b592929e202c398b/src/main.cpp#L1419

I tried it, it didn't work. Nothing could be played at all.
I give an example of how the AAC VBR stream plays.

https://user-images.githubusercontent.com/105736517/170811196-abd0f188-7ac2-4a23-9226-fc4511771b93.mp4

Sanc0Pansa, 2022-05-28:
If someone wants to test this on a search page, there are many stations where this problem occurs. For example, on the card of the 70's, the station 101.3 WKRD - Rockford's Best Mix (ninth in a row). 
I will add once again that this happens when playing via HELIX I2S (specifically for me PCM 5102). The VS1053 on the ESP32-Radio V1 works. I have not tested the combination of ESP32-Radio V2 with VS1053.

Edzelf, 2022-05-31:
Please give an example of a station with this problem.

Sanc0Pansa, 2022-05-31:
> Please give an example of a station with this problem.

for example:
ice24.securenetsystems.net/WRKD
ice6.securenetsystems.net/KCLC2
ice66.securenetsystems.net/R42N
fr.radio-streamhosting.com:8000/metalheartradio64aac
stream.open.fm/108

Edzelf, 2024-02-20:
Mono stations are handled correctly now.