Edzelf/ESP32Radio-V2 #50: Problems at compilation Arduino IDE  1.8.20

Fabri54, 2022-05-19:
Hi, I'm moving to version 2 but at compilation time I get the following error messages:

/var/folders/jm/8ykp7mhd767fgdf5t617f21c0000gn/T/arduino_build_301023/sketch/LCD2004.cpp.o:(.bss.dline+0x0): multiple definition of `dline'
/var/folders/jm/8ykp7mhd767fgdf5t617f21c0000gn/T/arduino_build_301023/sketch/LCD1602.cpp.o:(.bss.dline+0x0): first defined here
/Users/fabriziobianchi/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: Warning: size of symbol `dline' changed from 40 in /var/folders/jm/8ykp7mhd767fgdf5t617f21c0000gn/T/arduino_build_301023/sketch/LCD1602.cpp.o to 80 in /var/folders/jm/8ykp7mhd767fgdf5t617f21c0000gn/T/arduino_build_301023/sketch/LCD2004.cpp.o
collect2: error: ld returned 1 exit status

I've a BLUETFT display, the rotary encoder and IR receiver. The hardware works fine with version 1 even if I'm experiencing some problems but they may be subject of another post.
Has anybody any suggestion?
Thank you

Edzelf, 2022-05-19:
Try to do a "clean" and/or remove .pio and .vscode directories.
Show us your config.h

Fabri54, 2022-05-21:
I did it, a few weeks ago when I first tried V2 but I had left some libraries in the working directory as I have a Nextion and an OLED display and want to leave what necessary to play with them. 
This  time I followed to the letter the instructions and left the minimum  needed for BLUETFT in the working directory. 
The .ino sketch compiles, is linked and can be written to the ESP32 without any issue. 
However I don't see where I can assign the pins like V1 and at the reset I get

17:40:01.010 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.
17:40:01.010 -> D: Version Sun, 15 May 2022 10:30:00 GMT.  Free memory 113792
17:40:01.010 -> D: Display type is BLUETFT
17:40:01.114 -> D: SPIFFS is okay, space 1378241, used 0
17:40:01.114 -> D: Found partition 'nvs     ' at offset 0x009000 with size    20480
17:40:01.150 -> D: Found partition 'otadata ' at offset 0x00E000 with size     8192
17:40:01.150 -> D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
17:40:01.150 -> D: Read 0 keys from NVS
17:40:01.150 -> D: pin_spi_sck set to 18
17:40:01.150 -> D: pin_spi_miso set to 19
17:40:01.150 -> D: pin_spi_mosi set to 23
17:40:01.150 -> D: GPIO0 is HIGH
17:40:01.150 -> D: GPIO2 is HIGH
17:40:01.184 -> D: GPIO4 is HIGH
17:40:01.184 -> D: GPIO5 is HIGH
17:40:01.184 -> D: GPIO12 is HIGH
17:40:01.217 -> D: GPIO13 is HIGH
17:40:01.217 -> D: GPIO14 is HIGH
17:40:01.217 -> D: GPIO15 is HIGH
17:40:01.217 -> D: GPIO16 is HIGH
17:40:01.251 -> D: GPIO17 is HIGH
17:40:01.251 -> D: GPIO18 is HIGH
17:40:01.251 -> D: GPIO19 is HIGH
17:40:01.284 -> D: GPIO21 is HIGH
17:40:01.284 -> D: GPIO22 is HIGH
17:40:01.284 -> D: GPIO23 is HIGH
17:40:01.284 -> D: GPIO25 is HIGH
17:40:01.320 -> D: GPIO26 is HIGH
17:40:01.320 -> D: GPIO27 is HIGH
17:40:01.320 -> D: GPIO32 is HIGH
17:40:01.353 -> D: GPIO33 is HIGH
17:40:01.353 -> D: GPIO34 is LOW, probably no PULL-UP
17:40:01.353 -> D: GPIO35 is HIGH
17:40:01.387 -> D: GPIO39 is LOW, probably no PULL-UP
17:40:01.387 -> D: Start BLUETFT display
17:40:02.162 -> D: Create list with acceptable WiFi networks
17:40:02.162 -> D: Added Vodafone-34350430-2p4 to list of networks
17:40:03.294 -> D: Scan Networks
17:40:05.000 -> D: Scan completed
17:40:05.000 -> D: Number of available networks: 4
17:40:05.000 -> D:  1 - Vodafone-xxxxxxxx-yy     Signal: -51 dBm, Encryption WPA2_PSK, Acceptable
17:40:05.035 -> D:  2 - FASTWEB-1234567_EXT      Signal: -86 dBm, Encryption WPA_WPA2_PSK, 
17:40:05.035 -> D:  3 - Vodafone-7654321_EXT    Signal: -91 dBm, Encryption WPA2_PSK, 
17:40:05.035 -> D:  4 - DIRECT-04-HP OfficeJet 5200 Signal: -92 dBm, Encryption WPA2_PSK, 
17:40:05.035 -> D: End of list
17:40:05.169 -> D: Slow SPI, Testing VS1053 read/write registers...
17:40:05.169 -> D: VS1053 not properly installed!
17:40:05.268 -> D: Connect to network
17:40:07.282 -> D: Try WiFi Vodafone-34350430-2p4
17:40:11.159 -> D: SSID = Vodafone-34350430-2p4
17:40:11.193 -> D: IP = 192.168.1.31
17:40:11.193 -> D: Start web server
17:40:11.193 -> D: Network found. Starting mp3 client, mqtt and OTA
17:40:11.193 -> D: MDNS responder started
17:40:12.191 -> D: Rotary encoder is disabled (-1/-1/-1)
17:40:12.191 -> D: Sync TOD
17:40:12.225 -> D: Sync TOD, new value is 17:40:12
17:40:14.237 -> D: nvs_get_str failed 1102 for key preset, keylen is 6, len is 150!
17:40:14.237 -> D: Contents: 
17:40:14.237 -> D: updateNr 0 <= 0 to 0, relative is 0
17:40:14.237 -> D: updateNr result is 0
17:40:14.237 -> D: nextPreset is 0
17:40:14.237 -> D: Radiofuncs cmd is 1
17:40:14.275 -> D: Connect to host 
17:40:14.275 -> D: Request  failed!

and the TFT screen remains white
I assume that the lack of pin assignment, with the exception of ski, miso and most may be a significant inconvenient but I'm so dumb that I don't find the file where they are set like V1.

Edzelf, 2022-05-21:
With your browser, go to http://192.168.1.31 and then to the config page.  Here you add lines for your configuration.  There is a example configuration available on that page.  The documentation lists all possible settings.

Fabri54, 2022-05-21:
OK, I will do but I'm a bit baffled. Why not allowing to customise a config file to be loaded with the sketch or with the ESP32init.ino ?  
Before sending this reply I tried connecting the Web Browser   and I get connection timed out.
What I'm missing?

Edzelf, 2022-05-21:
I see that your SPIFFS is still empty.  You have to upload the files in the data directory to SPIFFS.
And you are right, you may use the init sketch to preload the configuration,  but I think the web interface is more convenient.

Fabri54, 2022-05-21:
Hem... here comes my lack of experience. I still cannot connect my browser to the IP address assigned by the router. Initially I got "connection timed out", now has turned into a polite "sorry". Consequently I cannot use the web interface to config the parameters, pins, etc.
Now, which files should I load into the SPIFFS and how?
My apologies for being a pain in the neck but I'm just a little step above "beginner" and here I'm stuck. Thank you

Fabri54, 2022-05-21:
I'm sorry, I had completely forgot how to use SPIFFS. I've loaded  the files contained into the Data folder. 
Now I can use the web interface.  I'll try to configure all the parameters tomorrow

Fabri54, 2022-05-22:
Latest update. I've copied some config and pinning data in to the config windows of the web browser.
Rotary Encoder works only for mute and vol+ / vol- commands
IR remote works for all the exa codes in the configuration.
the SD is not seen
TFT display luminosity significantly reduced. At boot the screen is almost unreadable while it's OK (low luminosity) when playing an internet radio stream..
Any suggestion about how to overcome these issues?
here is the diagnostic output at boot:
11:53:57.773 -> D⸮⸮H⸮]A⸮0xee
11:53:57.773 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:53:57.773 -> mode:DIO, clock div:1
11:53:57.803 -> load:0x3fff0018,len:4
11:53:57.803 -> load:0x3fff001c,len:1216
11:53:57.803 -> ho 0 tail 12 room 4
11:53:57.803 -> load:0x40078000,len:10944
11:53:57.803 -> load:0x40080400,len:6388
11:53:57.803 -> entry 0x400806b4
11:54:01.134 -> 
11:54:01.134 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.
11:54:01.134 -> D: Version Sun, 15 May 2022 10:30:00 GMT.  Free memory 113792
11:54:01.167 -> D: Display type is BLUETFT
11:54:01.235 -> D: SPIFFS is okay, space 1378241, used 33383
11:54:01.235 -> D: Found partition 'nvs     ' at offset 0x009000 with size    20480
11:54:01.272 -> D: Found partition 'otadata ' at offset 0x00E000 with size     8192
11:54:01.272 -> D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
11:54:01.272 -> D: Read 55 keys from NVS
11:54:01.272 -> D: pin_ir set to 35
11:54:01.272 -> D: pin_enc_clk set to 25
11:54:01.272 -> D: pin_enc_dt set to 26
11:54:01.272 -> D: pin_enc_sw set to 27
11:54:01.272 -> D: pin_tft_cs set to 15
11:54:01.272 -> D: pin_tft_dc set to 2
11:54:01.272 -> D: pin_vs_cs set to 5
11:54:01.272 -> D: pin_vs_dcs set to 16
11:54:01.272 -> D: pin_vs_dreq set to 4
11:54:01.307 -> D: pin_spi_sck set to 18
11:54:01.307 -> D: pin_spi_miso set to 19
11:54:01.307 -> D: pin_spi_mosi set to 23
11:54:01.307 -> D: GPIO0 is HIGH
11:54:01.307 -> D: GPIO2 is HIGH
11:54:01.307 -> D: GPIO4 is HIGH
11:54:01.344 -> D: GPIO5 is HIGH
11:54:01.344 -> D: GPIO12 is HIGH
11:54:01.344 -> D: GPIO13 is HIGH
11:54:01.344 -> D: GPIO14 is HIGH
11:54:01.378 -> D: GPIO15 is HIGH
11:54:01.378 -> D: GPIO16 is HIGH
11:54:01.378 -> D: GPIO17 is HIGH
11:54:01.378 -> D: GPIO18 is HIGH
11:54:01.411 -> D: GPIO19 is HIGH
11:54:01.411 -> D: GPIO21 is HIGH
11:54:01.411 -> D: GPIO22 is HIGH
11:54:01.444 -> D: GPIO23 is HIGH
11:54:01.444 -> D: GPIO25 is HIGH
11:54:01.444 -> D: GPIO26 is HIGH
11:54:01.478 -> D: GPIO27 is HIGH
11:54:01.478 -> D: GPIO32 is HIGH
11:54:01.478 -> D: GPIO33 is HIGH
11:54:01.478 -> D: GPIO34 is LOW, probably no PULL-UP
11:54:01.515 -> D: GPIO35 is HIGH
11:54:01.515 -> D: GPIO39 is LOW, probably no PULL-UP
11:54:01.515 -> D: gpio_00 will execute uppreset = 1
11:54:01.515 -> D: gpio_12 will execute upvolume = 2
11:54:01.515 -> D: gpio_13 will execute downvolume = 2
11:54:01.515 -> D: Enable pin 35 for IR
11:54:01.515 -> D: Start BLUETFT display
11:54:02.302 -> D: Create list with acceptable WiFi networks
11:54:02.302 -> D: Added Vodafone-34350430-2p4 to list of networks
11:54:02.302 -> D: Added Vodafone-34350430-2p4 to list of networks
11:54:02.302 -> D: Added NG_BIAN to list of networks
11:54:03.425 -> D: Scan Networks
[... 31 lines cut ...]
11:54:25.592 -> D: Connect to host airspectrum.cdnstream1.com:8000/1261_192
11:54:25.625 -> D: Song stopped correctly after 0 msec
11:54:25.791 -> D: send GET command
11:54:25.924 -> D: Switch to HEADER
11:54:25.924 -> D: Headerline: Content-Type: audio/mpeg
11:54:25.924 -> D: Headerline: icy-br:192
11:54:25.924 -> D: Headerline: ice-audio-info: ice-samplerate=44100;ice-bitrate=192;ice-channels=2
11:54:25.924 -> D: Headerline: icy-br:192
11:54:25.924 -> D: Headerline: icy-description:Magic Oldies Florida HD 192 Kbps with the Best Oldies on the Planet!
11:54:25.958 -> D: Headerline: icy-genre:Oldies 60s, 70s, 80s
11:54:25.958 -> D: Headerline: icy-name:Magic Oldies Florida
11:54:25.958 -> D: Headerline: icy-private:0
11:54:25.958 -> D: Headerline: icy-pub:1
11:54:25.958 -> D: Headerline: icy-url:http://www.magicoldiesflorida.com
11:54:25.958 -> D: Headerline: Server: Icecast 2.4.0-kh12
11:54:25.958 -> D: Headerline: Cache-Control: no-cache, no-store
11:54:25.958 -> D: Headerline: Access-Control-Allow-Origin: *
11:54:25.992 -> D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
11:54:25.992 -> D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
11:54:25.992 -> D: Headerline: Connection: Close
11:54:25.992 -> D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
11:54:25.992 -> D: Headerline: icy-metaint:16000
11:54:25.992 -> D: Switch to DATA, bitrate is 192, metaint is 16000
11:54:26.393 -> D: Streamtitle found, 34 bytes
11:54:26.393 -> D: StreamTitle='Donovan - Atlantis ';

Fabri54, 2022-05-22:
Last thing, in the configuration I've deleted the pin correspondence for I2S signals

Edzelf, 2022-05-22:
The backlight of your display is probable connected to 3.3 Volt instead of 5 Volt.  The SD is probably not configured in your config.h and the Chip Select pin for it is not configured in NVS.
I do not know what is wrong with the rotary encoder, but you should see click and rotation events in the serial debug output.

Fabri54, 2022-05-22:
The backlight is indeed connected to 3.3V but there were no luminosity issues before moving from V1 to V2.
The  SDCARD is defined in config.h
Pin 15 is assigned to sd_cs (only thing inherited from V1 is that the TFT and SD chip selects are the same signal, pin 15 (V1 worked only in that way)
I do see the results of rotating and clicking the rotary encoder but only mute works
12:18:21.803 -> D: Single click
12:18:21.803 -> D: Mute set to 1
12:18:33.417 -> D: Single click
12:18:33.417 -> D: Mute set to 0

Edzelf, 2022-05-22:
I mean the cs of the SD, not of the display.  The brightness of the display backlight does not depend on  software.  Rotary switch:  does the double click and triple click and maybe the long click work as programmed?

Fabri54, 2022-05-22:
The chip select of TFT and SD with V1 had to be connected to the same GPIO = 15  in order to have most of the basic functionalities. 
double click and triple click and maybe the long click are detected but no action follows
13:23:32.768 -> D: Triple click
13:23:32.768 -> D: No tracks on SD  (SD card is not detected and mounted but may be the cs in common with TFT)
13:23:38.120 -> D: Long click
13:23:38.120 -> D: Radiofuncs cmd is 2
13:23:38.290 -> D: Stopping client
13:23:49.001 -> D: Long click
13:23:49.001 -> D: Radiofuncs cmd is 2
What is the long click supposed to do?

Edzelf, 2022-05-22:
Good idea about the chip select, but it will not work :)
I will check the functionality of double-, triple- and long click later.  It should be in the document somewhere.

Fabri54, 2022-05-22:
It's weird, perhaps I was expecting a quicker reaction time but, the double click is now working: it allows to chose a station among the presets (double click, enter in the selection mode, rotate the encoder until you see the desired station on the display, click to confirm

Edzelf, 2022-05-23:
I have added some more info on the rotary encoder in chapter 12 of the document.

Fabri54, 2022-05-23:
Great! Thank you. 
In the meantime I moved the SD card chip select to GPIO14 and now the SD is mounted and read (not all the times but I suspect the wiring and the breadboard being the culprits of some hiccups). With the triple click I can access the selection of files and pick one for playback. Web interface seems to work as well.

closed 2022-05-23 by Fabri54