Edzelf/ESP32Radio-V2 #51: @ issues... only one error

CdnLaser, 2022-05-24:
First I apologize, I'm NOT a software guy, so i'm a bit lost with the software. I am using PlatformIO and the compiling mostly went ok.
I have a problem with the VS1023 when compiled, I will include the dump.
The second problem is the NEXTION display, it only shows the 'demo mode' of the Nextion and I'm not sure how to get around it. I have 5 v and Ground or the display wouldnt work at at and I have the TX going to Pin 16 of the ESP32 and the TX going to Pin 17 of the ESP32. The Pins are assigned properly (or appear to be looking at the code.

Here is the VS1023 errors (almost at the bottom .

Any assistance would be appreciated for this great looking project.

Cheers

[Starting] Opening the serial port - COM8
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

D: Starting ESP32-radio running on CPU 1 at 240 MHz.
D: Version Sun, 15 May 2022 10:30:00 GMT.  Free memory 110580
D: Display type is NEXTION
D: SPIFFS is okay, space 1378241, used 33634
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 48 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: Pin 5 is already reserved!
D: pin_i2s_bck set to 5
D: Pin 32 is already reserved!
D: pin_i2s_lck set to 32
D: Pin 4 is already reserved!
D: pin_i2s_din set to 4
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is HIGH
D: GPIO39 is HIGH
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Enable pin 35 for IR
D: Start NEXTION display
D: Init Nextion, pins 16, 17
D: Nextion command cls BLACK
D: Nextion command page 0
D: NEXTION output 'Starting......' to 1
D: NEXTION output 'Sun, 15 May 2022' to 2
D: NEXTION output 'By Ed Smallenburg' to 3
D: Create list with acceptable WiFi networks
D: Added MYWIFI to list of networks
D: Added SSID1 to list of networks
D: Added SSID2 to list of networks
D: Scan Networks
: Scan completed
D: Number of available networks: 7
D:  1 - Spare Hub                 Signal: -67 dBm, Encryption WPA_WPA2_PSK, 
D:  2 - New_Setup                 Signal: -75 dBm, Encryption WPA2_PSK, 
D:  3 - NewSetup2_guest           Signal: -77 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - NTGR_VMB_1473370562       Signal: -78 dBm, Encryption WPA2_PSK, 
D:  5 - MK                        Signal: -91 dBm, Encryption WPA2_PSK, 
D:  6 - Worldline4062B7           Signal: -91 dBm, Encryption WPA_WPA2_PSK, 
D:  7 - BELL826                   Signal: -95 dBm, Encryption WPA2_PSK, 
D: End of list
D: updateNr 0 <= 0 to 6, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to network
D: NEXTION output 'Connect to network' to 4

Edzelf, 2022-05-24:
First note that pins 5, 32 and 4 are double defined.  You have to correct that in the config page of the webinterface.  Then you have to use the Nextion editor to program the Nextion outside of the ESP32.

CdnLaser, 2022-05-24:
Thank you Edzelf for the fast response!!!
Within the web interface, do I need to redefine those pins or can I just '#' infront of them?
Also with the display, me being the software dummy I am, saw the TFT and the HMI file in the nextion directory and thought they would get compiled within the build... oops. Thanks for explaining :)

Cheers!!!

CdnLaser, 2022-05-24:
Awesome... will give it all a go this evening !!

Cheers,
Russ

closed 2022-05-25 by CdnLaser