Edzelf/ESP32Radio-V2 #53: WiFi & Bluetooth

wa8lbz, 2022-05-28:
I am building the ESP32 Radio V2 project but without using internal speakers or ear phones. My radio consists of a EZSBC ESP32 board; PCM5102A I2S MP3 decoder board; VS1838B IR receiver board; Nextion NX4024T032 display unit; and KCX_BT_EMITTER Bluetooth transmitter board. My idea is to control the radio with an IR remote control and transmit audio to external speakers and earphones by way of Bluetooth transmission.

I chose the EZSBC ESP32 board because it can be powered by applying +5V to its VIN pin. With some of the ESP32 DOIT DEVKT V1 boards, this cannot be done. The +5V has to be applied through the USB connector on the board. This was not convenient for me.

All of the above mentioned boards have been bread boarded and are working as expected. But after a short period of time (1 to 5 minutes) of operation the radio stops working. I discovered by troubleshooting that the Bluetooth transmitter is causing the problem. When the Bluetooth transmitter is disconnected, the radio will play for hours with no problem.

It turns out that WiFi and Bluetooth frequencies are in the same band. WiFi 2.3995 to 2.4845 GHz and Bluetooth 2.4000 to 2.4835 GHz. I suspect my problem is do to interference between the WiFi and Bluetooth signals.

I posted this to find out is anyone else is using a Bluetooth transmitter in their ESP32 Radio V2 project and if they are experiencing the problem I described above. Any comments will be greatly appreciated.

tsctrl, 2022-05-29:
you can apply up to 12volt to the esp vcc pin right?

wa8lbz, 2022-05-29:
No. The EZSBC.com website says the voltage applied to the VIN pins should be no higher than +8VDC.

wa8lbz, 2022-05-29:
Apparently, my WiFi and Bluetooth coexistence problem is solved. The problem was a laying on the antenna of the Bluetooth transmitter board. My guess is that the wire de-tuned the Bluetooth transmitter to the point that it interfered with the WiFi signal. After removing the wire, the radio plays for extended periods of time with no interruptions.

WerkAG, 2022-10-10:
I'm also using ESP32 Radio V2  and the KCX_BT_EMITTER on a custom designed PCB (which can be fitted with the Bluetooth TX or the PAM8403 amplifier).  It works for days without issues.

[image]

...and the end result in a 3D printed box:

[image]

WerkAG, 2022-11-09:
Soon I will make available on GitHub the gerber files of a new PCB with support for the bluetooth KCX_BT_EMITTER module and/or the PAM8403 amplifier. 

[image]

[image]