Edzelf/ESP32Radio-V2 #58: Doesn't connect to any WiFi if multiple access points are available

Fabri54, 2022-06-21:
Hi, I've eventually built and put into a wooden container the ESP32 radio. It's mostly functional (some hiccups when accessing the SD card)  but it shows an unexpected behaviour when it's located in a room where the two WiFi networks I've at home overlap themselves (different names, different passwords, of course).
In a two-three meter range from the AP the ESP32 Radio connects to the closest WiFi but when it's placed further it doesn't connect neither to the still physically closest AP nor to the other one (physically closest may not necessarily means a stronger signal level/quality) which is further, and it reverts to creating its own AP.
Has anybody encountered a similar problem and/or know if there is a solution to it?
Thank you

thb584Git, 2022-08-12:
On testing my ESP32-radio pcb with adding several changes to the configuration I ended up with problems with Wifi and streaming dropouts. Got this fixed be erasing flash and completly new programming. Maybe you give it a try - good luck

Fabri54, 2022-08-13:
Many thanks for the suggestion but I'm afraid it's not going to work. I've built a first prototype which has seen multiple installations of the SW but the final version has been implemented on a brand new ESP32, completely clean.
I may give a try but not in the near future. In case I'll post the results. 
Have a good day