Edzelf/ESP32Radio-V2 #59: Unable to build a second ESP32 Radio

Fabri54, 2022-06-24:
Hi,
it's with a lot of shame that I'm writing this post. 
I've built a first ESP32 Radio V2, boxed it and it is now playing on the background  while I'm typing. It has been built copying the prototype breadboard into a more structured and clean one. IR, Rotary, SD (Red TFT)  are  all working and I simply transplanted the ESP32 - WROOM - 32 from the protoboard to the final one. Actually I have used a slightly better audio amp and, with a proper single power supply (9V/3A + 7805 to provide 5V to ESP32), I didn't even need the decoupling audio transformers to get a more than decent audio.
However the point here is that I'm now playing with the protoboard (not a single wire has been changed) but I cannot get the ESP32 Radio working anymore. I've tried two ESP32 modules. I've erased all the flash with the esptool.py.  I've written the content of the Data folder into the SPIFFS and I've checked that the files are there and read the defaultprefs.txt. It contains exactly what is needed and it's a copy  of the one which is running on the working model but, whatever I do I get the following output on the serial monitor (using Arduino IDE 1.8.20)

12:36:27.313 -> ⸮⸮ ⸮ ⸮⸮⸮ ⸮@Q*R⸮⸮VVHh⸮⸮}⸮⸮⸮⸮0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:36:27.313 -> mode:DIO, clock div:1
12:36:27.313 -> load:0x3fff0018,len:4
12:36:27.313 -> load:0x3fff001c,len:1216
12:36:27.313 -> ho 0 tail 12 room 4
12:36:27.313 -> load:0x40078000,len:10944
12:36:27.313 -> load:0x40080400,len:6388
12:36:27.313 -> entry 0x400806b4
12:36:30.620 -> 
12:36:30.620 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.
12:36:30.620 -> D: Version Sun, 15 May 2022 10:30:00 GMT.  Free memory 113792
12:36:30.653 -> D: Display type is BLUETFT
12:36:30.722 -> D: SPIFFS is okay, space 1378241, used 33634
12:36:30.756 -> D: Found partition 'nvs     ' at offset 0x009000 with size    20480
12:36:30.756 -> D: Found partition 'otadata ' at offset 0x00E000 with size     8192
12:36:30.756 -> D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
12:36:30.756 -> D: Read 6 keys from NVS
12:36:30.756 -> D: pin_spi_sck set to 18
12:36:30.756 -> D: pin_spi_miso set to 19
12:36:30.756 -> D: pin_spi_mosi set to 23
12:36:30.756 -> D: GPIO0 is HIGH
12:36:30.789 -> D: GPIO2 is HIGH
12:36:30.789 -> D: GPIO4 is HIGH
12:36:30.789 -> D: GPIO5 is HIGH
12:36:30.789 -> D: GPIO12 is HIGH
12:36:30.822 -> D: GPIO13 is HIGH
12:36:30.822 -> D: GPIO14 is HIGH
12:36:30.822 -> D: GPIO15 is HIGH
12:36:30.857 -> D: GPIO16 is HIGH
12:36:30.857 -> D: GPIO17 is HIGH
12:36:30.857 -> D: GPIO18 is HIGH
12:36:30.895 -> D: GPIO19 is HIGH
12:36:30.895 -> D: GPIO21 is HIGH
12:36:30.895 -> D: GPIO22 is HIGH
12:36:30.895 -> D: GPIO23 is HIGH
12:36:30.929 -> D: GPIO25 is HIGH
12:36:30.929 -> D: GPIO26 is HIGH
12:36:30.929 -> D: GPIO27 is HIGH
12:36:30.929 -> D: GPIO32 is HIGH
12:36:30.964 -> D: GPIO33 is HIGH
12:36:30.964 -> D: GPIO34 is LOW, probably no PULL-UP
12:36:30.964 -> D: GPIO35 is HIGH
12:36:30.998 -> D: GPIO39 is LOW, probably no PULL-UP
12:36:30.998 -> D: Start BLUETFT display
12:36:31.761 -> D: Create list with acceptable WiFi networks
12:36:31.761 -> D: Added Vodafone-34350430-2p4 to list of networks
12:36:32.895 -> D: Scan Networks
12:36:34.997 -> D: Scan completed
12:36:34.997 -> D: Number of available networks: 12
12:36:34.997 -> D:  1 - FASTWEB-UF6NAC_EXT        Signal: -81 dBm, Encryption WPA2_PSK, 
12:36:34.997 -> D:  2 - Vodafone-C01400156_EXT    Signal: -82 dBm, Encryption WPA2_PSK, 
12:36:35.032 -> D:  3 - CASA DEL GATTO            Signal: -88 dBm, Encryption WPA2_PSK, 
12:36:35.032 -> D:  4 - WOW FI - FASTWEB          Signal: -88 dBm, Encryption ????, 
12:36:35.032 -> D:  5 - FASTWEB-UF6NAC            Signal: -88 dBm, Encryption WPA2_PSK, 
12:36:35.032 -> D:  6 - Vodafone-A85617501        Signal: -88 dBm, Encryption WPA2_PSK, 
12:36:35.032 -> D:  7 - Vodafone-WiFi             Signal: -88 dBm, Encryption OPEN, 
12:36:35.068 -> D:  8 - WOW FI - FASTWEB          Signal: -89 dBm, Encryption ????, 
12:36:35.068 -> D:  9 - HOLLY-WIFI                Signal: -91 dBm, Encryption WPA2_PSK, 
12:36:35.068 -> D: 10 - Mesh_B10                  Signal: -92 dBm, Encryption WPA2_PSK, 
12:36:35.068 -> D: 11 - TIM-18580995              Signal: -94 dBm, Encryption WPA2_PSK, 
12:36:35.068 -> D: 12 - NG_BIAN                   Signal: -96 dBm, Encryption WPA2_PSK, 
12:36:35.068 -> D: End of list
12:36:35.068 -> D: updateNr 0 <= 0 to 0, relative is 0
12:36:35.103 -> D: updateNr result is 0
12:36:35.103 -> D: nextPreset is 0
12:36:35.211 -> D: Slow SPI, Testing VS1053 read/write registers...
12:36:35.211 -> D: VS1053 not properly installed!
12:36:35.314 -> D: Connect to network
12:36:37.334 -> D: Try WiFi Vodafone-34350430-2p4
12:36:39.538 -> D: WiFi Failed!  Trying to setup AP with name ESP32-Radio and password ESP32-Radio.
12:36:39.538 -> D: IP = 1.0.0.0
12:36:39.538 -> D: Start web server
12:36:40.554 -> D: Rotary encoder is disabled (-1/-1/-1)
12:36:42.545 -> D: updateNr 0 <= 0 to 0, relative is 0
12:36:42.580 -> D: updateNr result is 0
12:36:42.580 -> D: nextPreset is 0

It's driving me mad. Any suggestion where to look for  the cause of this unexpected behaviour?
Many thanks

Fabri54, 2022-06-25:
The ESP32 Radio now works but I had to connect to its AP and insert the entire content of the "defaultprefs.txt" file with a copy and paste operation in the Config page of the web server. Actually there was no configuration data with the exception of the first six lines.
I do not understand this behaviour. Have I done something wrong erasing the whole flash with the esptool.py ? When I write the content of the data folder to the SPIFFS the operation ends well and, as written, I can then read the list of files inside the SPIFFS as well as read what is inside each file.
The module has  the following configuration " ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 926100".
Please help to understand what I'm doing wrong

Edzelf, 2022-06-25:
You followed the correct procedure.  There should be a button on the config page to load the defaultprefs.txt file directly.
There is also an editable default network definition in config,h to directly connect in STA mode,  The pins however must be configured in the config page.

Fabri54, 2022-06-25:
Thank you but I must be missing something. 
The defaultprefs.txt file is first written into the SPIFFS (it is there, I can read it)  then when I load and run the ESP32Radio.ino  I expect the program to read the txt file and use it at boot , while when I look at Config using the web access to ESP32-Radio AP it is practically empty. Why?

Here below is the defaultprefs.txt file as it is written into the SPIFFS using Arduino IDE. In addition if in Config.h I insert into #define FIXEDWIFI "MYWIFI/password" one of my wifi networks, at boot  it doesn't connect to it and reverts to ESP32-Radio

# Example configuration
# Programmable input pins:
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
gpio_17 = resume
gpio_34 = station = icecast.omroep.nl:80/radio1-bb-mp3
#
# MQTT settings
mqttbroker = none
mqttport = 1883
mqttuser = none
mqttpasswd = none
mqttprefix = none
# Enter your WiFi network specs here:
wifi_00 = Vodafone-34350430-2p4/************
wifi_01 = NG_BIAN/************
#
volume = 72
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
preset = 0
preset_00 = 101smoothjazz.com/101-smoothjazz.m3u     #  0 - Smooth Jazz, California
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIA WWW.SHERADIO.COM
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
 
# Clock offset and daylight saving time
clk_server = pool.ntp.org                            # Time server to be used
clk_offset = 1                                       # Offset with respect to UTC in hours
clk_dst = 1                                          # Offset during daylight saving time (hours)
#
# IR codes
ir_629D = upvolume = 2
ir_A857 = downvolume = 2
ir_E21D = mute
ir_E01F = downpreset = 1
ir_906F = uppreset = 1
ir_30CF = preset = 1
ir_18E7 = preset = 2
ir_7A85 = preset = 3
ir_10EF = preset = 4
ir_38C7 = preset = 5
ir_5AA5 = preset = 6
ir_42BD = preset = 7
ir_4AB5 = preset = 8
ir_52AD = preset = 9  
#
# GPIO pinnings
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
#
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
#
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
#
pin_vs_cs = 5                                           # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 32                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
#
pin_sd_cs = 14					  # GPIO Pin number for SD card CS"
#

I don't want to waste your time as, at the end of the day, I have two working ESP32-radio, the official one inside a wooden case and the prototype, but I'd really understand If I'm doing something wrong and, of course, what.
Thank you and have a nice weekend

Edzelf, 2022-06-25:
No, the defaultprevs.txt is not read at start-up.  Only used for filling the config page on a button click.

Fabri54, 2022-06-25:
OK, eventually I understand. Thank you again.
One last thing, if you are interested I can provide the schematic of the power supply,  the picture and  the link of the power amp I'm using and that didn't require me to insert the two audio transformers to avoid the digital noise into the audio

closed 2022-07-20 by Fabri54