Edzelf/ESP32Radio-V2 #60: VS1053 not properly installed!

LYMarvin, 2022-06-27:
Hello! Thank you for your work. 
I use this version:
https://github.com/Edzelf/ESP32-Radio，VERSION     "Mon, 25 Mar 2020 10:45:00 GMT"
Network radio and MP3 play normally.Only the display is normal when using ILI9341, and the NEXTION  cannot be compiled.

The same development board.
I use this version:
https://github.com/Edzelf/ESP32Radio-V2 , VERSION     "Sun, 15 May 2022 10:30:00 GMT"

NEXTION  display is normal,but" VS1053 not properly installed!"
The serial port monitoring results are as follows:

D: Starting ESP32-radio running on CPU 1 at 240 MHz.
D: Version Sun, 15 May 2022 10:30:00 GMT.  Free memory 113792
D: Display type is NEXTION
D: SPIFFS is okay, space 1378241, used 0
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 6 keys from NVS
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Start NEXTION display
D: Init Nextion, pins 16, 17
D: Nextion command cls BLACK
D: Nextion command page 0
D: NEXTION output 'Starting......' to 1
D: NEXTION output 'Sun, 15 May 2022' to 2
D: NEXTION output 'By Ed Smallenburg' to 3
D: Create list with acceptable WiFi networks
D: Added TP-LINK_0562 to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 30
D:  1 - TP-LINK_0562              Signal: -63 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  2 - FAST_CDEF                 Signal: -80 dBm, Encryption WPA_WPA2_PSK, 
D:  3 - JCG-635400-2.4G           Signal: -81 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - CMCC-401                  Signal: -86 dBm, Encryption WPA_WPA2_PSK, 
D:  5 - 666                       Signal: -88 dBm, Encryption WPA_WPA2_PSK, 
D:  6 - TP-LINK_927B              Signal: -89 dBm, Encryption WPA_WPA2_PSK, 
D:  7 - 2003                      Signal: -89 dBm, Encryption WPA_WPA2_PSK, 
D:  8 - Tenda_447C58              Signal: -89 dBm, Encryption WPA_PSK, 
D:  9 - TP-LINK_C94A.             Signal: -90 dBm, Encryption WPA_WPA2_PSK, 
D: 10 - CMCC-2m5b                 Signal: -90 dBm, Encryption WPA_WPA2_PSK, 
D: 11 - TP-LINK_49E778            Signal: -90 dBm, Encryption WPA_WPA2_PSK, 
D: 12 - TP-LINK_EF65              Signal: -90 dBm, Encryption WPA_WPA2_PSK, 
D: 13 - TP-LINK_61BE              Signal: -91 dBm, Encryption WPA_WPA2_PSK, 
D: 14 - Xiaomi_01E1               Signal: -91 dBm, Encryption WPA_WPA2_PSK, 
D: 15 - HUAWEI-3MFNG7             Signal: -92 dBm, Encryption WPA2_PSK, 
D: 16 - FAST_1482                 Signal: -92 dBm, Encryption WPA_WPA2_PSK, 
D: 17 - CMCC-U9Pb                 Signal: -92 dBm, Encryption WPA_WPA2_PSK, 
D: 18 - CU_63fr                   Signal: -92 dBm, Encryption WPA2_PSK, 
D: 19 - CU_yc6b                   Signal: -92 dBm, Encryption WPA2_PSK, 
D: 20 - CMCC-N2tf                 Signal: -93 dBm, Encryption WPA_WPA2_PSK, 
D: 21 - TP-LINK_9B28              Signal: -94 dBm, Encryption WPA_WPA2_PSK, 
D: 22 - CMCC-bbPj                 Signal: -94 dBm, Encryption WPA_WPA2_PSK, 
D: 23 - TP-LINK_EDB5              Signal: -94 dBm, Encryption WPA_WPA2_PSK, 
D: 24 - 2503                      Signal: -95 dBm, Encryption WPA_WPA2_PSK, 
D: 25 - @PHICOMM_F4               Signal: -95 dBm, Encryption WPA_WPA2_PSK, 
D: 26 - CMCC-G6Ah                 Signal: -95 dBm, Encryption WPA_WPA2_PSK, 
D: 27 - 十全十美              Signal: -95 dBm, Encryption WPA_WPA2_PSK, 
D: 28 - FAST_B8D4                 Signal: -96 dBm, Encryption WPA_WPA2_PSK, 
D: 29 - FAST_4492                 Signal: -97 dBm, Encryption WPA_WPA2_PSK, 
D: 30 - TP-LINK_77EE              Signal: -97 dBm, Encryption WPA_WPA2_PSK, 
D: End of list
D: updateNr 0 <= 0 to 0, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to network
D: NEXTION output 'Connect to network' to 4
D: Try WiFi TP-LINK_0562
D: NEXTION output 'SSID = TP-LINK_0562' to 5
D: SSID = TP-LINK_0562
D: IP = 192.168.1.105
D: NEXTION output 'IP = 192.168.1.105' to 6
D: Nextion command page 1
D: Start web server
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 11:51:22
D: updateNr 0 <= 0 to 0, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: NEXTION command seen BE ??"Starting......"
D: NEXTION command seen 74 2.txt="Sun, 15 May 2022"s%??
D: Radiofuncs cmd is 1
D: Connect to host 
D: Request  failed!
D: Nextion t0.txt="ESP32-Radio"
D: Nextion t1.txt=""

Looking forward to your support, thank you very much!

[Boot serial port monitoring information.txt](https://github.com/Edzelf/ESP32Radio-V2/files/8990992/Boot.serial.port.monitoring.information.txt)

Edzelf, 2022-06-27:
You should at least define the pins for the VS1053 in the config page of the web interface.

LYMarvin, 2022-06-28:
Serial port display IP address is: IP = 192.168.1.105
However, the web interface cannot be opened when accessing this address！

LYMarvin, 2022-06-28:
accessing this address，display "sorry"!

Edzelf, 2022-06-28:
Did you upload the contents of the data directory to SPIFFS?

LYMarvin, 2022-06-29:
> Did you upload the contents of the data directory to SPIFFS?

I compiled it with Arduino IDE，Use the same method to upload the program：https://github.com/Edzelf/ESP32-Radio   "Mon, 25 Mar 2020 10:45:00 GMT"
Failed to understand SPIFs.

Edzelf, 2022-06-29:
Google is your friend.  See for instance https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/.

LYMarvin, 2022-07-28:
defaultprefs.txt

> Google is your friend. See for instance https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/.

I used the "esp32 sketch data upload" tool once, and Arduino also uploaded it, but it didn't work.
What files have been uploaded by the "esp32 sketch data upload" tool? Is it just "defaultprefs.txt"? thank you!

Edzelf, 2022-07-28:
Upload is for all files in the data sub directory.  During upload the files are listed.  And I think the files are also listed in the debug output of the sketch.

LYMarvin, 2022-07-30:
> 
**I still haven't solved this problem     sad**
**After writing the file in data to the SPIFs file, the serial port monitoring results are as follows:**
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

**Debug output:**

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.
D: Version Sun, 15 May 2022 10:30:00 GMT.  Free memory 113792
D: Display type is ILI9341
D: SPIFFS is okay, space 1378241, used 32881
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 26
D: pin_enc_dt set to 25
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_sd_cs set to 14
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 13
D: pin_vs_dreq set to 4
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is LOW, probably no PULL-UP
D: GPIO26 is HIGH
D: GPIO27 is LOW, probably no PULL-UP
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_32 will execute upvolume = 2
D: gpio_33 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start ILI9341 display
D: Create list with acceptable WiFi networks
D: Added TP-LINK_0562 to list of networks
D: Added TP-LINK_0562 to list of networks
D: Added TP-LINK_2B26 to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 16
D:  1 - TP-LINK_0562              Signal: -59 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  2 - JCG-635400-2.4G           Signal: -75 dBm, Encryption WPA_WPA2_PSK, 
D:  3 - CU_9xyQ                   Signal: -83 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - 666                       Signal: -84 dBm, Encryption WPA_WPA2_PSK, 
D:  5 - 3433ea32                  Signal: -85 dBm, Encryption WPA_WPA2_PSK, 
D:  6 - hy                        Signal: -86 dBm, Encryption WPA_WPA2_PSK, 
D:  7 - Xiaomi                    Signal: -88 dBm, Encryption WPA2_PSK, 
D:  8 - CU_63fr                   Signal: -88 dBm, Encryption WPA2_PSK, 
D:  9 - TP-LINK_49E778            Signal: -89 dBm, Encryption WPA_WPA2_PSK, 
D: 10 - CMCC-N2tf                 Signal: -89 dBm, Encryption WPA_WPA2_PSK, 
D: 11 - TOTOLINK                  Signal: -90 dBm, Encryption WPA_WPA2_PSK, 
D: 12 - FAST_B8D4                 Signal: -90 dBm, Encryption WPA_WPA2_PSK, 
D: 13 - yaohao                    Signal: -91 dBm, Encryption WPA_WPA2_PSK, 
D: 14 - HUAWEI-yuan               Signal: -91 dBm, Encryption WPA2_PSK, 
D: 15 - QMM0526                   Signal: -93 dBm, Encryption WPA_WPA2_PSK, 
D: 16 - 2503                      Signal: -94 dBm, Encryption WPA_WPA2_PSK, 
D: End of list
D: updateNr 0 <= 0 to 8, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: Connect to network
D: SSID = TP-LINK_0562
D: IP = 192.168.1.104
D: Start web server
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 19:30:13
D: SD okay, start scan
D: Locate mp3 files on SD, may take a while...
D: Space 143832
D: 14 tracks on SD
D: updateNr 0 <= 11 to 8, relative is 0
D: updateNr result is 8
D: nextPreset is 8
D: Radiofuncs cmd is 1
D: SDfuncs cmd is 2
D: Connect to host skonto.ls.lv:8002/mp3
D: Song stopped correctly after 0 msec

** click"VOL-   Volume is now 2  no more"**

Edzelf, 2022-07-30:
Most of the functions seem to work now.  Program crashes as soon as the mp3 stream starts.  Check you supply voltage.