Edzelf/ESP32Radio-V2 #63: No keys found in NVS partition

Gooser5005, 2022-08-10:
Hi there!

First of all, thank you very much for your great work! 

I'm facing an issue regarding the NVS data. After uploading the firmware and the data directory the monitor output shows, that no keys could have been read from the NVS partition. As a result there is no proper pin assignment and all connected devices TFT 1.8", VS1053, rotary knob and IR-receiver don't work.

I'm using the most recent version from May, 5th 2022 in PlatformIO. I already tried to switch platform from espressif32 to espressif32@3.1.0 but with no success. 

There is no issue with NVS data in V1 of ESP32Radio.  

Is there anything I can do to isolate the cause of the error?

Many thanks for your work and your support!

Kind regards
Gerd

Below please find the monitor output from both platform settings espressif32 and espressif32@3.1.0

espressif32@3.1.0
=================
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.
D: Version Sun, 15 May 2022 10:30:00 GMT.  Free memory 113792
D: Display type is BLUETFT
D: SPIFFS is okay, space 1378241, used 33634
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 0 keys from NVS
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: Start BLUETFT display
D: Create list with acceptable WiFi networks
D: Added xxx to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 7
D:  1 - Mxxx
D:  2 - Sxxx
D:  3 - Mxxx
D:  4 - Mxxx
D:  5 - Sxxx
D:  6 - Sxxx
D:  7 - Fxxx
D: End of list
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to network
D: Try WiFi Sxxx
D: SSID = xxx
D: IP = xxx.xxx.xxx.xxx
D: Start web server
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 20:42:52
D: nvs_get_str failed 1102 for key preset, keylen is 6, len is 150!
D: Contents:
D: updateNr 0 <= 0 to 0, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: Radiofuncs cmd is 1
D: Connect to host
D: Request  failed!
...

espressif32
===========
D: Starting ESP32-radio running on CPU 1 at 240 MHz.
D: Version Sun, 15 May 2022 10:30:00 GMT.  Free memory 110580
D: Display type is BLUETFT
D: SPIFFS is okay, space 1378241, used 33634
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 0 keys from NVS
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: Start BLUETFT display
E (3444) gpio: gpio_set_level(226): GPIO output gpio_num error
E (3445) gpio: gpio_set_level(226): GPIO output gpio_num error
E (3448) gpio: gpio_set_level(226): GPIO output gpio_num error
...
... (infinite loop with "E (xxxx) gpio: gpio_set_level(226): GPIO output gpio_num error" messages)

Edzelf, 2022-08-10:
So your NVS is empty.  Use the webserver to edit the settings in the config page.  You may set the defaults first with the "Default" button, then edit the settings.  See document chapter 2.2 for examples.

In config.h you can set the credentials for a default WiFi network if you do not want to use AP mode of WiFi.

Gooser5005, 2022-08-10:
I did it the same way you advised and now it works perfectly fine :).

Thanks a lot Ed!

Gooser5005, 2022-08-10:
Closing this issue

closed 2022-08-10 by Gooser5005