Edzelf/ESP32Radio-V2 #64: VS1053B - Possible wrong detection?

WerkAG, 2022-08-30:
I have a VS1053/1003 module with a VS1053B chip. If I force the result of the testComm routine to be true, the module works well and I get sound with MP3 and AAC streams. 

Could it be that the VS1053B isn't being correctly detected? 

Bellow are the errors show on serial monitor.

```
20:39:58.527 -> D: Slow SPI, Testing VS1053 read/write registers...
20:39:58.527 -> D: VS1053 SPI error. SB:012C R1:0000 R2:0000
20:39:58.527 -> D: VS1053 SPI error. SB:0258 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:0384 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:04B0 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:05DC R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:0708 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:0834 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:0960 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:0A8C R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:0BB8 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:0CE4 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:0E10 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:0F3C R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:1068 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:1194 R1:0000 R2:0000
20:39:58.667 -> D: VS1053 SPI error. SB:12C0 R1:0000 R2:0000
20:39:58.687 -> D: VS1053 SPI error. SB:13EC R1:0000 R2:0000
20:39:58.727 -> D: VS1053 SPI error. SB:1518 R1:0000 R2:0000
20:39:58.727 -> D: VS1053 SPI error. SB:1644 R1:0000 R2:0000
20:39:58.727 -> D: VS1053 SPI error. SB:1770 R1:0000 R2:0000
20:39:58.767 -> D: This is not a VS1053, but a VS1001 instead!
20:39:58.847 -> D: Fast SPI, Testing VS1053 read/write registers again...
20:39:58.847 -> D: VS1053 SPI error. SB:0003 R1:0000 R2:0000
20:39:58.887 -> D: VS1053 SPI error. SB:0006 R1:0000 R2:0000
20:39:58.887 -> D: VS1053 SPI error. SB:0009 R1:0000 R2:0000
20:39:58.887 -> D: VS1053 SPI error. SB:000C R1:0000 R2:0000
20:39:58.887 -> D: VS1053 SPI error. SB:000F R1:0000 R2:0000
20:39:58.927 -> D: VS1053 SPI error. SB:0012 R1:0000 R2:0000
20:39:58.927 -> D: VS1053 SPI error. SB:0015 R1:0000 R2:0000
20:39:58.927 -> D: VS1053 SPI error. SB:0018 R1:0000 R2:0000
20:39:58.927 -> D: VS1053 SPI error. SB:001B R1:0000 R2:0000
20:39:58.967 -> D: VS1053 SPI error. SB:001E R1:0000 R2:0000
20:39:58.967 -> D: VS1053 SPI error. SB:0021 R1:0000 R2:0000
20:39:58.967 -> D: VS1053 SPI error. SB:0024 R1:0000 R2:0000
20:39:58.967 -> D: VS1053 SPI error. SB:0027 R1:0000 R2:0000
20:39:59.007 -> D: VS1053 SPI error. SB:002A R1:0000 R2:0000
20:39:59.007 -> D: VS1053 SPI error. SB:002D R1:0000 R2:0000
20:39:59.007 -> D: VS1053 SPI error. SB:0030 R1:0000 R2:0000
20:39:59.007 -> D: VS1053 SPI error. SB:0033 R1:0000 R2:0000
20:39:59.047 -> D: VS1053 SPI error. SB:0036 R1:0000 R2:0000
20:39:59.047 -> D: VS1053 SPI error. SB:0039 R1:0000 R2:0000
20:39:59.047 -> D: VS1053 SPI error. SB:003C R1:0000 R2:0000
20:39:59.047 -> D: This is not a VS1053, but a VS1001 instead!
```

Here is my config:

```
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqttbroker = none
mqttpasswd = none
mqttport = 1883
mqttprefix = none
mqttuser = none
#
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_tft_blx = 33
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 32                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
```

pin_vs_dcs works on pin 32 and 16 (tried both)

Edzelf, 2022-08-31:
Please define (uncomment) the line with VS1003 in include/config.h, neer line 18.
//#define DEC_VS1003                                      // Hardware decoder for MP3

WerkAG, 2022-08-31:
Firstly many thanks for your work on the wonderful ESP32 Radio software, and for your quick answer.

Already tried with #define DEC_VS1053 and #define DEC_VS1003 (one at a time, of course). As said, the radio works perfectly, and I have sound on the output of the VS1053 module, for that I just have to force the  testComm() routine to return true (okay = true; ). Just wondering why the registers aren't been read (all return zero)?  

For testing I wrote a small code using another VS1053 library. Despite I managed to make it play audio streams, I couldn't also read anything else from the registers than zero. So whatever the issue is, it doesn't just happen with the ESP32Radio software.  I will continue to investigate.

Edzelf, 2022-09-01:
The testcomm function is not reading all registers, but tries to write to the volume register and read is back.  So it seems that reading the volume register does not work in your case.  If this is a problem, replace the VS1053 module.

WerkAG, 2022-09-09:
Thanks for your answer. I understand what the testcomm function does. Despite the radio is working well I will replace the VS1053 module as soon as I get a new one.
I liked so much this project that I designed a PCB for it. Today I received the first prototype, just finished assembling it and its working. 

[image]

Zero strange noises on the speakers :-) Instead of transformers I used something similar to what is described here:
https://esp8266-server.de/Internetradio.html

trip5, 2023-08-04:
> I designed a PCB for it.

OMG that looks simple and good.  Any chance you got this through PCBWay?  Would you share?  There's more than a few requests around these forums for PCBs that work.

WerkAG, 2023-08-05:
Of course I share it. You can find the gerber files here: https://github.com/WerkAG/ESP32Radio_PCB_v3.1

chainsawdave, 2024-01-14:
@WerkAG Did you solve your no sound issue? Also, how do you force the test com routine? I have tried two VS1053 boards from different sources and both give the same error. SPI error. SB:1770 R1:0000 R2:0000

aly-fly, 2024-01-14:
Did you try to lower the SPI clock frequency, verified good GND connection, reduced SPI wires length?

trip5, 2024-01-14:
Is this still going?  Actually I had an issue with an Esplora 1.8 inch TFT that caused the VS1053 to not be detected correctly.

Disconnecting the display allowed it to pass and I tried various solutions but in the end I just went to this line:

https://github.com/Edzelf/ESP32Radio-V2/blob/6c4c4921fef1cbc745fee9511596c4402b8e3a02/lib/codecs/src/VS1053.cpp#L163

And added `okay = true ;` and it worked fine... ish.

Went with another display anyways... The Esplora display isn't stable.

chainsawdave, 2024-01-14:
> Did you try to lower the SPI clock frequency, verified good GND connection, reduced SPI wires length?

Thanks for the suggestion. So I found two wires with poor continuity. I fixed  this and the errors (not VS1053, VS1005) has gone. Now I don't see any errors but still no sound

Edzelf, 2024-01-15:
Show the full logging.

chainsawdave, 2024-01-15:
> Show the full logging.

Here is what I can see.
[ 10931][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 7
[ 10932][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 10936][I][main.cpp:867] listNetworks(): [main] Scan completed
[ 10941][I][main.cpp:875] listNetworks(): [main] Number of available networks: 7
[ 10949][I][main.cpp:893] listNetworks(): [main]  1 - Devices                   Signal: -47 dBm, Encryption WPA2_PSK, Acceptable
[ 10960][I][main.cpp:893] listNetworks(): [main]  2 - Home network              Signal: -48 dBm, Encryption WPA2_PSK, Not accepttable
[ 10972][I][main.cpp:893] listNetworks(): [main]  3 - Devices                   Signal: -79 dBm, Encryption WPA2_PSK, Acceptable
[ 10983][I][main.cpp:893] listNetworks(): [main]  4 - Home network              Signal: -79 dBm, Encryption WPA2_PSK, Not accepttable
[ 10995][I][main.cpp:893] listNetworks(): [main]  5 - DIRECT-CB-HP              Signal: -87 dBm, Encryption WPA2_PSK, Not accepttable
[ 11006][I][main.cpp:893] listNetworks(): [main]  6 - Devices                   Signal: -89 dBm, Encryption WPA2_PSK, Acceptable
[ 11018][I][main.cpp:893] listNetworks(): [main]  7 - Home network              Signal: -90 dBm, Encryption WPA2_PSK, Not accepttable
[ 11029][I][main.cpp:897] listNetworks(): [main] End of list
[ 11060][I][main.cpp:966] nextPreset(): [main] nextPreset is 0
[ 11074][I][main.cpp:2717] setup(): [main] Connect to network
[ 11076][V][WiFiGeneric.cpp:343] _arduino_event_cb(): STA Stopped
[ 11078][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 3 - STA_STOP
[ 12097][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 12101][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[ 12102][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[ 19315][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 7
[ 19316][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 19321][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[ 19325][I][WiFiMulti.cpp:119] run(): [WIFI] 7 networks found
[ 19331][D][WiFiMulti.cpp:151] run():        0: [6][74:AC:B9:2D:9C:06] Home network (-48) *
[ 19339][D][WiFiMulti.cpp:149] run():  --->   1: [6][7A:AC:B9:2D:9C:06] Devices (-48) *
[ 19346][D][WiFiMulti.cpp:151] run():        2: [11][44:D9:E7:0B:2C:22] Home network (-79) *
[ 19355][D][WiFiMulti.cpp:149] run():  --->   3: [11][46:D9:E7:0B:2C:22] Devices (-81) *
[ 19362][D][WiFiMulti.cpp:149] run():  --->   4: [1][82:2A:A8:09:81:0D] Devices (-87) *
[ 19370][D][WiFiMulti.cpp:151] run():        5: [6][4A:9E:BD:6E:90:CB] DIRECT-CB-HP  (-87) *
[ 19378][D][WiFiMulti.cpp:151] run():        6: [1][80:2A:A8:09:81:0D] Home network (-88) *
[ 19386][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 7A:AC:B9:2D:9C:06 SSID: Devices Channel: 6 (-48)
[ 19397][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 20051][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: Devices, BSSID: 7a:ac:b9:2d:9c:06, Channel: 6, Auth: WPA2_PSK
[ 20052][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 20077][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:2.2.2.102
[ 20078][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 20081][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 2.2.2.102, MASK: 255.255.255.0, GW: 2.2.2.1
[ 20090][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[ 20095][D][WiFiMulti.cpp:175] run(): [WIFI] SSID: Devices
[ 20101][D][WiFiMulti.cpp:176] run(): [WIFI] IP: 2.2.2.102
[ 20106][D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 7A:AC:B9:2D:9C:06
[ 20112][D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 6
[ 20117][I][main.cpp:1638] connectwifi(): [main] SSID = Devices
[ 20122][I][main.cpp:1642] connectwifi(): [main] IP = 2.2.2.102
[ 20129][I][main.cpp:2727] setup(): [main] Start web server
[ 20135][I][main.cpp:2737] setup(): [main] Network found. Starting clients
[ 20143][I][main.cpp:2771] setup(): [main] MDNS responder started
[ 21148][I][main.cpp:2809] setup(): [main] Rotary encoder is enabled
[ 21149][I][main.cpp:4372] gettime(): [main] Sync TOD
[ 22819][I][main.cpp:4385] gettime(): [main] TOD synced
[ 22922][I][main.cpp:966] nextPreset(): [main] nextPreset is 1
[ 22923][I][main.cpp:3403] radiofuncs(): [main] Radiofuncs cmd is 1
[ 22924][I][main.cpp:1400] connecttohost(): [main] Connect to host airspectrum.cdnstream1.com:8114/1648_128
[ 22998][I][main.cpp:2541] onConnect(): [main] Connected to host at 144.217.252.222 on port 8114
[ 23034][I][main.cpp:1461] connecttohost(): [main] send GET command
[ 23090][I][main.cpp:3651] handlebyte_ch(): [main] Switch to HEADER
[ 23090][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: Content-Type: audio/mpeg
[ 23093][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-br:128
[ 23099][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
[ 23111][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-br:128
[ 23117][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-description:Today's Easy Hits from the 70s, 80s, 90s through Today!
[ 23129][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-genre:Easy Listening
[ 23137][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-name:Easy Hits Florida
[... cut ...]
[ 23220][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
[ 23229][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-metaint:16000
[ 23236][I][main.cpp:3742] handlebyte_ch(): [main] Switch to DATA, bitrate is 128, metaint is 16000
[ 23376][I][main.cpp:1294] showstreamtitle(): [main] Streamtitle found, 36 bytes
[ 23376][I][main.cpp:1295] showstreamtitle(): [main] StreamTitle='Madonna - Take a Bow ';
[ 50255][W][AsyncTCP.cpp:950] _poll(): rx timeout 4

Edzelf, 2024-01-15:
I'm interested in the start of the logging.

chainsawdave, 2024-01-15:
Is this right?

0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (512) esp_core_dump_flash: No core dump

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3054][I][main.cpp:2582] setup(): [main] Version Thu, 14 Dec 2023 09:00:00 GMT.  Free memory 110580
[  3056][I][main.cpp:2583] setup(): [main] Display type is DUMMY
[  3149][I][main.cpp:2593] setup(): [main] SPIFFS is okay, space 1016801, used 33383
[  3152][I][main.cpp:2614] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3155][I][main.cpp:2614] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3164][I][main.cpp:2614] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
[  3177][I][main.cpp:2486] fillkeylist(): [main] Read 48 keys from NVS
[  3189][I][main.cpp:1910] readIOprefs(): [main] pin_ir set to 35
[  3190][I][main.cpp:1910] readIOprefs(): [main] pin_enc_clk set to 25
[  3193][I][main.cpp:1910] readIOprefs(): [main] pin_enc_dt set to 26
[  3199][I][main.cpp:1910] readIOprefs(): [main] pin_enc_sw set to 27
[  3205][I][main.cpp:1910] readIOprefs(): [main] pin_tft_cs set to 15
[  3211][I][main.cpp:1910] readIOprefs(): [main] pin_tft_dc set to 2
[  3217][I][main.cpp:1910] readIOprefs(): [main] pin_vs_cs set to 5
[  3223][I][main.cpp:1910] readIOprefs(): [main] pin_vs_dcs set to 32
[  3229][I][main.cpp:1910] readIOprefs(): [main] pin_vs_dreq set to 4
[  3235][I][main.cpp:1910] readIOprefs(): [main] pin_spi_sck set to 18
[  3241][I][main.cpp:1910] readIOprefs(): [main] pin_spi_miso set to 19
[  3247][I][main.cpp:1910] readIOprefs(): [main] pin_spi_mosi set to 23
[  3264][I][main.cpp:2655] setup(): [main] GPIO0 is HIGH
[  3274][I][main.cpp:2655] setup(): [main] GPIO2 is LOW, probably no PULL-UP
[  3284][I][main.cpp:2655] setup(): [main] GPIO4 is LOW, probably no PULL-UP
[  3294][I][main.cpp:2655] setup(): [main] GPIO5 is HIGH
[  3304][I][main.cpp:2655] setup(): [main] GPIO12 is HIGH
[  3314][I][main.cpp:2655] setup(): [main] GPIO13 is HIGH
[  3324][I][main.cpp:2655] setup(): [main] GPIO14 is HIGH
[  3334][I][main.cpp:2655] setup(): [main] GPIO15 is HIGH
[  3344][I][main.cpp:2655] setup(): [main] GPIO16 is HIGH
[  3354][I][main.cpp:2655] setup(): [main] GPIO17 is HIGH
[  3364][I][main.cpp:2655] setup(): [main] GPIO18 is LOW, probably no PULL-UP
[  3374][I][main.cpp:2655] setup(): [main] GPIO19 is LOW, probably no PULL-UP
[  3384][I][main.cpp:2655] setup(): [main] GPIO21 is HIGH
[  3394][I][main.cpp:2655] setup(): [main] GPIO22 is HIGH
[  3404][I][main.cpp:2655] setup(): [main] GPIO23 is LOW, probably no PULL-UP
[  3414][I][main.cpp:2655] setup(): [main] GPIO25 is LOW, probably no PULL-UP
[  3424][I][main.cpp:2655] setup(): [main] GPIO26 is LOW, probably no PULL-UP
[  3434][I][main.cpp:2655] setup(): [main] GPIO27 is HIGH
[  3444][I][main.cpp:2655] setup(): [main] GPIO32 is LOW, probably no PULL-UP
[  3454][I][main.cpp:2655] setup(): [main] GPIO33 is HIGH
[  3464][I][main.cpp:2655] setup(): [main] GPIO34 is LOW, probably no PULL-UP
[  3474][I][main.cpp:2655] setup(): [main] GPIO35 is LOW, probably no PULL-UP
[  3484][I][main.cpp:2655] setup(): [main] GPIO39 is LOW, probably no PULL-UP
[  3485][I][main.cpp:1756] readprogbuttons(): [main] gpio_00 will execute uppreset = 1
[  3488][I][main.cpp:1756] readprogbuttons(): [main] gpio_12 will execute upvolume = 2
[  3496][I][main.cpp:1756] readprogbuttons(): [main] gpio_13 will execute downvolume = 2
[  3503][I][main.cpp:1756] readprogbuttons(): [main] gpio_14 will execute stop
[  3510][I][main.cpp:1756] readprogbuttons(): [main] gpio_17 will execute resume
[  3518][I][main.cpp:1756] readprogbuttons(): [main] gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
[  3528][I][main.cpp:2667] setup(): [main] Enable pin 35 for IR
[  3534][I][main.cpp:2672] setup(): [main] Start DUMMY display
[  3539][E][main.cpp:2689] setup(): [main] Display not activated
[  3544][I][main.cpp:2278] mk_lsan(): [main] Create list with acceptable WiFi networks
[  3552][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Devices
[  3559][I][main.cpp:2308] mk_lsan(): [main] Added Devices to list of networks
[  3566][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Devices
[  3572][I][main.cpp:2308] mk_lsan(): [main] Added Devices to list of networks
[  3580][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: SSID2
[... 56 lines cut ...]
[ 21759][I][main.cpp:1400] connecttohost(): [main] Connect to host airspectrum.cdnstream1.com:8114/1648_128
[ 21833][I][main.cpp:2541] onConnect(): [main] Connected to host at 144.217.252.222 on port 8114
[ 21869][I][main.cpp:1461] connecttohost(): [main] send GET command
[ 21924][I][main.cpp:3651] handlebyte_ch(): [main] Switch to HEADER
[ 21924][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: Content-Type: audio/mpeg
[ 21927][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-br:128
[ 21933][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
[ 21944][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-br:128
[ 21951][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-description:Today's Easy Hits from the 70s, 80s, 90s through Today!
[... cut ...]
[ 22047][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: Connection: Close
[ 22054][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
[ 22063][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-metaint:16000
[ 22070][I][main.cpp:3742] handlebyte_ch(): [main] Switch to DATA, bitrate is 128, metaint is 16000
[ 22163][I][main.cpp:1294] showstreamtitle(): [main] Streamtitle found, 37 bytes
[ 22164][I][main.cpp:1295] showstreamtitle(): [main] StreamTitle='Honey Cone - Want Ads ';

Edzelf, 2024-01-15:
Yes, that is what I would like to see.
I noticed that most of the lines for the VS1053 (18,19, 23, 4, 32) are LOW.  They should be high normally, see the document chapter 19..
You may also try to set the volume to >75 in the web interface.

chainsawdave, 2024-01-15:
> Yes, that is what I would like to see. I noticed that most of the lines for the VS1053 (18,19, 23, 4, 32) are LOW. They should be high normally, see the document chapter 19.. You may also try to set the volume to >75 in the web interface.

Thank you Ed. The Volume was already maxed out. I have it at 98% right . Love the web interface by the way. Good for noobs like me. 
Checked out Chapter 19. I see what you mean.  Could this be poor connections to the VS1053? Checked mine were good. Could the WROOM ESP32 be faulty? Or have I done something wrong in the config file (below). It does seem that all the connections that should be HIGH are LOW including the encoder which tested OK.

Cheers
David
($)pin_i2s_bck = 5                                      # GPIO Pin number for I2S "BCK"
($)pin_i2s_din = 34                                     # GPIO Pin number for I2S "DIN"
($)pin_i2s_lck = 32                                     # GPIO Pin number for I2S "L(R)CK"
#
clk_dst = 1                                          # Offset during daylight saving time (hours)
clk_offset = 1                                       # Offset with respect to UTC in hours
clk_server = pool.ntp.org                            # Time server to be used
#
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
gpio_17 = resume
gpio_34 = station = icecast.omroep.nl:80/radio1-bb-mp3
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqttbroker = none
mqttpasswd = none
mqttport = 1883
mqttprefix = none
mqttuser = none
#
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 32                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                     # GPIO Pin number for VS1053 "DREQ"
#
preset = 1
preset_00 = stream.live.vc.bbcmedia.co.uk/bbc_world_service #  0 - BBC World Service
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIAMI 256k
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
#
toneha = 14
tonehf = 0
tonela = 3
tonelf = 0
#
volume = 98
#

Edzelf, 2024-01-16:
Remove the VS1053.  Then check if the GPIO lines (19,23) are HIGH.  If LOW, the ESP32 is faulty.  If HIGH, the VS1023 (or the wiring) is faulty.

chainsawdave, 2024-01-16:
> Remove the VS1053. Then check if the GPIO lines (19,23) are HIGH. If LOW, the ESP32 is faulty. If HIGH, the VS1023 (or the wiring) is faulty.

Removed ESP32 GPIO lines 18,19, 23, 5, 4, 32 were HIGH. (as per chapter 19). Checked wiring and continuity from ESP32 to VS1053. These were fine apart from GPIO 5. Selectively removed connections GPIO 18,19, 23, 5, 4, 32 in turn. Each one returned HIGH when nothing was connected.  So it's looking more like the VS1053. I'll order up another. It's possible a managed to fry the board as I did have a few bad connections initially.
Again thanks for your patience. I think this is an awesome project and would like to share once I get it up and running.

chainsawdave, 2024-02-03:
> Remove the VS1053. Then check if the GPIO lines (19,23) are HIGH. If LOW, the ESP32 is faulty. If HIGH, the VS1023 (or the wiring) is faulty.

Hi Ed Here's an update. Got new VS1053 boards to try. The one I had was faulty you were right. However, I'm still not getting any sound out.  Love to get your thoughts.

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (512) esp_core_dump_flash: No core dump⸮⸮ѥѥ⸮⸮found!
E (513) esp_core_dump_flash: No core dump partition found!
[     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3054][I][main.cpp:2582] setup(): [main] Version Thu, 14 Dec 2023 09:00:00 GMT.  Free memory 110580
[  3056][I][main.cpp:2583] setup(): [main] Display type is DUMMY
[  3149][I][main.cpp:2593] setup(): [main] SPIFFS is okay, space 1016801, used 33383
[  3152][I][main.cpp:2614] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3155][I][main.cpp:2614] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3164][I][main.cpp:2614] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
[  3177][I][main.cpp:2486] fillkeylist(): [main] Read 48 keys from NVS
[  3186][I][main.cpp:1910] readIOprefs(): [main] pin_ir set to 35
[  3187][I][main.cpp:1910] readIOprefs(): [main] pin_enc_clk set to 25
[  3193][I][main.cpp:1910] readIOprefs(): [main] pin_enc_dt set to 26
[  3199][I][main.cpp:1910] readIOprefs(): [main] pin_enc_sw set to 27
[  3205][I][main.cpp:1910] readIOprefs(): [main] pin_tft_cs set to 15
[  3211][I][main.cpp:1910] readIOprefs(): [main] pin_tft_dc set to 2
[  3217][I][main.cpp:1910] readIOprefs(): [main] pin_vs_cs set to 5
[  3223][I][main.cpp:1910] readIOprefs(): [main] pin_vs_dcs set to 32
[  3229][I][main.cpp:1910] readIOprefs(): [main] pin_vs_dreq set to 4
[  3235][I][main.cpp:1910] readIOprefs(): [main] pin_spi_sck set to 18
[  3241][I][main.cpp:1910] readIOprefs(): [main] pin_spi_miso set to 19
[  3247][I][main.cpp:1910] readIOprefs(): [main] pin_spi_mosi set to 23
[  3264][I][main.cpp:2655] setup(): [main] GPIO0 is HIGH
[  3274][I][main.cpp:2655] setup(): [main] GPIO2 is LOW, probably no PULL-UP
[  3284][I][main.cpp:2655] setup(): [main] GPIO4 is HIGH
[  3294][I][main.cpp:2655] setup(): [main] GPIO5 is HIGH
[  3304][I][main.cpp:2655] setup(): [main] GPIO12 is HIGH
[  3314][I][main.cpp:2655] setup(): [main] GPIO13 is HIGH
[  3324][I][main.cpp:2655] setup(): [main] GPIO14 is HIGH
[  3334][I][main.cpp:2655] setup(): [main] GPIO15 is HIGH
[  3344][I][main.cpp:2655] setup(): [main] GPIO16 is HIGH
[  3354][I][main.cpp:2655] setup(): [main] GPIO17 is HIGH
[  3364][I][main.cpp:2655] setup(): [main] GPIO18 is HIGH
[  3374][I][main.cpp:2655] setup(): [main] GPIO19 is HIGH
[  3384][I][main.cpp:2655] setup(): [main] GPIO21 is HIGH
[  3394][I][main.cpp:2655] setup(): [main] GPIO22 is HIGH
[  3404][I][main.cpp:2655] setup(): [main] GPIO23 is HIGH
[  3414][I][main.cpp:2655] setup(): [main] GPIO25 is HIGH
[  3424][I][main.cpp:2655] setup(): [main] GPIO26 is LOW, probably no PULL-UP
[  3434][I][main.cpp:2655] setup(): [main] GPIO27 is HIGH
[  3444][I][main.cpp:2655] setup(): [main] GPIO32 is HIGH
[  3454][I][main.cpp:2655] setup(): [main] GPIO33 is HIGH
[  3464][I][main.cpp:2655] setup(): [main] GPIO34 is LOW, probably no PULL-UP
[  3474][I][main.cpp:2655] setup(): [main] GPIO35 is LOW, probably no PULL-UP
[  3484][I][main.cpp:2655] setup(): [main] GPIO39 is LOW, probably no PULL-UP
[  3485][I][main.cpp:1756] readprogbuttons(): [main] gpio_00 will execute uppreset = 1
[  3488][I][main.cpp:1756] readprogbuttons(): [main] gpio_12 will execute upvolume = 2
[  3496][I][main.cpp:1756] readprogbuttons(): [main] gpio_13 will execute downvolume = 2
[  3503][I][main.cpp:1756] readprogbuttons(): [main] gpio_14 will execute stop
[  3510][I][main.cpp:1756] readprogbuttons(): [main] gpio_17 will execute resume
[  3518][I][main.cpp:1756] readprogbuttons(): [main] gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
[  3528][I][main.cpp:2667] setup(): [main] Enable pin 35 for IR
[  3534][I][main.cpp:2672] setup(): [main] Start DUMMY display
[  3539][E][main.cpp:2689] setup(): [main] Display not activated
[  3544][I][main.cpp:2278] mk_lsan(): [main] Create list with acceptable WiFi networks
[  3552][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Devices
[... 62 lines cut ...]
[ 26264][I][main.cpp:1400] connecttohost(): [main] Connect to host airspectrum.cdnstream1.com:8114/1648_128
[ 26672][I][main.cpp:2541] onConnect(): [main] Connected to host at 144.217.252.222 on port 8114
[ 26674][I][main.cpp:1461] connecttohost(): [main] send GET command
[ 26880][I][main.cpp:3651] handlebyte_ch(): [main] Switch to HEADER
[ 26880][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: Content-Type: audio/mpeg
[ 26883][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-br:128
[ 26889][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
[ 26901][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-br:128
[... cut ...]
[ 27003][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: Connection: Close
[ 27010][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
[ 27019][I][main.cpp:3672] handlebyte_ch(): [main] Headerline: icy-metaint:16000
[ 27026][I][main.cpp:3742] handlebyte_ch(): [main] Switch to DATA, bitrate is 128, metaint is 16000
[ 27181][I][main.cpp:1294] showstreamtitle(): [main] Streamtitle found, 58 bytes
[ 27182][I][main.cpp:1295] showstreamtitle(): [main] StreamTitle='Tom Petty - Don't Come Around Here No More ';

Edzelf, 2024-02-03:
The dt line of the rotary encoder is low.  Fix this first.  It should be pulled up to high.Verzonden vanaf mijn Galaxy

"target": "https://github.com/Edzelf/ESP32Radio-V2/issues/64#issuecomment-1925422649",
"url": "https://github.com/Edzelf/ESP32Radio-V2/issues/64#issuecomment-1925422649",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

Edzelf, 2024-02-05:
I also mis the lines with:
```
Slow SPI, Testing VS1053 read/write registers...
Fast SPI, Testing VS1053 read/write registers again...
```
Did you change the source code?

chainsawdave, 2024-02-06:
Hi Ed Quick update. The encoder for the DT was faulty as is GPIO 26 on the
ESP32.  I haven't knowingly changed any source code. I previously had an
error when the code ran the Slow SPI and Fast SPI test but then I think my
VS1053 board was faulty. Not seeing it with the new board. I will first try
assigning a different GPIO to the DT on the encoder. I have tried loading
the code on a different ESP32 board I get an error saying it has an
incompatible crystal. It loads ok but won't connect to WiFi. Anyway, I will
let you know how I get on.

Edzelf, 2024-02-07:
Are you sure you are still using the VS1053?  Check config.h.

chainsawdave, 2024-02-07:
Hi Ed Yes It's wired to a VS1053 Config file below.
//***************************************************************************************************
// config.h                                                                                         *
//***************************************************************************************************
// Configuration definition for your particular set-up.                                             *
//***************************************************************************************************
//
#define NAME "ESP32-Radio"                              // Define name of the radio, also AP SSID
                                                          // Default is "ESP32-Radio"

//#define SDCARD                                          // Experimental: For SD card support (reading MP3-files)

//#define ETHERNET                                        // For wired Ethernet (WT32-ETH-01 or similar)

#define FIXEDWIFI "Devices/Millstone"                       // Add a fixed SSID to the list (WiFi only)

//#define ENABLEOTA                                       // OTA feature

// Define (just one) type of MP3/AAC decoder
**#define DEC_VS1053                                        // Hardware decoder for MP3, AAC, OGG**
//#define DEC_VS1003                                      // Hardware decoder for MP3 only
//#define DEC_HELIX                                       // Software decoder for MP3, AAC. I2S output
//#define DEC_HELIX_AI                                    // Software decoder for AI Audio kit (AC101)
//#define DEC_HELIX_INT                                   // Software decoder for MP3, AAC. DAC output
                                                          // Needs platform = espressif32@3.1.0 !!!!

// Define (just one) type of display.  See documentation.
//#define BLUETFT                                           // Works also for RED TFT 128x160
//#define ST7789                                          // 240x240 TFT (SPI)
//#define OLED1306                                        // 64x128 I2C OLED SSD1306
//#define OLED1309                                        // 64x128 I2C OLED SSD1309
//#define OLED1106                                        // 64x128 I2C OLED SH1106
#define DUMMYTFT                                        // Dummy display
//#define LCD1602I2C                                      // LCD 1602 display with I2C backpack
//#define LCD2004I2C                                      // LCD 2004 display with I2C backpack
//#define ILI9341                                         // ILI9341 240*320
//#define NEXTION                                         // Nextion display
//

// Define ZIPPYB5 if a ZIPPY B5 Side Switch is used instead of a rotary switch
//#define ZIPPYB5

// End of configuration parameters.

chainsawdave, 2024-02-11:
Hi Ed The replacement parts (rotary encode and new ESP32 boards finally arrived. I still can't get the radio to stream. 
[(POWERON_RESET),boot0x17 (SPI_FAST_ log Feb 2024.txt](https://github.com/Edzelf/ESP32Radio-V2/files/14234273/POWERON_RESET.boot0x17.SPI_FAST_.log.Feb.2024.txt)
[configuration.txt](https://github.com/Edzelf/ESP32Radio-V2/files/14234274/configuration.txt)
Hopefully you can see something in there. I've checked the continuity of the connections so i don't think it's that. Also I get different errors depending on the boards I plug in. Cheers D..

Edzelf, 2024-02-12:
I'm pretty sure that you have changed something in the main.cpp file or you ar using an old version.  In the logging there should be a line saying:
`setup(): [main] Playtask started`
The first debug line should say:
`[I][main.cpp:2584] setup(): [main] Version Thu, 26 Dec 2023 14:30:00 GMT.`
Please use the last version of the software.

chainsawdave, 2024-02-17:
Thanks  Ed I will re download and try again,
Cheers
David

chainsawdave, 2024-02-19:
Hi Ed. After much fiddling
[ets Jul 29 2019 122146 log for Ed.txt](https://github.com/Edzelf/ESP32Radio-V2/files/14334162/ets.Jul.29.2019.122146.log.for.Ed.txt)

I've managed to get all the pins to HIGH. It was a combination of mis-wiring and dry joints. I downloaded the latest code. The board is a VS1053 Arduino shield. Ive' tried with two different boards with the same results, Can you see anyhting in the code?
Cheers
David

Edzelf, 2024-02-19:
I just uploaded a new version.  The new version allows SPDIF output and I corrected the handling of mono stations.

In your logging I still see SPI errors.  So the wiring to the VS1053 is not okay.
You may try to lower the SPI frequency at line 205 in VS1053.cpp, but normally the VS1053 should work at 5 MHz.