Edzelf/ESP32Radio-V2 #68: URL of a podcast gives a crashe. Radio restart

fanfanlatulipe26, 2022-11-16:
This address crashes and the radio restart.
api.europe1.fr/podcast/mp3/itunes-1146402624/4146989/podcast.mp3

This URL is OK in Mozilla, Edge
As far as I understand the is a redirect to a new URL, and this one makes a new redirect o itself ....  

```
D: Radiofuncs cmd is 1
D: Stopping client
D: Connect to host api.europe1.fr/podcast/mp3/itunes-1146402624/4146989/podcast.mp3
D: send GET command
D: Switch to HEADER
D: Headerline: Content-Type: text/html; charset=UTF-8
D: Headerline: Content-Length: 1240
D: Headerline: Connection: close
D: Headerline: Cache-Control: max-age=0, public
D: Headerline: Location: https://ais.cloud-services.paris/europe1/prod/audio/emissions/hondelatte-raconte-1749439/Cecile-Combettes-meurtre-et-mystere-au-monastere-Le-recit-3131668.mp3?awCollectionId=174
D: Headerline: X-S-Maxage: 0
D: Headerline: X-Server: (null)
D: Headerline: X-Varnish-Age: 0
D: Headerline: X-Varnish-Cache: MISS
D: Headerline: X-Varnish-Hits: 0
D: Headerline: X-Varnish-Hostname: ip-172-30-232-11
D: Headerline: X-Cache: Miss from cloudfront
D: Headerline: X-Amz-Cf-Pop: CDG3-C2
D: Headerline: X-Amz-Cf-Id: v0gLYf4Dww5QxSlm8FycPdEojjz_E2xCWxokRnTgbK5JaJt-Sa7g7w==
D: Redirect
D: Switch to HEADER
D: Radiofuncs cmd is 1
D: Connect to host ais.cloud-services.paris/europe1/prod/audio/emissions/hondelatte-raconte-1749439/Cecile-Combettes-meurtre-et-mystere-au-monastere-Le-recit-3131668.mp3?awCollectionId=1749439&awEpisode
D: send GET command
D: Switch to HEADER
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Instance-id: efc7d5e3f7c6e61b79fa6f4fe9373c05
D: Headerline: Server: Axians AdService 9.0.5
D: Headerline: Connection: close
D: Headerline: Location: http://ais.cloud-services.paris:80/europe1/prod/audio/emissions/hondelatte-raconte-1749439/Cecile-Combettes-meurtre-et-mystere-au-monastere-Le-recit-3131668.mp3?awCollectionId=1
D: Headerline: Content-Length: 0
D: Headerline: X-Node: adn-eqxd1-e13
D: Headerline: Content-Disposition: inline
D: Headerline: Set-Cookie: AISSessionId=6321f25b9a24bdff_5417511_3WpbXRaO__0000003tZFG; Path=/; Domain=ais.cloud-services.paris; Max-Age=6000; Expires=Wed, 16 Nov 2022 09:57:07 GMT
D: Redirect
D: Song stopped correctly after 20 msec
D: Radiofuncs cmd is 1
D: Connect to host ais.cloud-services.paris:80/europe1/prod/audio/emissions/hondelatte-raconte-1749439/Cecile-Combettes-meurtre-et-mystere-au-monastere-Le-recit-3131668.mp3?awCollectionId=1749439&awEpis
D: send GET command

Stack smashing protect failure!

abort() was called at PC 0x400f3cc7 on core 1

Backtrace:0x40083f2d:0x3ffb25500x4008cd7d:0x3ffb2570 0x40092859:0x3ffb2590 0x400f3cc7:0x3ffb2610 0x400d5699:0x3ffb2630 0x400d57ae:0x3ffb27d0 0x400d86f5:0x61446174  |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...

```

Edzelf, 2022-11-16:
The last redirect points to a https location.  This is not supported.

fanfanlatulipe26, 2022-11-16:
I found the problem: there is an overflow of char array getreq used for building the  GET request.
Its size is 
 `char        getreq[300] ;` 
In this example the redirect URL are very large.
I changed the declaration to 
`char        getreq[500] ;`
and it is OK now.
In fact in this example the size of the GET command is 356.