Edzelf/ESP32Radio-V2 #71: New to this need a little help please, thank you

scaudle, 2022-12-06:
Hello and thanks for a great project. This is my first time with plaformio and a project of this size. Im new to all this. 

I compiled ok and loaded ok which is great but it does not seem to work for me. I go to the ip address in the web and i get "SORRY"

Here is my terminal from running build.

any help on pointing me to the place to fix would be fantastic. I am very excited about this project.

Thanks a million to any one who helps me. :) Have a great day

Warning! Ignore unknown configuration option `monitor_flags` in section [env:esp32]
--- forcing DTR inactive
--- forcing RTS inactive
--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

D: Starting ESP32-radio running on CPU 1 at 240 MHz.
D: Version Thu, 03 Nov 2022 13:00:00 GMT.  Free memory 110580
D: Display type is NEXTION
D: SPIFFS is okay, space 1378241, used 0
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1507328
D: Read 6 keys from NVS
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is LOW, probably no PULL-UP
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: Start NEXTION display
D: Init Nextion, pins 16, 17
D: Nextion command cls BLACK
D: Nextion command page 0
D: NEXTION output 'Starting......' to 1
D: NEXTION output 'Thu, 03 Nov 2022' to 2
D: NEXTION output 'By Ed Smallenburg' to 3
D: Create list with acceptable WiFi networks
D: Added Littles-place to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 12
D:  1 - Littles-place             Signal: -46 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - DIRECT-68-HP 1200 Neverstop Signal: -47 dBm, Encryption WPA2_PSK,
D:  3 - Littles-place             Signal: -48 dBm, Encryption WPA2_PSK, Acceptable
D:  4 - Littles-place             Signal: -71 dBm, Encryption WPA2_PSK, Acceptable
D:  5 - Drslow                    Signal: -78 dBm, Encryption WPA2_PSK,
D:  6 - ASUS 84                   Signal: -82 dBm, Encryption WPA2_PSK, 
D:  7 - cofamilia                 Signal: -84 dBm, Encryption WPA2_PSK,
D:  8 - Seo Family ASUS           Signal: -87 dBm, Encryption WPA2_PSK, 
D:  9 - NETGEAR93a                Signal: -88 dBm, Encryption WPA2_PSK,
D: 10 - This is the Way           Signal: -90 dBm, Encryption WPA2_PSK, 
D: 11 - chofam                    Signal: -92 dBm, Encryption WPA2_PSK,
D: 12 - HP-Print-E2-LaserJet Pro MFP Signal: -92 dBm, Encryption OPEN,
D: End of list
D: updateNr 0 <= 0 to 0, relative is 0
D: updateNr result is 0
D: nextPreset is 0
E (6395) gpio: gpio_set_level(226): GPIO output gpio_num error
E (6396) gpio: gpio_set_level(226): GPIO output gpio_num error
E (6409) gpio: gpio_set_level(226): GPIO output gpio_num error
E (6409) gpio: gpio_set_level(226): GPIO output gpio_num error
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to network
D: NEXTION output 'Connect to network' to 4
D: Try WiFi Littles-place
D: NEXTION output 'SSID = Littles-place' to 5
D: SSID = Littles-place
D: IP = 192.168.86.20
D: NEXTION output 'IP = 192.168.86.20' to 6
D: Nextion command page 1
D: Start web server
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 06:19:27
D: updateNr 0 <= 0 to 0, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: Radiofuncs cmd is 1
D: Connect to host
D: Request  failed!
D: Nextion t0.txt="ESP32-Radio"
D: Nextion t1.txt=""
E (19203) gpio: gpio_set_level(226): GPIO output gpio_num error
E (19204) gpio: gpio_set_level(226): GPIO output gpio_num error
E (19209) gpio: gpio_set_level(226): GPIO output gpio_num error
E (19215) gpio: gpio_set_level(226): GPIO output gpio_num error

Edzelf, 2022-12-06:
"Sorry" means that you forgot to upload the files in the "data" directory into SPIFFS.  See also the 4th debug output line.Verzonden vanaf mijn Galaxy

"target": "https://github.com/Edzelf/ESP32Radio-V2/issues/71",
"url": "https://github.com/Edzelf/ESP32Radio-V2/issues/71",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

scaudle, 2022-12-06:
Edzelf, thank you , I have followed your post, I guess the file system worked as I now have a webpage. But there is no presets, and I am still unable to get it working. 
So sorry to be a pain, but thank you so much for your help. 
I am using a nodeMCU 32s from amazon. ESP WROOM 32 ESP32 Development Board 2.4GHz WiFi Dual Cores Microcontroller Integrated with Antenna RF Low Noise Amplifiers Filters
and I have a VS1053. I do not have the monitor wired up yet. 

Warning! Ignore unknown configuration option `monitor_flags` in section [env:esp32]
--- forcing DTR inactive
--- forcing RTS inactive
--- Terminal on COM3 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
D: NEXTION output 'SSID = Littles-place' to 5
D: SSID = Littles-place
D: IP = 192.168.86.30
D: NEXTION output 'IP = 192.168.86.30' to 6
D: Nextion command page 1
D: Start web server
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 11:22:41
D: nvs_get_str failed 1102 for key preset, keylen is 6, len is 150!
D: Contents: 
D: updateNr 0 <= 0 to 0, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: Radiofuncs cmd is 1
D: Connect to host
D: Request  failed!
D: Nextion t0.txt="ESP32-Radio"
D: Nextion t1.txt=""
E (21960) gpio: gpio_set_level(226): GPIO output gpio_num error
E (21961) gpio: gpio_set_level(226): GPIO output gpio_num error

Edzelf, 2022-12-06:
You need to fill the preferences in the web-interface (config page).
See the document page 3 "Preferences".

Edzelf, 2022-12-06:
Pin 4 and 5 are double defined and conflicting.

scaudle, 2022-12-06:
Hi :)

Ok here is a small issue. My ESP 32S does not have pin 16
Which shows in the esp32_radio_init.Ino. Line 71 that connects to VS1053 “DCS”

I changed 16 to 22 as in the docs it says under AI tanker kit that 21 and 22 are free, which might mean
For something else but this is what I used. 

I have not changed pin 4 or 5 and they are listed as 4 = DREQ and 5 = CS on the VS1053, 
Is there a different file I would look at to solve this. 

I see on the web config that 4 and 5 are listed more than once but I did not list these, used default and only doc to pin 22

Here is web config

clk_dst = 1                                          # Offset during daylight saving time (hours)
clk_offset = 1                                       # Offset with respect to UTC in hours
clk_server = pool.ntp.org <http://pool.ntp.org/>                            # Time server to be used
#
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
gpio_17 = resume
gpio_34 = station = icecast.omroep.nl:80/radio1-bb-mp3 <http://icecast.omroep.nl/radio1-bb-mp3>
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqttbroker = none
mqttpasswd = none
mqttport = 1883
mqttprefix = none
mqttuser = none
#
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
pin_i2s_bck = 5                                      # GPIO Pin number for I2S "BCK"
pin_i2s_din = 4                                      # GPIO Pin number for I2S "DIN"
pin_i2s_lck = 32                                     # GPIO Pin number for I2S "L(R)CK"
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 22                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
#
preset = 7
preset_00 = 109.206.96.34:8100 <http://109.206.96.34:8100/>                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 <http://airspectrum.cdnstream1.com:8114/1648_128> #  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050 <http://us2.internet-radio.com:8050/>              #  2 - CLASSIC ROCK MIAMI 256k
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 <http://airspectrum.cdnstream1.com:8000/1261_192> #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 <http://airspectrum.cdnstream1.com:8008/1604_128> #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105 <http://us1.internet-radio.com:8105/>              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3 <http://icecast.omroep.nl/radio1-bb-mp3>       #  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032 <http://205.164.62.15:10032/>                      #  7 - 1.FM <http://1.fm/> - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3 <http://skonto.ls.lv:8002/mp3>                    #  8 - Skonto 128k
preset_09 = 94.23.66.155:8106 <http://94.23.66.155:8106/>                        #  9 - *ILR CHILL and GROOVE
preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
preset_12 = 149.255.59.164:8132 <http://149.255.59.164:8132/>                      # 12 - Brit_Com
#
toneha = 3
tonehf = 3
tonela = 3
tonelf = 3
#
volume = 100
#
wifi_00 = Littles-place/Molly100
wifi_01 = Littles-place/Molly100

Edzelf, 2022-12-07:
First remove lines:
pin_i2s_bck = 5                                      # GPIO Pin number for I2S "BCK"
pin_i2s_din = 4                                      # GPIO Pin number for I2S "DIN"
pin_i2s_lck = 32                                     # GPIO Pin number for I2S "L(R)CK"
if you are not using I2S device.  This will solve the GPIO conflicts.\

The VS1003 (printing on the chip, not on the board) will work for mp3 stations.