Edzelf/ESP32Radio-V2 #75: Not getting expected performance

ma261065, 2023-02-02:
First of all *THANK YOU* for being the only guy on the internet who switches the VS1053 from MIDI mode to MP3 mode in his driver. You wouldn't believe how many days I wasted until I found your code!!!

I have built the project, and it is running well. However, mine seems to max out at around 128kb/s streams. If I pick any higher-rate streams from the radio search, or use this test file: ia803405.us.archive.org/20/items/mjq-1983-montreal-jazz-festival-cbc/01-Introduction.mp3 - then the sound is choppy.

Given that you seem to get up to 320kb/s (according to your documentation), I wonder whether there is something wrong with my setup (using ESP32-WROOM-32 module) or if this is expected. I have tried both with a VS1053 and using the software MP3 decoder via an I2S DAC.

Can you play the test file above without choppiness?

Edzelf, 2023-02-02:
Thanks for your kind words.
I have tried the station.  I had the same choppy sound.
I also tried a 320 kbps station from the "search" menu, no problem there.

ma261065, 2023-02-03:
Thanks for verifying.

So, for example, does this 320kbps station play OK for you? (It's super choppy for me when using both I2S & VS1053 out)

25303.live.streamtheworld.com:80/DEEP_RADIO.mp3

If it is also choppy for you, could you please share an example of a 320kbps station that plays smoothly for you?

Edzelf, 2023-02-03:
Good station with 320kbs stream (here):
```
server-27.stream-server.nl:8192/stream
ekrrockradio4.scs.ms:8400/
www.rmgoldserver.eu:8963/
mp3.wunschradio.de/wunschradio-60er.mp3
live.radiocrash.net/
audio.674.fm/stream
88.198.10.229:9500/stream
148.251.43.231:8938/stream
strm112.1.fm/ambientpsy_mobile_mp3
188.40.97.185:8179/stream
streaming5.elitecomunicacion.es:8030/live.mp3
mediaserv38.live-streams.nl:8027/live
```
25303.live.streamtheworld.com:80/DEEP_RADIO.mp3 plays perfect here.

ma261065, 2023-02-03:
Thanks!!

Interesting, every one of them is choppy on my end. Pressing "Test" on the main web page shows "Chunks in queue 0" more often than not on those stations. When it's on a lower rate stream the "Chunks in queue" is always close to 400 and the sound is fine.

I've tried moving closer to my WiFi router with no change. I wonder if I'm using a different version of some of the libraries to you? Any other thoughts?

Edzelf, 2023-02-03:
Better try a network of a friend in the neighborhood.

pablobertagna, 2023-05-31:
Hi everyone. First things first, thanks for sharing this great project. @ma261065, I was wondering if you could solve this issue. I'm having just the same problem... choppy sound above 128kbps. I'm using a WROOM-32 module too. Pretty sure it has nothing to do with Internet speed (I tested with several ISPs). Any help would be appreciated. Thanks in advance.

aly-fly, 2023-06-01:
You have to find a local radio station, preferably with a server within 50 km radius. 
This radio has a problem, has very small buffer and pulls lots of very small packets. Any longer connection can't handle this type of load.

z34pec, 2023-06-02:
I have similar question:

naxidigital-80s128ssl.streaming.rs:8042/;stream.nsv

this stream can not be played. Does somebody knows why?

I tried to find new stream for the same station but look like it is the only one that is available. It play ok on other devices but on esp32 radio no (talking about this one, edzelf's).

aly-fly, 2023-10-09:
@Edzelf  I tried to find a setting or part of the code where network client buffer size is defined. Not successful. Please help.

Random local station: Free memory is 73716, chunks in queue 395, bitrate 192 kbps
It seems we could allocate approx +60 kb for the increased buffer? This would give 0.3 second worth of play time at 192 kbps. Not much, but worth trying for the sake of reducing number of packets and frequency of the packets on the network. 
Where could I change this in the code?

Edzelf, 2023-10-10:
The network client buffer size cannot be changed as far as I know.  You may change the application buffer by changing "QSIZ" in the main program.  Note that WiFi may be unstable if the free memory is limited.
Chunks to the VS1053 are always 32 bytes.  Blocks from the network will be received (async) by the call-back function "handleData ()".