Edzelf/ESP32Radio-V2 #85: esp32radio v2 with 2 esp32!!!

clear-sky, 2023-03-13:
As far as I understand, due to technical problems, the Bluetooth transmitter cannot be integrated with the radio, so why not use two chips? In this case, both the project will be more intimate and its educational aspect will be higher!!!

Edzelf, 2023-03-13:
Agree. Just connect a Bluetooth transmitter to the audio output.

WerkAG, 2023-03-14:
Here is an example using a  Bluetooth transmitter...<br><br>

[image]

clear-sky, 2023-03-14:
hi all 
i mean using an esp32 as radio reciever via wifi and another esp32 as bluetooth sender

WerkAG, 2023-03-14:
Why complicate? Edzelf suggestion its easy to implement, doesn't require any software changes and works great.

closed 2023-03-17 by clear-sky