Edzelf/ESP32Radio-V2 #86: NVS bin creation support

autagi, 2023-03-18:
Hi,
I'm newbie. I want to use Arduino IDE to manage and upload this project to my ESP32.
I found Espressif download tool to flash bin file into NVS memory.
I haven't PlatformIO.
I think I must have a tool to create bin file of data directory (web server files).
I found "NVS Partition Generator Utility" but I need cvs file with parameters and I don't know where can I find them.
Is there a more easy way to flash data files?

thanks,
Gianluigi

Edzelf, 2023-03-18:
There is no need to fill the NVS.  The program itself will take care of that.
The data directory should be uploaded to SPIFFS, but the Arduino IDE has the tool to do that.

autagi, 2023-03-18:
Thank you for the fast support.
I will follow your guide