Edzelf/ESP32Radio-V2 #88: How to use Nextion "soft" buttons & battery monitor?

NickBramley, 2023-03-22:
I am using a Nexion NX4832 which is working great! How do I get the soft buttons to work, when I touch NEXT the ESP32 is not seeing the command. I have "Nextion out" connected to GPIO 21. GPIO 16 is being used for the VS1053 XDCS.
See:-

D: pin_nxt_rx set to 21
D: pin_nxt_tx set to 17 in the config file

and
D: pin_nxt_rx set to 21
D: pin_nxt_tx set to 17

D: Init Nextion, pins 16, 17 in the log

After a "long press" of the rotary encoder the audio stops. How to get it going again without rebooting?

I am planning to put the battery monitor circuit (using the potential divider as suggested) connected to GPIO34. How do I add this info please?
Thanks for any help!

Existing config file:-

clk_dst = 0                                          # Offset during daylight saving time (hours)
clk_offset = 0                                       # Offset with respect to UTC in hours
clk_server = pool.ntp.org                            # Time server to be used
#
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
gpio_17 = resume
gpio_34 = station = icecast.omroep.nl:80/radio1-bb-mp3
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqttbroker = none
mqttpasswd = none
mqttport = 1883
mqttprefix = none
mqttuser = none
#
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
pin_i2s_bck = 5                                      # GPIO Pin number for I2S "BCK"
pin_i2s_din = 4                                      # GPIO Pin number for I2S "DIN"
pin_i2s_lck = 32                                     # GPIO Pin number for I2S "L(R)CK"
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_nxt_rx = 21                                      # GPIO Pin number for Nextion "TX"
pin_nxt_tx = 17                                      # GPIO Pin number for Nextion "RX"
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 16                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
#
preset = 15
preset_00 = 78.129.202.200:8040                      #  0 - Radio Caroline
preset_01 = server.webnetradio.net:5430/stream       #  1 - Easy Hits Florida 128k
preset_02 = server.webnetradio.net:5418/stream       #  2 - CLASSIC ROCK Florida 256k
preset_03 = server.webnetradio.net:5282/stream       #  3 - Magic Oldies Florida
preset_04 = server.webnetradio.net:5218/stream       #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = 204.141.167.19:9170/stream               #  5 - New Country 93
preset_06 = 50.31.185.139:80/1058                    #  6 - 113.fm Route66
preset_07 = 144.217.79.175:6986/stream               #  7 - Classic Hits 109 Country Hits
preset_08 = 192.99.177.195:8006/stream               #  8 - Motown Sound
preset_09 = 216.235.89.162:80/1992_128.mp3           #  9 - Blues Music Fan Radio
preset_10 = 185.33.21.112:80/blues_64                #  10 - 1.FM Blues Radio
preset_11 = jenny.torontocast.com:8008/stream        #  11 - A Mississippi Blues
preset_12 = 62.149.196.16:8080/stream                #  12 - Rock Radio Station
preset_13 = 65.21.202.84:8029/stream                 #  13 - Wicked Rock
preset_14 = 46.105.100.126:9085/stream               #  14 - D Rock
preset_15 = 198.178.123.17:10922                     # 15 - San Francisco's 70s Hits
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 61
#
wifi_00 = ***************
wifi_01 = SSID2/PASSWD2

log file:-

D: Starting ESP32-radio running on CPU 1 at 240 MHz.
D: Version Thu, 03 Nov 2022 13:00:00 GMT.  Free memory 110580
D: Display type is NEXTION
D: SPIFFS is okay, space 1318001, used 33634
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1441792
D: Found partition 'coredump' at offset 0x3F0000 with size    65536
D: Read 54 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_nxt_rx set to 21
D: pin_nxt_tx set to 17
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: Pin 5 is already reserved!
D: pin_i2s_bck set to 5
D: pin_i2s_lck set to 32
D: Pin 4 is already reserved!
D: pin_i2s_din set to 4
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Enable pin 35 for IR
D: Start NEXTION display
D: Init Nextion, pins 16, 17
D: Nextion command cls BLACK
D: Nextion command page 0
D: NEXTION output 'Starting......' to 1
D: NEXTION output 'Thu, 03 Nov 2022' to 2
D: NEXTION output 'By Ed Smallenburg' to 3
D: Create list with acceptable WiFi networks
D: Added MYSSID to list of networks
D: Added EE- to list of networks
D: Added SSID2 to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - EE
D:  2 - Ni
D: End of list
D: updateNr 0 <= 0 to 15, relative is 0
D: updateNr result is 0
D: nextPreset is 0
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: Connect to network
D: NEXTION output 'Connect to network' to 4
D: NEXTION output 'SSID = EE' to 5
D: SSID = EE-Cowans Yard
D: IP = 192.168.1.100
D: NEXTION output 'IP = 192.168.1.100' to 6
D: Nextion command page 1
D: Start web server
D: Network found. Starting mp3 client, mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 14:32:05
D: updateNr 0 <= 15 to 15, relative is 0
D: updateNr result is 15
D: nextPreset is 15
D: Radiofuncs cmd is 1
D: Connect to host 198.178.123.17:10922
D: Song stopped correctly after 0 msec
D: send GET command
D: Nextion t0.txt="ESP32-Radio"
D: Nextion t1.txt=""
D: Switch to HEADER
D: Headerline: icy-name:San Francisco's 70's HITS!
D: Headerline: icy-genre:70s
D: Headerline: icy-url:http://www.70sRadioHits.com
D: Headerline: content-type:audio/mpeg
D: Headerline: icy-pub:1
D: Headerline: icy-metaint:16384
D: Headerline: icy-br:128
D: Switch to DATA, bitrate is 128, metaint is 16384
D: Nextion t2.txt="San Francisco's 70's HITS!"
D: Streamtitle found, 62 bytes
D: StreamTitle='Robert Palmer - Every Kinda People - Double Fun';
D: Nextion t1.txt="Robert Palmer\rEvery Kinda People - Double Fun"

Thanks for any help

Nick

NickBramley, 2023-03-23:
Thanks Ed! Is that a modified V2 or is it V1?

Edzelf, 2023-03-23:
Modified V2.

NickBramley, 2023-03-23:
Brilliant!!!!!..... All working...thanks again.

NickBramley, 2023-03-23:
I found the new version but it appeared to be in the V1 project on Github rather than the V2...??
Very pleased with it! It is a brilliant bit of software, well done and congrats on a great project.

Edzelf, 2023-03-23:
I guess I did the wrong upload.

NickBramley, 2023-04-20:
Hi Ed How can I identify the version to get the full functionality from a Nextion screen. The copy that I downloaded after this was posted has gone somewhere I can't find it. BTW I have completed a new PCB for this project....just waiting for delivery.

Edzelf, 2023-04-20:
Look at the definition of "VERSION" in main.cpp.  It should read:
#define VERSION     "Tue, 22 Nov 2022 15:30:00 GMT"

NickBramley, 2023-05-05:
Hi Ed,

                I am using the version as described but it is not responding to inputs from the Nextion screen.

 

Still getting nothing from the PCM5102 either

 

Hey…. I really appreciate you taking the time to support your wonderful software so well….!!

 

Nick