Edzelf/ESP32Radio-V2 #89: Compiling for ESP32-S3 - code not executing past

aranFish, 2023-03-31:
Hello, thank you for such a great, well-documented project.

Because I have a spare board, I'm attempting to compile your project for the ESP32-S3 ([using this "FeatherS3" board](https://esp32s3.com/feathers3.html)).

I've managed to get the code to compile and upload successfully, by making the following changes to platformio.ini:
```
board = um_feathers3
build_flags = -DARDUINO_USB_MODE=1;
```

I havn't yet wired up my VS1003 or TFT modules. I would just like to observe to see if the module will set up a wireless AP first.

After compiling and uploading, and then uploading the filesystem, I open the serial monitor. This is what I see:

`D: 8257524/313224 - Starting ESP32-radio running on CPU 1 at 240 MHz.
D: 8257524/313224 - Version Tue, 22 Nov 2022 15:30:00 GMT.  Free memory 8257524
D: 8257524/313224 - Display type is DUMMY
D: 8257524/308864 - SPIFFS is okay, space 1016801, used 33634
D: 8257524/308828 - Found partition 'nvs     ' at offset 0x009000 with size    20480
D: 8257524/308828 - Found partition 'otadata ' at offset 0x00E000 with size     8192
D: 8257524/308828 - Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
D: 8257524/308764 - Read 0 keys from NVS
D: 8257524/308692 - pin_spi_sck set to 18
D: 8257524/308692 - pin_spi_miso set to 19
D: 8257524/308692 - pin_spi_mosi set to 23
D: 8257524/308692 - GPIO0 is HIGH
D: 8257524/308692 - GPIO2 is HIGH
D: 8257524/308692 - GPIO4 is HIGH
D: 8257524/308692 - GPIO5 is HIGH
D: 8257524/308692 - GPIO12 is HIGH
D: 8257524/308692 - GPIO13 is HIGH
D: 8257524/308692 - GPIO14 is HIGH
D: 8257524/308692 - GPIO15 is HIGH
D: 8257524/308692 - GPIO16 is HIGH
D: 8257524/308692 - GPIO17 is HIGH
D: 8257524/308692 - GPIO18 is HIGH
`

... and there, the code execution stops. I have looked at other Issues threads, and see that I should see other GPIO statuses, and then the WiFi access point being set up, but the code execution never reaches that point.

Might you have any suggestions that I could try, before I give up and purchase a normal ESP32 board?

Many thanks to you.

Aran

Edzelf, 2023-04-01:
Sorry, I not have one.  So I cannot test it here.

aranFish, 2023-04-01:
OK thank you

closed 2023-04-01 by aranFish

M4rkoV2, 2024-09-19:
I have the same problem with esp32-s3. Mine resets

`Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3243][I][main.cpp:2431] setup(): [main] Version Fri, 05 Jul 2024 13:40:00 GMT.  Free memory 286708
[  3252][I][main.cpp:2432] setup(): [main] Display type is DUMMY
[  3325][I][main.cpp:2442] setup(): [main] SPIFFS is okay, space 956561, used 0
[  3385][I][main.cpp:2463] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3395][I][main.cpp:2463] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3404][I][main.cpp:2463] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1048576
[  3414][I][main.cpp:2463] setup(): [main] Found partition 'coredump' at offset 0x3F0000 with size    65536
[  3426][I][main.cpp:2334] fillkeylist(): [main] Read 0 keys from NVS
[  3433][I][main.cpp:1841] readIOprefs(): [main] 'pin_spi_sck  ' set to 18
[  3440][I][main.cpp:1841] readIOprefs(): [main] 'pin_spi_miso ' set to 19
[  3447][I][main.cpp:1841] readIOprefs(): [main] 'pin_spi_mosi ' set to 23
[  3464][I][main.cpp:2502] setup(): [main] GPIO0 is HIGH
[  3479][I][main.cpp:2502] setup(): [main] GPIO2 is HIGH
[  3494][I][main.cpp:2502] setup(): [main] GPIO4 is HIGH
[  3509][I][main.cpp:2502] setup(): [main] GPIO5 is HIGH
[  3524][I][main.cpp:2502] setup(): [main] GPIO12 is HIGH
[  3539][I][main.cpp:2502] setup(): [main] GPIO13 is HIGH
[  3554][I][main.cpp:2502] setup(): [main] GPIO14 is HIGH
[  3569][I][main.cpp:2502] setup(): [main] GPIO15 is HIGH
[  3584][I][main.cpp:2502] setup(): [main] GPIO16 is HIGH
[  3599][I][main.cpp:2502] setup(): [main] GPIO17 is HIGH
[  3614][I][main.cpp:2502] setup(): [main] GPIO18 is HIGH
[  3629][I][main.cpp:2502] setup(): [main] GPIO19 is HIGH
[  3644][I][main.cpp:2502] setup(): [main] GPIO21 is HIGH
[  3649][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  3665][I][main.cpp:2502] setup(): [main] GPIO22 is LOW, probably no PULL-UP
[  3672][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  3688][I][main.cpp:2502] setup(): [main] GPIO23 is LOW, probably no PULL-UP
[  3695][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  3711][I][main.cpp:2502] setup(): [main] GPIO25 is LOW, probably no PULL-UP
[  3728][I][main.cpp:2502] setup(): [main] GPIO26 is HIGH
[  3743][I][main.cpp:2502] setup(): [main] GPIO27 is HIGH
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c3
  #0  0x403743c3 in _DoubleExceptionVector at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:463

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
`

Edzelf, 2024-10-17:
The latest version supports ESP32-S3.  Not fully tested yet.  I tested the simplest version (with DEC_HELIX_SPDIF output and dummy display).