Edzelf/ESP32Radio-V2 #95: difficulty compiling on Arduino IDE

The-Valveman, 2023-04-19:
I am struggling to use this version as suggested to me in another issue that i am having with the earlier version (SD cards not working)
I am trying to compile and upload this to the ESP32 using the instructions in the manual there seems to be a couple of conflictions in the instructions:-
I try to follow the below instructions but there are 2 issues

1 when i try to rename the file "main.cpp" the arduino IDE does not open as it appears i cannot change the .cpp extension to .INO itried to just rename the filt to "ESP32Radio.ino" but it doesn't open in the Arduino IDE

2 in the instruction below it tells me to remove the following folders:  ”doc”, ”tools”, ”ESP32-Radio.code-workspace”, ”LICENCE”, ”platformio.ini” and ”README.md”

then it states the work directory should contain the following:  “data”, “doc”, “nextion”, “tools”, the source files and maybe “lib”.

Do I have to replace the "tools" and "Docs" folders back into the workspace directory? as when i have carried out the above the tools and docs folders are not in the workspace directory.

Thanks in advance for your help Peter

Exerpt form your manual:-

Goto this directory and remove ”doc”, ”tools”, ”ESP32-Radio.code-workspace”, ”LICENCE”, ”platformio.ini” and ”README.md”
Move the 3 files in “src” to your work directory. And remove the (now empty) “src” directory.
Move all files from “include” to your work directory and remove the “include” directory.
Edit the file config.h to meet your configuration (display, codec, ..).
Move the files in “lib/codecs/src” to your work directory.  If you use VS1053, you may skip the aac* and mp3* files.
Depending on your type of display, move the files in “lib/ILI9341/src” to your work directory,
or move the files in “lib/LCD1602/src” to your work directory,
or move the files in “lib/LCD2004/src” to your work directory,
or move the files in “lib/NEXTION/src” to your work directory,
or move the files in “lib/bluetft/src” to your work directory,
or move the files in “lib/dummytft/src” to your work directory,
or move the files in “lib/OLED/src” to your work directory,
or move the files in “lib/ST7789/src” to your work directory.
Remove the “lib” directory if you want to save some diskspace.
Rename the file “main.cpp” to “ESP32Radio.ino”.  The Arduino IDE should start.
Your work directory will now contain “data”, “doc”, “nextion”, “tools”, the source files and maybe “lib”.
Start the Arduino IDE by double clicking “ESP32Radio.ino”.
Select the right board (ESP32 Dev Module) in Tools and set the right Port.
Edit the “config.h” file to your needs.
You can now verify the sketch.  You probably have to install some libraries.

The-Valveman, 2023-04-25:
This would have been a great project if i could get it to work I have version 1 working but the SD Card feature is not working. I was suggested to use the version 2 of this project unfortunately i am a fairy newcomer to programming and am reasonably comfortable using the Arduino IDE as per the post above i am hitting a brick wall when trying to set this up. I have tried to open it in Platformio but i have no clue with what i am doing in Platform i have followed instructions online but i am getting errors trying and help is not forthcoming so it looks like i am going have to shelve this project which is unfortunate!!

Edzelf, 2023-04-25:
Just show the errors you get with PlatformIO and I can help you.

The-Valveman, 2023-04-29:
Hi Ed thankyou for the reply i have been away for a couple of days hence the delay responding.
I have finally got the code to compile and upload in PLatformio But have now hit another brick wall
I followed the instructions to upload via the terminal but now i am stuck the instructions say to connect the the ESP32Radio and use the password "ESP32Radio" i have tried this via a web browser and via the terminal the ESP32 Radio after stating "wiFi Failed!" is gives an ip address of 1.0.0.0 i have even tried this address but still to no avail What do you use to connect to the ESP32Radio? I have even tried to edit the  config text file with my WiFi credentials but it wont connect it seems to try to connect to a network called "Laptop" I really want this to work as i like the web interface ability to add/edit the preset  stations without me having to make a keyboard on the nextion screen to enter my preferred stations as well as not having to hard code them. this internet radio is going to be incorporated into a receiver that also incorporates a DAB/FM tuner which is working.

Edzelf, 2023-04-29:
If the Radio cannot find a configured WiFi network, it will create its own.  This network has no Internet connection, so no station can be found.
But you can use your PC/laptop/tablet/phone to connect to this WiFi Network "ESPRadio".  Usually there is a "WiFi" icon to click on.  The password is also "ES32Radio".  Once the connection is made, you can use your Internet Browser to go to http://192.168.4.1 and use the config page of the webinterface to complete your settings.
The IP showed in de serial output "1.0.0.0" is strange, but you may try http://1.0.0.0 if 192.168.4.1 does not work.
There is also a simple solution.
Edit the file include/config.h and put your WiFi network name and password in line 15, like this:
#define FIXEDWIFI "nnnnn/ppppp"
nnnnn is the network name, ppppp is the password.  Recompile and upload to the radio.  Then the radio will connect to your network right away.

The-Valveman, 2023-04-29:
Hi Ed 
I have connected to the WiFi network set up by the ESP32Radio and accessed the config page on my phone i have added my home WiFi credentials as well as the pins for the I2S output then clicked the c
'SAVE' button then restarted the ESP32 but it still will not connect to my home WiFi i have double checked the ssid and password which are correct but it seems to be ignoring the config file as when i  restart it still say 'WiFi failed! trying to set up AP with the name ESP32-Radio and password ESP32_Radio' as per below copied from the serial monitor:-

D: 110580/259960 - Starting ESP32-radio running on CPU 1 at 240 MHz.
D: 110580/259960 - Version Tue, 22 Nov 2022 15:30:00 GMT.  Free memory 110580
D: 110580/259960 - Display type is NEXTION
D: 110580/255600 - SPIFFS is okay, space 1016801, used 33634
D: 110580/255564 - Found partition 'nvs     ' at offset 0x009000 with size    20480
D: 110580/255564 - Found partition 'otadata ' at offset 0x00E000 with size     8192
D: 110580/255564 - Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
D: 110580/255500 - Read 40 keys from NVS
D: 110580/255316 - pin_ir set to 35
D: 110580/255316 - pin_tft_cs set to 15
D: 110580/255316 - pin_tft_dc set to 2
D: 110580/255316 - pin_sd_cs set to 13
D: 110580/255316 - pin_spi_sck set to 18
D: 110580/255316 - pin_spi_miso set to 19
D: 110580/255316 - pin_spi_mosi set to 23
D: 110580/255428 - GPIO0 is HIGH
D: 110580/255428 - GPIO2 is LOW, probably no PULL-UP
D: 110580/255428 - GPIO4 is HIGH
D: 110580/255428 - GPIO5 is HIGH
D: 110580/255428 - GPIO12 is HIGH
D: 110580/255428 - GPIO13 is HIGH
D: 110580/255428 - GPIO14 is HIGH
D: 110580/255428 - GPIO15 is HIGH
D: 110580/255428 - GPIO16 is HIGH
D: 110580/255428 - GPIO17 is HIGH
D: 110580/255428 - GPIO18 is HIGH
D: 110580/255428 - GPIO19 is HIGH
D: 110580/255428 - GPIO21 is HIGH
D: 110580/255428 - GPIO22 is HIGH
D: 110580/255428 - GPIO23 is HIGH
D: 110580/255428 - GPIO25 is HIGH
D: 110580/255428 - GPIO26 is HIGH
D: 110580/255428 - GPIO27 is HIGH
D: 110580/255428 - GPIO32 is HIGH
D: 110580/255428 - GPIO33 is HIGH
D: 110580/255428 - GPIO34 is LOW, probably no PULL-UP
D: 110580/255428 - GPIO35 is LOW, probably no PULL-UP
D: 110580/255428 - GPIO39 is LOW, probably no PULL-UP
D: 110580/255428 - gpio_00 will execute uppreset = 1
D: 110580/255428 - gpio_12 will execute upvolume = 2
D: 110580/255196 - Enable pin 35 for IR
D: 110580/254812 - Start NEXTION display
D: 110580/254812 - Init Nextion, pins -1, -1
D: 110580/252768 - Nextion command cls BLACK
D: 110580/252768 - Nextion command page 0
D: 110580/252768 - NEXTION output 'Starting......' to 1
D: 110580/252768 - NEXTION output 'Tue, 22 Nov 2022' to 2
D: 110580/252768 - NEXTION output 'By Ed Smallenburg' to 3
D: 110580/252768 - Create list with acceptable WiFi networks
D: 110580/252552 - Added BT-6FF8FG to list of networks
D: 110580/252440 - Added SSID2 to list of networks
D: 110580/206340 - Scan Networks
D: 110580/205556 - Scan completed
D: 110580/205556 - Number of available networks: 9
D: 110580/205556 -  1 - BT-6FF8FG                 Signal: -60 dBm, Encryption WPA2_PSK, Acceptable
D: 110580/205556 -  2 - BTWi-fi                   Signal: -63 dBm, Encryption OPEN, 
D: 110580/205508 -  3 - Three_003179              Signal: -76 dBm, Encryption WPA2_PSK, 
D: 110580/205476 -  4 - PETER_WORKSHOP            Signal: -85 dBm, Encryption WPA2_PSK, 
D: 110580/205488 -  5 - BTWi-fi                   Signal: -90 dBm, Encryption OPEN, 
D: 110580/205476 -  6 - BT-HXCKWJ                 Signal: -90 dBm, Encryption WPA2_PSK, 
D: 110580/205460 -  7 - SKY8D9C6                  Signal: -91 dBm, Encryption WPA2_PSK, 
D: 110580/205460 -  8 - BTWi-fi                   Signal: -91 dBm, Encryption OPEN, 
D: 110580/205444 -  9 - BT-MHF6K7                 Signal: -93 dBm, Encryption WPA2_PSK, 
D: 110580/205444 - End of list
D: 110580/205028 - nextPreset is 0
D: 110580/190612 - Connect to network
D: 110580/190612 - NEXTION output 'Connect to network' to 4
D: 110580/190832 - WiFi Failed!  Trying to setup AP with name ESP32-Radio and password ESP32-Radio.
D: 110580/188176 - IP = 1.0.0.0
D: 110580/188216 - NEXTION output 'IP = 1.0.0.0' to 5
D: 110580/188092 - Nextion command page 1
D: 110580/188092 - Start web server
D: 110580/169696 - Rotary encoder is disabled (-1/-1/-1)
D: 110580/169696 - nextPreset is 6
D: 110580/167320 - Starting I2S playtask..
D: 63476/105620 - Output to I2S, pins -1, -1 and -1
D: 63476/105620 - I2S setpin error!
D: 63476/105420 - Nextion t0.txt="ESP32-Radio"
D: 63476/105420 - Volume set to 72

Edzelf, 2023-04-29:
The ESP should connect to network "BT-6FF8FG".
This sometimes fails.  Check your powersupply.  It should be like 5 Vol, 2 Amp.
Sometimes it helps to clear all the Flash memory (in the PlatformIO ([image]) menu, "Erase Flash".

The-Valveman, 2023-04-29:
> Sometimes it helps to clear all the Flash memory (in the PlatformIO ([image:
> afbeelding]

Hurrah it's working. Erasing the flash sorted it thankyou.  All i got to do
now is get the SD card working and change the I2S pins as i am not using
the encoder for control I will use pins 25 26 27 for the I2S as these pins
are used on my pcb for this device and integrate it into the receiver using
the receiver MCU to parse the Nextion commands to the ESP when the Internet
radio or SD card functions are used.

The-Valveman, 2023-04-29:
Success clearing the flash sorted it thankyou
All I have to now is get the SD Card working

closed 2023-04-29 by The-Valveman