Edzelf/ESP32Radio-V2 #97: DAC Problem?

wa8lbz, 2023-04-26:
I have 17 presets in the ESP32 Radio 2.0 and they all play perfectly but one. 198.178.123.8:8454/;stream/1 sounds speeded up like playing a 33rpm record at 78rpm. I am using the PCM5102A DAC with the FLT, DMP, SCL and FMT pins set low (grounded). The XMT pins set high (3.3V).

I started the ESP32 Radio 2.0 project using the VS1053 DAC and don't remember having this problem. The source has not changed because when I type the above IP address into my browser it plays perfectly. Is it posssible that this a DAC problem?

Edzelf, 2023-04-28:
This is a mono station.  The software does not handle this correct in combination with I2S devices. Sofar I did not find a solution.  The VS1053 works correct.

Edzelf, 2024-02-20:
Mono stations shoud work in the latest version.