Edzelf/ESP32Radio-V2 #100: Unable to find a function for stop/resume command

dkotte, 2023-05-18:
I am unable to find a function for stop/resume command.
I have to send a stop/resume command independent of previous state. "Mute" toggles with every command. 
This is a problem, if I didn't know the current state.
Within main.cpp I've found:
//**************************************************************************************************
//                                     A N A L Y Z E C M D                                         *
//**************************************************************************************************
// Hand...
//   stop                                   // Stop playing                                        *
//   resume                                 // Resume playing                                      *
... but no corresponding function ... and it doesn't work

Edzelf, 2023-05-19:
In the next release, there will be 2 buttons for the mute function:

[image]

trip5, 2023-11-25:
I know it's a bit of an old issue here but perhaps still relevant to the original question...

I'd like for the radio to be always on but to not use data (I'm planning on putting inside a massage chair with no display).

I did my best to dig into the code and it looks like mute merely puts the volume to zero.  And perhaps this was the original request... is it possible to add "pause" and "play" to both mqtt and a gpio/IR command?

trip5, 2023-11-26:
Actually I do have a workaround for now.

Make a preset that points to nothing (ie preset_199 = .mp3) and that basically stops the radio. This can be activated via MQTT... 
 and probably a GPIO.

Although it looks like I may be better off just putting in a power switch as the speakers have a bit of a static/hum to them while the radio is on.