Edzelf/ESP32Radio-V2 #103: sd module fails

stefi01, 2023-05-31:
So I managed to get everything working including the ir but the sd module fails, I tried the sd Card built into the  ILI9341 2.8" screen that I used and works fine  but no luck with the sd on the back so tried a 3.3v and a 5v stand alone module that you get with Arduino kits, still failed and not recognized, the wiring is correct, MOSI TO MOSI pin 23, MISO TO MISO pin 19,  SCK pin 18, SD CS TO  21 and verified the pin in the config file,, just confused how the pins match and yet it fails to initialize.

aly-fly, 2023-05-31:
I have similar situation - micro SD card reader on the back of the LCD does not work. There are some extra components and one chip on PCB which I haven't analyzed yet.

But the large (classic) SD-card reader on the MP3 decoder module works fine.
Supplied with 3.3V from ESP32 dev board.

stefi01, 2023-05-31:
I never used the mp3 module as built my own but it has no sd card slot, sounds from the radio side works fine so need still to work out how to get a stand alone sd card module working, wiring is all good and is wired according to the main sketch

aly-fly, 2023-05-31:
Check that you have short wires, max 10 cm. Connect very good ground. Reduce SPI clock frequency.

stefi01, 2023-06-01:
> 

or do I change something here to lower it? not sure what to change

  write_register ( SCI_CLOCKF, 6 << 12 ) ;              // Normal clock settings
    // multiplyer 3.0 = 12.2 MHz
    //SPI Clock to 4 MHz. Now you can set high speed SPI clock.

Edzelf, 2023-06-01:
Do not change the SPI speed for the VS1053.  Change the speed for the SD card in SDcard.h, line 24:

`#define SDSPEED 8000000                               // SPI speed of SD card`