Edzelf/ESP32Radio-V2 #104: Issue with ethEvent function - SYSTEM_EVENT doesn't work but ARDUINO_EVENT does

goscickiw, 2023-06-04:
I have a WT32-ETH01 module, and I compiled the program in PlatformIO for it to work via Ethernet. However it didn't seem to be able to make a connection to the Icecast server, despite obtaining an IP address and being able to make a DNS request.

I set the debug level to 5 so that I get all messages in the terminal, and I saw that some lines showed that it got an IP address, but one showed just `IP = ` without the address. I looked through the code and found that this should display the address from `ipaddress` variable, which is set in the `ethEvent` function.

Then I saw that what should have been displayed as:
`ETH Started`
`ETH cable connected`
`IPv4: [address], 100 Mbps, FULL_DUPLEX`
actually displayed as:
`ETH event 18`
`ETH event 20`
`ETH event 22`

So the `ethEvent` function didn't detect the events properly. I have tested the WT32-ETH01 before using some [example code](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino), and it was working, and I noticed that it uses:
`ARDUINO_EVENT_ETH_START`
`ARDUINO_EVENT_ETH_CONNECTED`
`ARDUINO_EVENT_ETH_GOT_IP`
`ARDUINO_EVENT_ETH_DISCONNECTED`
`ARDUINO_EVENT_ETH_STOP`
instead of:
`SYSTEM_EVENT_ETH_START`
`SYSTEM_EVENT_ETH_CONNECTED`
`SYSTEM_EVENT_ETH_GOT_IP`
`SYSTEM_EVENT_ETH_DISCONNECTED`
`SYSTEM_EVENT_ETH_STOP`
So I replaced these and now everything is working fine.

However I thought that you must have had it working with `SYSTEM_EVENT_...` instead of `ARDUINO_EVENT_...`, otherwise you would have used `ARDUINO_EVENT_...` instead. So what could be causing it to not work in my case?

Edzelf, 2023-06-04:
You ar right about renaming the events.  It has something to do with newer versions of some libraries.  I got an IP address now, but still no connection.  I will try to find a solution.

goscickiw, 2023-06-05:
Thank you for answering. So would it be better if I leave these events as `ARDUINO`, or change them back to `SYSTEM` and use an older library?

Edzelf, 2023-06-05:
The best way is to use the new definiotions.

goscickiw, 2023-06-05:
Thanks. This issue is solved for me now, I'm not closing it yet because I'm not sure if you want it closed.

Edzelf, 2023-06-05:
So you do have connection with the server?  I still have problems, but it could be the WT32 itself.

goscickiw, 2023-06-05:
I can connect and play sound, but only with an MP3 stream as described in #105.

I only remember changing the board to wt32-eth01 and the COM port in platformio.ini, and later changing the `SYSTEM` events to `ARDUINO` when troubleshooting before it started to work. When I'm back at home I will check if I changed anything else.