Edzelf/ESP32Radio-V2 #105: OGG stream does not play

goscickiw, 2023-06-04:
I have a WT32-ETH01 board connected to VS1053. I use these pins:
```
pin_vs_cs = 15                                       # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 4                                       # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 35                                     # GPIO Pin number for VS1053 "DREQ"
#
pin_spi_sck = 14                                     # GPIO SPI SCK
pin_spi_miso = 36                                    # GPIO SPI MISO
pin_spi_mosi = 12                                    # GPIO SPI MOSI
```
MP3 streams work well, but when I try to open an OGG stream it connects, but doesn't play any sound. If I wait too long before switching back to MP3 the entire program crashes. I have measured the pins with a logic probe and it seems like it's trying to send something to the VS1053, but there's no sound. I'm not sure if this is an issue with VS1053 or with the program.

Here is the debug output (the line that says `Request stream.nightride.fm/nightride.ogg failed!` doesn't seem to be important, it shows the same with an MP3 stream despite everything working normally, and it seems to be because the `connecttohost` function is made in a way that shows the "failed" message even if it successfully connects):

```
D: 110580/195896 - Connect to host stream.nightride.fm/nightride.ogg
D: 110580/195520 - Connected to host at 51.81.80.91 on port 80
D: 110580/195608 - send GET command
D: 110580/194024 - Request stream.nightride.fm/nightride.ogg failed!
D: 110580/194008 - Switch to HEADER
D: 110580/194008 - Headerline: Accept-Encoding: identity
D: 110580/194008 - Headerline: Access-Control-Allow-Origin: *
D: 110580/194008 - Headerline: Allow: GET, DELETE, OPTIONS
D: 110580/194008 - Headerline: Cache-Control: no-cache, no-store
D: 110580/194008 - Headerline: Content-Type: application/ogg
D: 110580/193960 - Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: 110580/193960 - Headerline: Ice-Audio-Info: channels=2;samplerate=48000
D: 110580/193960 - Headerline: Icy-Name: nightride.ogg
D: 110580/193960 - Headerline: Icy-Pub: 1
D: 110580/193960 - Headerline: Pragma: no-cache
D: 110580/193960 - Headerline: Server: Caddy
D: 110580/193960 - Headerline: Server: Icecast 2.4.99.3
D: 110580/193960 - Switch to DATA, bitrate is 0, metaint is 0
E (202583) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (202583) task_wdt:  - async_tcp (CPU 0/1)
E (202583) task_wdt: Tasks currently running:
E (202583) task_wdt: CPU 0: IDLE
E (202583) task_wdt: CPU 1: loopTask
E (202583) task_wdt: Aborting.

abort() was called at PC 0x400fb1b8 on core 0

Backtrace: 0x40083fc9:0x3ffbef0c |<-CORRUPTED

  #0  0x40083fc9:0x3ffbef0c in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408

ELF file SHA256: 89d16df0d7b209fe

E (11978) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

Edzelf, 2023-06-04:
I tested it and I have the same bad results.  I will try to find the cause.

goscickiw, 2023-06-05:
Thanks for answering. As for the "request failed" message, would the following changes in main.cpp be a good way to fix it?
* In `connecttohost()`, add a `return true ;` right after `mp3client->write ( getreq, strlen ( getreq ) ) ;`
* In `radiofuncs()`, use `connected = connecttohost() ;` instead of
```
connecttohost() ;
connected = true ;
```

Edzelf, 2023-06-05:
Your fix is okay.  But it won't solve the ogg problem.
The staiton at stream.nightride.fm/nightride.ogg sends "Ogg Opus".  Ogg Opus is not supported by the VS1053, Ogg Vorbis should play well.
Try `soul.audiogrooves.net/deeply.ogg`

goscickiw, 2023-06-05:
Thanks, looks like I will have to use the MP3 version of this station. The fix was just for the debug message.

I will test the link when I'm back at home, but I presume that it will work, now that I know the cause of the issue.

goscickiw, 2023-06-05:
I tested with the link, the first time I got the same result as before, but after a few more tries it seems like sometimes it loads and sometimes there's the same issue. Also it stops playing after a few minutes which doesn't happen with an MP3 stream. So it seems loke there's still some issues with OGG.

Also I noticed that after it crashes and reboots, it doesn't seem to initialize VS1053 properly and gets stuck. I presume it's because it would have to reset VS1053, and I currently have the VS1053's reset pin pulled up to 5V with a 1k resistor. Would this be correct?

Also there doesn't seem to be a setting for VS1053 reset pin, but there is one for `pin_shutdown` or `pin_shutdownx`. Should I set it and connect to the reset pin?

Edzelf, 2023-06-05:
If you use a reset button (pulling the EN pin down), you can connect the reset pin of the VS1053 to the EN pin.
the shutdown pin is used to disable/enable an amplifier.
I cannot reproduce the ogg-error, but I tested `icecast.ithost.it:8000/radiosei.ogg`.

goscickiw, 2023-06-05:
Now I have downloaded the newest version of the software with today's commits, now both of the OGG links you provided start properly 100% of the time. There still seems to be some issues with stopping after a few minutes and then crashing, more specifically with the first link `soul.audiogrooves.net/deeply.ogg`. However I noticed a new debug message  right before it crashes:
```
[469240][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[469440][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[469640][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[469840][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[470040][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[470240][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[470440][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[470640][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[470840][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[471040][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[471240][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[471440][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[471640][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[471840][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[472040][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[472240][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[472440][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[472640][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[472840][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[473040][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[473240][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[473440][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[473640][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[473840][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
[474040][E][main.cpp:3597] handlebyte_ch(): [main] MP3 packet dropped!
```
Also it starts making weird noises when it starts displaying these messages, and doesn't stop until I power cycle the device. I wonder if this could just be the SPI wiring, which is currently on a breadboard instead of proper PCB traces.

Also I haven't connected a reset button yet, until now I have reset by power cycling. I meant that the ESP32 itself would need some way to reset the VS1053 after it auto-reboots.

Edzelf, 2023-06-05:
The ESP32 will do a soft reset of the VS1053 at start-up.

goscickiw, 2023-06-05:
The soft reset doesn't seem to be enough after the crash and reboot, VS1053 is locked up until manual reset.