Edzelf/ESP32Radio-V2 #106: Need Help!!!

buddyroar, 2023-06-05:
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None, Disabled"

C:\Users\Buddy\Documents\Arduino\libraries\utils\utils.cpp:29:6: error: ambiguating new declaration of 'void dbgprint(const char*, ...)'

 void dbgprint ( const char* format, ... )

      ^~~~~~~~

In file included from C:\Users\Buddy\Documents\Arduino\libraries\utils\utils.cpp:5:

C:\Users\Buddy\Documents\Arduino\libraries\utils\utils.h:11:13: note: old declaration 'char* dbgprint(const char*, ...)'

 char*       dbgprint ( const char* format, ... ) ;                 // Print a formatted debug line

             ^~~~~~~~

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\Buddy\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.6\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Edzelf, 2023-06-05:
Looks like you have mixed versions.  Please load the complete zip-file, upload sketch and data dairectory.

buddyroar, 2023-06-05:
Could you explain what you mean by mixed versions?

closed 2023-06-05 by buddyroar

Edzelf, 2023-06-05:
Sorry, my fault.  Just a moment.

Edzelf, 2023-06-05:
Correct version is now available.

buddyroar, 2023-06-05:
Thank you, I will try to upload the new version once I’m home tonight.