Edzelf/ESP32Radio-V2 #107: Library Issue?

buddyroar, 2023-06-06:
I redownloaded the code from last time and received a new error than last time.

Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None, Disabled"

c:/users/buddy/documents/arduinodata/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\VS1053\VS1053.cpp.o:(.literal._ZN6VS10538testCommEPKc+0x18): undefined reference to `dbgprint(char const*, ...)'

c:/users/buddy/documents/arduinodata/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\VS1053\VS1053.cpp.o: in function `VS1053::testComm(char const*)':

C:\Users\Buddy\Documents\Arduino\libraries\VS1053/VS1053.cpp:112: undefined reference to `dbgprint(char const*, ...)'

c:/users/buddy/documents/arduinodata/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Buddy\Documents\Arduino\libraries\VS1053/VS1053.cpp:113: undefined reference to `dbgprint(char const*, ...)'

c:/users/buddy/documents/arduinodata/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Buddy\Documents\Arduino\libraries\VS1053/VS1053.cpp:132: undefined reference to `dbgprint(char const*, ...)'

c:/users/buddy/documents/arduinodata/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Buddy\Documents\Arduino\libraries\VS1053/VS1053.cpp:142: undefined reference to `dbgprint(char const*, ...)'

c:/users/buddy/documents/arduinodata/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\VS1053\VS1053.cpp.o:C:\Users\Buddy\Documents\Arduino\libraries\VS1053/VS1053.cpp:293: more undefined references to `dbgprint(char const*, ...)' follow

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\Buddy\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.6\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

buddyroar, 2023-06-06:
I am using a 128 x 160 LCD display with a ST7735S Driver Chip - along with the Adafruit ST7735 and ST7789 library.

Edzelf, 2023-06-07:
No need to import an VS1053 library from another source.  It is available in the /lib/codecs directory:
[image]