Edzelf/ESP32Radio-V2 #108: No audio

apadt, 2023-06-06:
Hi!
Finally uploading worked :-) 
Have connection via the Web interface, can change channels but if have no audio output. 
Checked in conf the vs1053 settings:
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 16                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
(Have no rotary encoder installed...)

Tips
Hartelijk dank/thanks
AJ

Edzelf, 2023-06-07:
In the logging, do you see the following lines without errors:
```
[ 11734][I][VS1053.cpp:127] testComm(): [VS1053] Slow SPI, Testing VS1053 read/write registers...
[ 11982][I][VS1053.cpp:127] testComm(): [VS1053] Fast SPI, Testing VS1053 read/write registers again...
```
This will test the SPI bus connection to the VS1053.

apadt, 2023-06-07:
Hi!
Thanks, now it works a small wire mistake :-) Thanks for the nice project!
Hartelijk dank vanuit Oslo!
AJ

chris200, 2024-03-01:
I am getting the SPI error, no audio.  Using the exact pin out in the main file.  Update config file, using Esp32, VS1053 (dated 2022-11-18), and a TFT with 9341.

The TFT is displaying correctly but no audio.

I hacked together the esp8266 library with the same pinouts and it plays the audio.  Any clues would be great.

chris200, 2024-03-01:
> I am getting the SPI error, no audio. Using the exact pin out in the main file. Update config file, using Esp32, VS1053 (dated 2022-11-18), and a TFT with 9341.
>
> The TFT is displaying correctly but [...]

I found the issue:  I unplugged the display and I got audio.  I had the touch pins connected to the ESP32.  As soon as I disconnected them everything works.