Edzelf/ESP32Radio-V2 #119: onError(): [main] MP3 host error DNS failed

Ephesian, 2023-07-13:
Hi
I am trying to connect to an internet radio that only has an ip address not a DNS entry and I get the following message

onError(): [main] MP3 host error DNS failed, which is what I would expect from giving IP address to DNS.

The radio is but stream is (http://192.111.140.11:8272/stream)

I suspect that no check is done and code assumes DNS in connecttohost

Eph.

dheydeck, 2023-07-16:
Hello- I have number of stations accessed via IP address rather than DNS:
```
preset_12 = 178.19.96.111:8007/                      # 12 - Radio Derf: Blues Delta
preset_14 = uk4-vn.mixstream.net:8018/               # 14 - Blue Skies Radio, London
preset_15 = 199.101.51.168:8004/stream               # 15 - Blues Connection
preset_16 = 149.255.59.3:8232/                       # 16 - Blues Radio: UK
preset_17 = hts06.kshost.com.br:10182/               # 17 - 61-Blues
preset_18 = 51.161.115.200:8008/                     # 18 - A Mississippi Blues
preset_19 = 62.171.171.202:8090/                     # 19 - RDMIX Relaxing Blues
preset_20 = 149.56.234.138:8173                      # 20 - 95.7 The Wolf
preset_21 = 192.99.4.205:8596                        # 21 - Radio Maquiavelli
preset_30 = 178.19.96.111:8005/                      # 30 - Radio Derf - Jazz
```
I suspect that your issue may be the `http://` prefix- perhaps try without?
Having said that, I've pasted the `http://192.111.140.11:8272/stream` url into the "Play" field in the Control tab and it plays (as does the bare address), albeit very choppily.

Edzelf, 2023-07-16:
I have also problems with 192.111.140.11:8272/stream.
I cannot reproduce the DNS error.

Edzelf, 2023-07-16:
The problem with 192.111.140.11:8272/stream is that the stream is mono.  The software does not handle this correctly.

Ephesian, 2023-07-17:
Thanks 

Yes its the HTTP:// my bad.