Edzelf/ESP32Radio-V2 #120: Sync TOD | Failed to obtain time

igor756, 2023-07-17:
Hello, thank you for nice project.
I connected only ESP32 + VS1053B (slightly different PCB design then your).

After uploading default config through web (no change, only wifi) I am receiving following error in serial monitor:
[ 46647][I][main.cpp:4347] gettime(): [main] Sync TOD
[ 51657][W][main.cpp:4350] gettime(): [main] Failed to obtain time!

The sound is OK and I can change radio stations in web.
Can you advise where the problem can be?

Thank you and have a nice day.

Edzelf, 2023-07-17:
It may take some time before the clock is synchronized.  The message will disappear after some time.

igor756, 2023-07-19:
Hmm, waited for 1 hour and no luck.
Added rotary encoder and now I am receiving:
[ 89830][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 91934][I][main.cpp:4347] gettime(): [main] Sync TOD
[ 96944][W][main.cpp:4350] gettime(): [main] Failed to obtain time!

Encoder works for mute/unmute and preset, though not working for volume control.
Any idea please?

Edzelf, 2023-07-25:
Are you sure you have a working NTP timeserver configured?
See paragraph 2.2 in the documentation.
```
clk_dst = 1
clk_offset = -1
clk_server = pool.ntp.org

```

igor756, 2023-08-01:
Now lot of different errors there, ie:
E (43750) gpio: gpio_set_level(226): GPIO output gpio_num error
E (43764) gpio: gpio_set_level(226): GPIO output gpio_num error
E (43769) gpio: gpio_set_level(226): GPIO ouGuru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (Playtask)
...
Backtrace: 0x400925f4:0x3ffdbfc0 0x400907ad:0x3ffdc000 0x4008ec14:0x3ffdc030 0x4008ebc4:0xa5a5a5a5 |<-CORRUPTED

I will try to change ESP32 and VS1053 for different pieces to see if it is not a hardware error.
Thank you for help so far.

Edzelf, 2023-08-01:
The error means that a gpio pinnumber is undefined, usually -1.Verzonden vanaf mijn Galaxy

"target": "https://github.com/Edzelf/ESP32Radio-V2/issues/120#issuecomment-1660547098",
"url": "https://github.com/Edzelf/ESP32Radio-V2/issues/120#issuecomment-1660547098",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

Edzelf, 2023-08-02:
I miss the following settings in the debug logging:
```
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4

```
Your pins are probably different, but you should see them in the output.