Edzelf/ESP32Radio-V2 #122: Schematic wiring diagram

BruceKenley, 2023-08-05:
Sorry to ask such a basic question where do l find a schematic?
Hope you can help
Does anyone have a pcb for sale?
Cheers
Bruce

Dudinu, 2023-08-05:
Hi Bruce !
This is my version of schematic for the excellent Web Radio of Edzelf. Hope it  helps.
[image]

BruceKenley, 2023-08-06:
Thanks l am running a il9341 tft. I think l will be able to change circuit for my tft.

pwilkowski, 2023-08-25:
I will be making custom PCB for this without any modules whatsoever, ie bare chips, with ability to order assembled on jlcpcb

Don't like the idea of putting together 5 different modules, will post soon link to repo.