Edzelf/ESP32Radio-V2 #123: Full Discrete schematic (no modules) - validation needed

pwilkowski, 2023-08-26:
Hello, I have created schematic for this radio with one goal in mind - no modules, everything done with discrete components and possible to order from JLC pcb fully assembled.

Additional features:
FM Radio on SI4702-C19-GMR chip
Audio DSP and selector on TLV320AIC3204
USBC connector power (its still usb 2.0)
CH376T as both USB stick or SD card reader

I can handle necessary code changes

Before i shot myself in the foot, i would like to confirm my schematic with you:
https://cdn.discordapp.com/attachments/710084652798312518/1145003214626897920/Esp32Radio.pdf

aly-fly, 2023-08-29:
Cool idea. However I don't fully understand why you want to use CH376 chip. To read SD card? This is not needed, because ESP32 can easily handle SD card directly connected to SPI bus. Or to be able to connect USB to the PC and access SD card without the need to remove it? Or to have support for external flash drive? A bit of overengineering if you ask me. I would remove this chip and connect SD card directly to ESP32. 
FM addition seems nice, but not sure if needed in this day and age. Could be useful when your WiFi is down, but this means major rework of the provided code, to be able to startup without wifi.
I am not sure if you checked the bias DC levels of the analog signal lines from multiple sources and destinations. It may be required to add DC blocking caps in the signal path. Just a word of warning.
I would remove the Preamp section, as you just need two analog switches to change between two sources. Or is FM chip without possibility of volume control?
PAM8403 seems a reasonable choice for tiny speakers. If you are trying to make a quality radio, add 2x RCA or a 3.5 mm stereo jack as a line-out connection for external amplifier, or possibility for a piggy-back of a serious amplifier board that can drive at least a few inch diameter speakers. 2x 20 Watt RMS minimum that can drive a nice bass.

pwilkowski, 2023-08-29:
@aly-fly 
CH367 is to read flash sticks which esp can't do and i have couple laying around. This chip costs like 0.5$ so its nothing.

SD card is connected to it so have unified reading mechanism, at least that is the idea. Not sure how it works in practice.

FM is needed because believe it or not, some people likes to hear old radio stations that not necessarily have web version. This will also allow it to be used without wifi.

I added preamp because i want to control typical functions like volume bass/treble and maybe a bit more fine tuning. If you know chip that does better job at this im open for suggestions.

As for signals i know FM radio ouputs like 80mVpp so it will need some opamp in between because VS chip is 1.8Vpp

Long story short, this is a radio that suppose to work with anything imaginable. I can add connectors before preamp but by default i want to use simple wideband speakers so it does work as garden radio.

aly-fly, 2023-08-29:
Ok. As someone might want to replicate your PCB, I would suggest going a bit wider, hence the comment for Line Out or optional powerful amp. My radio feeds two real 300 W speakers so it is more than a garden background music. :)

pwilkowski, 2023-08-29:
Also i forgot to mention but preamp is cheaper than trying to do any other kind of input selector. Do you have any suggestions on bumping peak levels of FM radio? ie preamp?